dsmrreader/dsmr-reader #798: Second smart meter

remyderuysscher, 2019-12-15:
Hi,

I would like to ask for support for an additional smartmeter. I have 2 installed for different purposes (house and EV-charging) with different energy contracts. It would be nice to have them both in 1 DSMR-Reader instance. 

It this possible?

dennissiemensma, 2019-12-16:
Hi Remy, thanks for asking. Unfortunately I don't think I can add support for it. 

The main reason being that both the datalogger and application only support a single smart meter simultaneously. The entire design of the project alwasy has been around a single meter.

You could try however to read both smart meters with a Python script and merge the input yourself, and then try sending it to the API of DSMR-reader. But I'm not sure if it will help you, as you're using different energy contracts and the data displayed will be off.

closed 2019-12-16 by dennissiemensma

remyderuysscher, 2019-12-16:
Hi Dennis, thanks for answering. The solution is then to run 2 DSMR-reader containers each on a different serial port.