dsmrreader/dsmr-reader #799: DSMR geupdate nu werkt niks meer

basvn, 2019-12-16:
Hoi Dennis,

Ik heb helaas weer problemen met een installatie die een jaar goed heeft gedraaid. Ik wou de installatie updaten omdat de pagina allemaal foutmeldingen geeft.. Zie http://winkel.eddy.frl/

Zodra ik het commando ./deploy.sh uitvoer als dsmr user, werkt het niet :

[!] Halted in pre-deployment: Either user ABORTED or script FAILED

De volgende commando's gebruikt:

```
sudo su - dsmr

source ~/.virtualenvs/dsmrreader/bin/activate
cd ~/dsmr-reader

pip3 install -r dsmrreader/provisioning/requirements/base.txt --force-reinstall

./deploy.sh
```

Heb je enig idee wat er mis kan zijn? supervisor backends draaien allemaal zo te zien ook.

dennissiemensma, 2019-12-16:
Ik denk je het beste kun beginnen met DEBUG-logging aanzetten en kijken wat er in de andere logs staat. Het is vast iets kleins: https://dsmr-reader.readthedocs.io/nl/v2/troubleshooting.html

basvn, 2019-12-17:
Ik heb debugging aangezet, en krijg dan de volgende errors te zien:

als ik ./deploy.sh 
```

 --- Clearing cache...                                                                                                                      
Traceback (most recent call last):                                                                                                          
  File "./manage.py", line 10, in <module>                                                                                                  
    execute_from_command_line(sys.argv)                                                                                                     
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/management/__init__.py", line 381, in execute_from_comman
d_line                                                                                                                                      
    utility.execute()                                                                                                                       
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/management/__init__.py", line 325, in execute            
    settings.INSTALLED_APPS                                                                                                                 
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/conf/__init__.py", line 79, in __getattr__                    
    self._setup(name)                                                                                                                       
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/conf/__init__.py", line 66, in _setup                         
    self._wrapped = Settings(settings_module)                                                                                               
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/conf/__init__.py", line 157, in __init__                      
    mod = importlib.import_module(self.SETTINGS_MODULE)                                                                                     
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/importlib/__init__.py", line 126, in import_module                                 
    return _bootstrap._gcd_import(name[level:], package, level)                                                                             
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import                                                                            
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load                                                                         
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked                                                                
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked                                                                         
  File "<frozen importlib._bootstrap_external>", line 674, in exec_module                                                                   
  File "<frozen importlib._bootstrap_external>", line 781, in get_code                                                                      
  File "<frozen importlib._bootstrap_external>", line 741, in source_to_code                                                                
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed                                                              
  File "/home/dsmr/dsmr-reader/dsmrreader/settings.py", line 47                                                                             
    LOGGING['loggers']['commands']['level'] = 'DEBUG'  
```

dennissiemensma, 2019-12-17:
Wat krijg je hier te zien:
```
sudo su - dsmr
git status
```

eterpstra, 2019-12-17:
@dennissiemensma 

Sorry, ik neem het even over van een vriend van mij . Ik zag dat ik onder zijn naam was ingelogd haha...

Als ik git-status doe dan zie ik:

```
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
```

en DSMR reader versie:

```
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
(dsmrreader) dsmr@winkelserver:~/dsmr-reader$ cat dsmrreader/__init__.py
from django.utils.version import get_version

VERSION = (2, 11, 3, 'final', 0)
```

dennissiemensma, 2019-12-17:
Wat krijg je hier te zien:
```
sudo su - dsmr
./manage.py --version
```

Draait dat goed of crasht Python daar ook?

eterpstra, 2019-12-17:
Die crasht:

```
IndentationError: unexpected indent
[!] Halted. Post-deployment script exited with non-zero code
(dsmrreader) dsmr@winkelserver:~/dsmr-reader$ ./manage.py --version
Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/management/__init__.py", line 381, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/management/__init__.py", line 325, in execute
    settings.INSTALLED_APPS
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/conf/__init__.py", line 79, in __getattr__
    self._setup(name)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/conf/__init__.py", line 66, in _setup
    self._wrapped = Settings(settings_module)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/conf/__init__.py", line 157, in __init__
    mod = importlib.import_module(self.SETTINGS_MODULE)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 674, in exec_module
  File "<frozen importlib._bootstrap_external>", line 781, in get_code
  File "<frozen importlib._bootstrap_external>", line 741, in source_to_code
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/dsmr/dsmr-reader/dsmrreader/settings.py", line 47
    LOGGING['loggers']['commands']['level'] = 'DEBUG'
    ^
IndentationError: unexpected indent
```

dennissiemensma, 2019-12-17:
Zonde. In een eerder comment miste net die onderste regel waar de foutmelding in staat.

De fout is helder. Er staat een spatie of tab of iets voor die regel met `LOGGING['loggers']['commands']['level'] = 'DEBUG'`. Die whitespaces ervoor moeten weg.

Daarna kun je het nogmaals proberen. En dan een deploy doen en daarna in de logs checken voor meer informatie. Eventueel handmatig even alles starten met `sudo supervisorctl start all`.

eterpstra, 2019-12-17:
Oke top!

Dat heeft het probleem inderdaad opgelost.. Ik heb nu echter dat de pagina niet realtime data bijhoud wat het volgende probleem was/ is.

De P1 data komt niet binnen. Ook is het nu als ik op ''Status'' druk, dat dan de webinterface crasht.

Kan ik hier ook een debug voor inschakelen? (Je kan het misschien zelf ook testen via de link)

dennissiemensma, 2019-12-17:
Ik denk dat je het beste even in de Supervisor log kan kijken van de backend: https://dsmr-reader.readthedocs.io/nl/v2/troubleshooting.html#supervisor

Dan zie je wat die aan het doen is en of er een achterstand is.

----

Kan het zijn dat het om een DSMR-v5 meter gaat en dan oude metingen niet opgeschoond worden?

Check even de grootte:
```
sudo su - postgres
psql dsmrreader

SELECT nspname || '.' || relname AS "relation",
    pg_size_pretty(pg_total_relation_size(C.oid)) AS "total_size"
  FROM pg_class C
  LEFT JOIN pg_namespace N ON (N.oid = C.relnamespace)
  WHERE nspname NOT IN ('pg_catalog', 'information_schema')
    AND C.relkind <> 'i'
    AND nspname !~ '^pg_toast'
  ORDER BY pg_total_relation_size(C.oid) DESC
  LIMIT 20;
```

dennissiemensma, 2019-12-17:
Ter vergelijking, met een DSMR-4 meter en hoop opschoning:
```
                      relation                      | total_size 
----------------------------------------------------+------------
 public.dsmr_datalogger_dsmrreading                 | 107 MB
 public.dsmr_consumption_electricityconsumption     | 61 MB
 public.dsmr_stats_hourstatistics                   | 3880 kB
 public.dsmr_consumption_gasconsumption             | 3592 kB
 public.dsmr_weather_temperaturereading             | 3272 kB
...
```

eterpstra, 2019-12-17:
Oke dat is denk ik het probleem ja.. Check die messages:

```
 public.dsmr_mqtt_message                               | 25 GB
 public.dsmr_datalogger_dsmrreading                     | 176 MB
 public.dsmr_consumption_electricityconsumption         | 26 MB
 public.dsmr_consumption_gasconsumption                 | 2168 kB
 public.dsmr_stats_hourstatistics                       | 416 kB
 public.dsmr_datalogger_meterstatistics                 | 408 kB
 public.dsmr_backend_scheduledprocess                   | 120 kB
 public.auth_user                                       | 96 kB
 public.django_admin_log                                | 96 kB
 public.django_session                                  | 96 kB
 public.auth_permission                                 | 88 kB
 public.dsmr_stats_daystatistics                        | 80 kB
 public.django_content_type                             | 72 kB
 public.django_migrations                               | 72 kB
 public.dsmr_backup_backupsettings                      | 64 kB
 public.dsmr_frontend_notification                      | 64 kB
 public.dsmr_mqtt_splittopictelegrammqttsettings        | 64 kB
 public.dsmr_mqtt_splittopicmeterstatisticsmqttsettings | 64 kB
 public.dsmr_mqtt_jsontelegrammqttsettings              | 64 kB
 public.dsmr_mqtt_mqttbrokersettings                    | 64 kB
```

eterpstra, 2019-12-17:
Ik zal mqtt eens uitschakelen

dennissiemensma, 2019-12-17:
Gooi eerst die tabel even leeg. In nieuwere versies van DSMR-reader zou dit trouwens niet meer moeten gebeuren, maar het schoont niet zelf een tabel op.

```
sudo su - postgres
psql dsmrreader

truncate dsmr_mqtt_message;
```

eterpstra, 2019-12-17:
Oke hij is bezig... Gaat wel even duren (Geen pi gelukkig maar een i3)

dennissiemensma, 2019-12-17:
Wellicht wil je daarna ook nog even 'vacuumen':
```
sudo su - postgres
vacuumdb -d dsmrreader -z -f -v
```

Kan even duren, maar zal met een i3 stukken sneller gaan.

eterpstra, 2019-12-17:
Topper!

Het is opgelost ! Dat was dus het probleem... 25 GB aan mqtt berichten haha.

Dank je wel Dennis!

closed 2019-12-17 by dennissiemensma