dsmrreader/dsmr-reader #801: Datalogger: komen geen berichten binnen

K-V-N, 2019-12-17:
Ik heb op een NUC een VM met DSMR Reader draaien en ik wil graag op een Pi 3+ de datalogger draaien. Er draait nu Ubuntu 19.10 op de Pi.

Eerder heeft met deze meter (E350) in elk geval het DSMR via Home Assistant gedraaid.

Als ik volgens de handleiding de DSMR datalogger only installeer op de Pi, komen er geen telegrammen binnen op de DSMR Reader. De reader is te bereiken vanuit de Pi. Als ik via systemctl dsmr_client_datalogger tail, krijg ik

> Starting...
> Starting...

Wat gaat er fout?

dennissiemensma, 2019-12-17:
Bedankt voor je melding. Ik heb wel wat meer informatie nodig.

Heb je in het datalogger-script wel de juiste URL en API key ingesteld?

K-V-N, 2019-12-17:
Heb je in het datalogger-script wel de juiste URL en API key ingesteld?

Ja, het IP-adres waarmee ik intern ook de GUI van DSMR Reader kan benaderen. Ik neem aan dat ik met het log ook zou moeten zien als er telegrammen binnenkomen om geparsed te worden? (Dus meer dan 'Starting...')?

dennissiemensma, 2019-12-17:
Welk script heb je gebruikt? De meeste recente van de docs? https://dsmr-reader.readthedocs.io/en/latest/installation/datalogger.html

Zo ja, er staan meerdere print()'s in en het lijkt er dan op dat het lezen van het telegram zelf misgaat. Ik zou echter wel een foutmelding verwachten. 

Nou zie ik echter dat er mogelijk nog een directive mist in de bijbehorende Supervisor config. Kun je eens proberen om deze volgende regel toe te voegen:
```
redirect_stderr=true
```

Waarbij het voorbeeld dan wordt:
```
    [program:dsmr_client_datalogger]
    command=/usr/bin/nice -n 5 /home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u /home/dsmr/dsmr_datalogger_api_client.py
    pidfile=/var/tmp/dsmrreader--%(program_name)s.pid
    user=dsmr
    group=dsmr
    autostart=true
    autorestart=true
    startsecs=1
    startretries=100
    stopwaitsecs=20
    redirect_stderr=true
    stdout_logfile=/var/log/supervisor/%(program_name)s.log
    stdout_logfile_maxbytes=10MB
    stdout_logfile_backups=3
```

Daarna:
```
    sudo supervisorctl reread
    sudo supervisorctl update
```

En opnieuw in de logs kijken.

K-V-N, 2019-12-17:
> Welk script heb je gebruikt? De meeste recente van de docs? https://dsmr-reader.readthedocs.io/en/latest/installation/datalogger.html

Ja, deze. 

> En opnieuw in de logs kijken.

```
pi@raspberrypi:~ $ sudo supervisorctl tail dsmr_client_datalogger
Starting...
Starting...
Starting...
```

Heb voor de zekerheid een nieuw Raspbian image geladen, maar krijg hetzelfde resultaat.

dennissiemensma, 2019-12-17:
Wat krijg je te zien als je het niet via Supervisor doet, maar handmatig:

```
sudo supervisorctl stop dsmr_client_datalogger
sudo su - dsmr
python3 -u /home/dsmr/dsmr_datalogger_api_client.py
```
Hier krijg ik zoiets te zien (ik heb twee instanties draaien), bij jou zou ik dan een foutmelding verwachten:
```
Starting...
Telegram read
Sending telegram to: /api/v1/datalogger/dsmrreading
Sending telegram to: /api/v1/datalogger/dsmrreading
Telegram read
Sending telegram to: /api/v1/datalogger/dsmrreading
Sending telegram to: /api/v1/datalogger/dsmrreading

```

K-V-N, 2019-12-17:
`(dsmrreader) dsmr@raspberrypi:~ $ python3 -u /home/dsmr/dsmr_datalogger_api_client.py
Starting...`

dennissiemensma, 2019-12-17:
Kun je eens met de hand proberen via `cu`? Daarmee check je of de user toegang heeft tot de poort. En of de poort wel juist is of een andere naam heeft.
https://dsmr-reader.readthedocs.io/en/latest/installation/explained.html#your-first-reading-optional

dennissiemensma, 2019-12-17:
Voor de duidelijkheid, in het script staat hardcoded `/dev/ttyUSB0`, maar wellicht wijkt het af.

En ik neem aan dat je geen andere applicatie aan de poort hebt hangen he? Er kan maar 1 applicatie tegelijkertijd luisteren op de seriele poort.

K-V-N, 2019-12-17:
> Voor de duidelijkheid, in het script staat hardcoded `/dev/ttyUSB0`, maar wellicht wijkt het af.
> 
> En ik neem aan dat je geen andere applicatie aan de poort hebt hangen he? Er kan maar 1 applicatie tegelijkertijd luisteren op de seriele poort.

Daar zal wel ergens aan liggen, `cat /dev/ttyUSB0` laat enkel lege data zien. Het moet wel dat device zijn, andere devices zitten er niet op.

dennissiemensma, 2019-12-17:
Ik weet niet of je het kunt catten. Je kunt het beste even handmatig testen met CU om het serieel uit te lezen. Stop dan wel even het Supervisor proces zodat ze elkaar niet in de weg zitten. Zie de link in mijn eerdere comment.

K-V-N, 2019-12-17:
Hmm. Dank voor je hulp. Als ik met de baudrate 115200 via`cu` kijk, gebeurt er niets. Als ik de baudrate  9600 gebruik werkt het wel. Het lijkt er dus op dat mijn meter DSMR 2.2 telegrammen verstuurt. Als ik de instellingen `read_telegram` aanpas naar DSMR 2.2, werkt komen we weer een paar stappen verder, maar de telegrammen kan de DSMR Reader niet lezen: `API error: Failed to parse telegram`

Als ik het python script aanpas om het telegram even te printen, komt dit er uit:

`/XMX5XMXABCE100080000

0-0:96.1.1(31333836353131302020202020202020)
1-0:1.8.1(04728.277*kWh)
1-0:1.8.2(05421.088*kWh)
1-0:2.8.1(00000.000*kWh)
1-0:2.8.2(00000.000*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(0000.46*kW)
1-0:2.7.0(0000.00*kW)
0-0:96.13.1()
0-0:96.13.0()
0-1:96.1.0(3238303131303038333037313836393133)
0-1:24.1.0(03)
0-1:24.3.0(191217210000)(00)(60)(1)(0-1:24.2.0)(m3)
(02493.238)
!
`

Volgens mij is deze meter vrij nieuw, vervangen vlak voordat ik het huis kocht.

dennissiemensma, 2019-12-17:
Kan het zijn dat je in DSMR-reader nog de datalogger-instellingen op DSMR v4/v5 hebt staan? Je kunt dit aanpassen via de webinterface als admin.

dennissiemensma, 2019-12-17:
Als ik namelijk je telegram in een automatisch test gooi, gaat het goed, mits de datalogger in DSMR-reader ook op v2 staat. Ik zal dat sowieso even opnemen in de documentatie als aandachtspuntje.

K-V-N, 2019-12-17:
> Kan het zijn dat je in DSMR-reader nog de datalogger-instellingen op DSMR v4/v5 hebt staan? Je kunt dit aanpassen via de webinterface als admin.

Check, dat was hem. Er komen telegrammen binnen en ze worden ook verwerkt! Tof :) Dank voor je hulp.

closed 2019-12-18 by dennissiemensma