dsmrreader/dsmr-reader #802: 502 Bad gateway

ThaStealth, 2019-12-17:
**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.11.0`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: Docker (ualex73/dsmr-reader-docker:latest)
* DSMR-protocol slimme meter *(v4 of v5)*: `v5`

Bij het opstarten van de applicatie gebeurd het volgende:
- Applicatie start en is korte tijd (< 1 minuut) bereikbaar via de website. Daarna krijg ik 504 (Gateway timeout) of 502 (Bad gateway) errors
- Zowel de logs van DSMR als Postgres geeft geen fouten, maar als ik de postgres db verplaatst en DSMR reader een nieuwe laat maken dan is de UI wel beschikbaar (en blijft beschikbaar).

Het lijkt erop alsof er een process start na 1 minuut wat niet meer stopt en waardoor de UI zich helemaal ophangt?

ThaStealth, 2019-12-17:
Nadat ik nog wat verder ben gaan experimenteren (de fout lijkt in de Postgres database te liggen) heb ik het volgende gedaan om de database te recoveren:

1. Export maken van de huidige database (ik gebruik Docker, dus niet alle commands zijn relevant:
```
//Verbinding maken met docker container van postgres
docker exec -it postgres /bin/bash

//dump van bestaande db maken (naar /var/lib/postgresql/data/ omdat dit een mountpoint is naar het host fs voor de container)
pg_dump -U dsmrreader > /var/lib/postgresql/data/dsmr.dat
```
2. Daarna de docker container/ postgres instance stoppen en de data directory mount wijzigen zodat er een nieuwe (lege) db word aangemaakt. En de dsmr.dat file moven van de oude locatie naar de nieuwe

3. Database backup opnieuw importeren
```
//Verbinding maken met docker container van postgres
docker exec -it postgres /bin/bash

//dump van bestaande db maken (naar /var/lib/postgresql/data/ omdat dit een mountpoint is naar het host fs voor de container)
psql -U dsmrreader< /var/lib/postgresql/data/dsmr.dat
```

Wellicht dat iemand anders hier in de toekomst zijn data kan recoveren ipv opnieuw te beginnen. Waardoor postgres besloten heeft dat de database corrupt is weet ik helaas niet, is niets raars gebeurd voor zover ik weet...

closed 2019-12-17 by ThaStealth

dennissiemensma, 2019-12-17:
@ThaStealth bedankt voor je melding en bevindingen. Wellicht is het goed om je SD-kaartje in de gaten te houden. Doorgaans raakt PostgreSQL niet zomaar corrupt.

Heb je ook nog de exacte foutmeldingen uit de PostgreSQL logs?