dsmrreader/dsmr-reader #803: Error bij stap volgens de handleiding

tokio69, 2019-12-18:
Beste,
ik loop vast tijdens de eerste setup van DSMR reader. Ik volg de handleiding en krijg een error op deze stap : 
sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/postgresql.txt

de error is deze : 
Requirement already satisfied: MarkupSafe>=0.23 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from jinja2->coreschema->coreapi<2.4,>=2.3->-r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt (line 2)) (1.1.1)
Building wheels for collected packages: psycopg2, psycopg2-binary
  Building wheel for psycopg2 (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-uxok69yt/psycopg2/setup.py'"'"'; __file__='"'"'/tmp/pip-install-uxok69yt/psycopg2/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-lr72zdvq --python-tag cp37
       cwd: /tmp/pip-install-uxok69yt/psycopg2/
  Complete output (40 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-aarch64-3.7
  creating build/lib.linux-aarch64-3.7/psycopg2
  copying lib/compat.py -> build/lib.linux-aarch64-3.7/psycopg2
  copying lib/errors.py -> build/lib.linux-aarch64-3.7/psycopg2
  copying lib/_json.py -> build/lib.linux-aarch64-3.7/psycopg2
  copying lib/tz.py -> build/lib.linux-aarch64-3.7/psycopg2
  copying lib/extensions.py -> build/lib.linux-aarch64-3.7/psycopg2
  copying lib/__init__.py -> build/lib.linux-aarch64-3.7/psycopg2
  copying lib/_range.py -> build/lib.linux-aarch64-3.7/psycopg2
  copying lib/_ipaddress.py -> build/lib.linux-aarch64-3.7/psycopg2
  copying lib/extras.py -> build/lib.linux-aarch64-3.7/psycopg2
  copying lib/_lru_cache.py -> build/lib.linux-aarch64-3.7/psycopg2
  copying lib/sql.py -> build/lib.linux-aarch64-3.7/psycopg2
  copying lib/errorcodes.py -> build/lib.linux-aarch64-3.7/psycopg2
  copying lib/pool.py -> build/lib.linux-aarch64-3.7/psycopg2
  running build_ext
  building 'psycopg2._psycopg' extension
  creating build/temp.linux-aarch64-3.7
  creating build/temp.linux-aarch64-3.7/psycopg
  aarch64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DPSYCOPG_VERSION=2.8.4 (dt dec pq3 ext lo64) -DPG_VERSION_NUM=110005 -DHAVE_LO64=1 -I/usr/include/python3.7m -I/home/dsmr/.virtualenvs/dsmrreader/include/python3.7m -I. -I/usr/include/postgresql -I/usr/include/postgresql/11/server -c psycopg/psycopgmodule.c -o build/temp.linux-aarch64-3.7/psycopg/psycopgmodule.o -Wdeclaration-after-statement
  In file included from psycopg/psycopgmodule.c:27:
  ./psycopg/psycopg.h:34:10: fatal error: Python.h: No such file or directory
   #include <Python.h>
            ^~~~~~~~~~
  compilation terminated.

  It appears you are missing some prerequisite to build the package from source.

  You may install a binary package by installing 'psycopg2-binary' from PyPI.
  If you want to install psycopg2 from source, please install the packages
  required for the build and try again.

  For further information please check the 'doc/src/install.rst' file (also at
  <http://initd.org/psycopg/docs/install.html>).

  error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for psycopg2

Alle stappen voordien allemaal zonder error of problemen.
Ik ben al eens helemaal herbegonnen met een verse install maar loop steeds daar op vast.
Ik gebruik een OrangePi Zero +2 H5 en de laatste Armbian 5.3.9 Buster. Mede omdat dit eMMC heeft ipv een SD kaartje en wifi.
Of dien ik echt een RPi te gebruiken met Raspbian?
Groeten
Tom

dennissiemensma, 2019-12-18:
Is de installatie van al deze packages ook gelukt?

```
sudo apt-get install -y postgresql postgresql-server-dev-all nginx supervisor git python3 python3-pip python3-virtualenv virtualenvwrapper
```

Ik denk dat je in dit geval `postgresql-server-dev-all` nodig hebt. 

DSMR-reader draait doorgaans op alle Linux-distro's waar bovenstaande packages goed installeren en draaien. Maar het kan natuurlijk zijn dat er wat exoten zijn die misschien andere packages gebruiken of ze niet ondersteunen.

dennissiemensma, 2019-12-18:
En anders kun je nog even zoeken op de inhoudelijke foutmelding: `./psycopg/psycopg.h:34:10: fatal error: Python.h: No such file or directory`

Bijvoorbeeld wat suggesties in https://github.com/psycopg/psycopg2/issues/900, al gaat het over een andere distro.

dennissiemensma, 2019-12-18:
Als alternatief kun je ook nog deze proberen:
```
sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install psycopg2-binary
```

In DSMR-reader staan al een tijd twee varianten van die driver, wellicht is het tijd dat ik de oude er definitief uit haal.

tokio69, 2019-12-19:
Deze packages nog eens installed maar was niet nodig :

postgresql-server-dev-all is already the newest version (200+deb10u3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

tokio69, 2019-12-19:
Ok, door de tweede tip met de uitleg van de error hier : 

en dan dit eerst te runnen : sudo apt install python3.7-dev
en dan verder te gaan kreeg ik geen error meer en kon verder gaan met de handleiding.

Onder het deel "Supervisor" heb ik wel weer een error, maak ik daar apart een issue van?
tom@orangepizeroplus2:~$ sudo supervisorctl reread
error: <class 'socket.error'>, [Errno 2] No such file or directory: file: /usr/lib/python2.7/socket.py line: 228

dennissiemensma, 2019-12-19:
Kan het zijn dat je `tom` gebruiker geen sudo rechten heeft? Anders kun je het nog even proberen als `root`. 

Het lijkt er namelijk wel op dat Supervisor goed geinstalleerd is, maar dat de client (supervisorctl) er niet bij kan.

En anders kun je als root even kijken of het proces in ieder geval draait:
```
sudo ps faux | grep supervisord | grep -v grep
```

dennissiemensma, 2019-12-26:
@tokio69 is dit gelukt?

closed 2019-12-31 by dennissiemensma

rippiedoos, 2020-06-06:
Wat ik mij echt afvraag, waarom er geen OS-packages worden gebruikt voor psycopg2. OK, in elk OS heten die dingen weer anders en ze zitten niet allemaal op 2.8.5, maar ze zitten allemaal wel ergens op 2.8. 

In RPM-systemen heet die package `python3-psycopg2`.
In Alpine Linux heet die package `py3-psycopg2`.
In DEB-systemen heet die package `python3-psycopg2`.

Het lijkt mij niet meer dan logisch om de driver van de bijbehorende PostgreSQL-release te gebruiken, zoals die op het OS ook wordt gebruikt. Enige wat je daarna ook nog zou moeten aanpassen is de virtualenv, dat die ook sys-packages meenemeent.

dennissiemensma, 2020-06-06:
@rippiedoos ik heb 5 jaar geleden gebruikt wat werkte en/of gangbaar was. Als er goede alternatieven zijn, dan wil ik daar best wel een keer naar kijken.

rippiedoos, 2020-06-06:
Door de package `python3-psycopg2` te installeren en de virtualenv aan te maken met `--system-site-packages` ipv `--no-site-packages` hoeven alle development-spullen er niet opgezet te worden. Zaken zoals gcc, g++, cpp en verschillende dev-packages (zoals postgresql-dev) zijn niet meer nodig. Ik heb dit met getest op Ubuntu 18.04 en ik heb de volgende packages ook meegenomen vanuit het OS, die heel erg dicht of helemaal in de buurt kwamen van de verwachte packages in base.txt:

- `python3-serial`
- `python3-coreapi`
- `python3-crcmod`

In relatie tot  site-packages is tegenwoord `--system-site-packages` de standaard. `--no-site-packages` is tegenwoordig deprecated en zou niet meer gebruikt moeten worden. En het scheelt een hoop gecompile door de psycopg2 vanuit het OS mee te nemen! De overige pip-packages zijn allemaal native python, dus daar hoeft weinig meer voor te gebeuren dan downloaden.