dsmrreader/dsmr-reader #804: Pi loopt vast door Postgress en dsmr?

nelbs, 2019-12-18:
Op mijn pi3B draai ik al geruime tijd DSMR-reader en hassbian. Tot vandaag ging dit prima.

Sinds vanmiddag merkte ik op dat de pi extreem traag is geworden. Zo traag dat ik vaak niet kan inloggen (SSH, Samba, browser) en de interface van homeassistant en dsmr-reader niet kan bereiken. 

Na een herstart is het systeem enkele minuten bereikbaar maar dit is van korte duur.  Onderstaand een screenshot van htop. Het geheugen zit bijna helemaal vol. Na het sluiten van homeassistant  kwam er geheugen vrij en werd het systeem weer even werkbaar. echter, binnen een kleine minuut liep het geheugen weer vol.  Ik heb dan ook het vermoeden dat postgress en/of dsmr hierin de boosdoener is. Ik heb vanmorgen toevallig wel dsmr-reader geupdate naar de laatste versie, kan dit misschien de oorzaak zijn?

Alvast bedankt voor de hulp

[image]

dennissiemensma, 2019-12-18:
Bedankt voor je melding. Kun je eens kijken hoe groot je database-backups zijn in `ls -lh /home/dsmr/dsmr-reader/backups/`?

Ik vermoed dat het komt doordat de backups nu in het geheugen gecomprimeerd worden en daardoor met pieken wellicht teveel gegeugen nodig hebben. Als dat zo is dan draai ik dat weer terug.

nelbs, 2019-12-18:
[image]

dennissiemensma, 2019-12-18:
Ja precies, je hebt een redelijk grote database, helemaal wanneer ze uitgepakt zijn. Ik zal #780 terugdraaien.

Je kunt even proberen of de vorige release weer beter gaat:

```
sudo su - dsmr
git checkout tags/v2.11.3
./post-deploy.sh
```

dennissiemensma, 2019-12-18:
Ik heb een fix klaarstaan, maar ik draai releases meestal zelf even 24 uur voor de zekerheid. 

Voor iedereen bij wie het een issue is (alleen grote databases vermoedelijk): https://github.com/dennissiemensma/dsmr-reader/issues/804#issuecomment-567218930

nelbs, 2019-12-18:
Top bedankt!

dennissiemensma, 2019-12-19:
Uitgebracht in nieuwe patch.

closed 2019-12-19 by dennissiemensma