dsmrreader/dsmr-reader #806: Power failure today. Raspberry Pi is responsive but website gives 'bad gateway'

Rawh, 2019-12-20:
Gaat er iets fout, maar weet je niet wat? [Bekijk de logfiles](https://dsmr-reader.readthedocs.io/nl/v2/troubleshooting.html) als startpunt.

**Wat gebruik je?**
* DSMR-reader versie *`2.12`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `Pi 2 Model B`
* DSMR-protocol slimme meter *v5`

*The story...*
Powerfailure at home this morning. Pi isn't behind UPS or anything. We restored power and Pi booted (seen the lights turn red/green) and expected it to run. Just gotten home (evening), website gives 502 bad gateway.

Logging into server tells us

> root@raspbian:/home/pi# supervisorctl
> dsmr_backend                     RUNNING   pid 448, uptime 11:25:15
> dsmr_datalogger                  RUNNING   pid 446, uptime 11:25:15
> dsmr_mqtt                        RUNNING   pid 3093, uptime 0:00:30
> dsmr_webinterface                FATAL     Exited too quickly (process log may have details)
> supervisor> stop dsmr_webinterface
> dsmr_webinterface: ERROR (not running)
> supervisor> start dsmr_webinterface
> dsmr_webinterface: ERROR (spawn error)
> supervisor>
> 

Tried the logs
`    /var/log/supervisor/dsmr_webinterface.log (Logs over de webinterface)`

> [2019-07-17 23:53:59 +0200] [17297] [INFO] Worker exiting (pid: 17297)
> [2019-07-17 23:54:00 +0200] [20856] [INFO] Booting worker with pid: 20856
> [2019-07-18 00:07:52 +0200] [20856] [INFO] Autorestarting worker after current request.
> [2019-07-18 00:07:52 +0200] [20856] [INFO] Worker exiting (pid: 20856)
> [2019-07-18 00:07:52 +0200] [24314] [INFO] Booting worker with pid: 24314
> [2019-07-18 00:23:56 +0200] [24314] [INFO] Autorestarting worker after current request.
Nothing useful, today isn't mentioned (20th)

Any tips?

Rawh, 2019-12-20:
Well I guess restarting it and waiting did "fix" it, it's hardly a correct way thought...

dennissiemensma, 2019-12-21:
Thanks for asking. It *might* be that your hardware just needs a few minutes. A Pi 2 is just barely speedy enough to keep up with DSMR-reader, which may explain the delay in booting the application again.

closed 2019-12-21 by dennissiemensma