dsmrreader/dsmr-reader #807: backup terug zetten

sjaakveenendaal, 2019-12-20:
DSMR was vast gelopen ik heb een backup in dropbox hoe krijg ik deze nu weer terug op het sd kaartje?

dennissiemensma, 2019-12-21:
Het hangt af van je skills met Linux. Je kunt waarschijnlijk via SFTP (bijvoorbeeld FileZilla) vermoedelijk inloggen via dezelfde gegevens als die je nu gebruikt voor SSH. En dan zet je de backup op een plek waar je bij kan.

sjaakveenendaal, 2019-12-21:
Filezilla geïnstalleerd vul mijn lokale ip adres in 192.168.172. en nog twee cijfers gebruiker admin en vervolgens mijn wachtwoord maar kom er niet in jullie hebben een prachtige software gemaakt maar voor een leek als ik toch wel erg moeilijk

dennissiemensma, 2019-12-22:
Kan je wel inloggen via SSH of heb je dat nog nooit gedaan?

sjaakveenendaal, 2019-12-22:
Nee heb nu wel Putty gedownload maar lukt ook niet

dennissiemensma, 2019-12-22:
Je kunt het beste de guide van raspberrypi zelf volgen: https://www.raspberrypi.org/documentation/remote-access/ssh/windows.md

sjaakveenendaal, 2019-12-22:
dank je wel maar kom er gewoon niet in

dennissiemensma, 2019-12-22:
Heb je daarnaast ook gecontroleerd of SSH uberhaupt aan staat? Op die pagina staat nog een verwijzing: https://www.raspberrypi.org/documentation/configuration/raspi-config.md

sjaakveenendaal, 2019-12-22:
dat was dus het probleem zoals ik al eerder zei ben niet echt handig 
volgende probleem is dat ik de bestanden niet met Filezilla over kan zetten op de pi de bestanden staan in een map op mijn bureaublad als ( dsmrreader-postgresql-partial-backup-2019-10-01.sql.gz )
wil dit dus terug plaatsen in /home/dsmr/dsmr-reader/backups/archive/2019/10 maar als ik dat doe krijg ik een foutmelding 

Overdracht afgerond
Alle overdrachten zijn afgerond. 104 bestanden konden niet overgedragen worden Filezilla
wat doe ik fout misschien iets met rechten?

dennissiemensma, 2019-12-22:
Wat je als alternatief kunt proberen is een wachtwoord instellen voor de `dsmr` gebruiker en dan daarmee inloggen via FileZilla.

```
sudo passwd dsmr
```

Voer daar twee keer een wachtwoord in wat je wilt hebben en probeer het dan via FileZilla als `dsmr` gebruiker.

sjaakveenendaal, 2019-12-24:
gedaan maar kan nog steeds geen map toevoegen via Filezilla in de dsmr directory

dennissiemensma, 2019-12-24:
En je logt wel in als `dsmr` via FileZilla en zit in de map (of submap) `/home/dsmr/`?

sjaakveenendaal, 2019-12-24:
log in met pi het standaard gebruikersnaam toch? en mijn wachtwoord

sjaakveenendaal, 2019-12-24:
oke dom dom hij doet het

sjaakveenendaal, 2019-12-24:
nog 1 vraag dan moet het lukken 
backup staat in home dsmr backup hoe verwijs ik dat in
zcat <PATH-TO-POSTGRESQL-BACKUP.sql.gz> | sudo sudo -u postgres psql dsmrreader

dennissiemensma, 2019-12-24:
Gewoon het absolute pad gebruiken:
```
zcat sudo mv /home/dsmr/bestandsnaam.ex | sudo sudo -u postgres psql dsmrreader
```

closed 2019-12-31 by dennissiemensma