dsmrreader/dsmr-reader #808: Vraag ivm partial backups

dimitrizetzsche, 2019-12-22:
Omwille van omstandigheden ben ik mijn RPI installatie verloren en heb ik enkel mijn partial backups kunnen recoveren.

Is er een aangewezen process om adhv de partial backups mijn DB te restoren? Ik zit nu op stap 7 (Application configuration & setup) v/h installatie proces.

dennissiemensma, 2019-12-22:
Je kunt de installatie als het goed is gewoon afmaken. Na afloop kun je dit uitvoeren: https://dsmr-reader.readthedocs.io/en/latest/installation/restore.html

Er staat daar dat het alleen op een lege database kan, maar bij een partial backup staan er alleen dagtotalen in en zou dat goed moeten gaan. Als het gelukt is, kun je in het Archief weer de dagen zien die bewaard waren.

dimitrizetzsche, 2019-12-23:
Heb een poging ondernomen maar ik krijg een error:
[image]

Wellicht omdat er al readings in mijn DB zitten met een bestaand key?

dennissiemensma, 2019-12-24:
Dat kan kloppen. Je zult het restoren tegelijkertijd moeten doen met de herinstallatie, anders krijg je dit en dat is een hoop gedoe met ID's.

Het makkelijkste is nu om even de dagstatistieken weg te gooien en dan weer opnieuw importeren.

1. Uitschakelen en deels wissen
```
sudo supervisorctl stop dsmr_backend

sudo su - postgres
psql dsmrreader

truncate table dsmr_stats_daystatistics;
truncate table dsmr_stats_hourstatistics;
```

2. Backup terugzetten

3. Inschakelen
```
sudo supervisorctl start dsmr_backend
```

closed 2019-12-31 by dennissiemensma