dsmrreader/dsmr-reader #809: Database Reset

quadrophenianl, 2019-12-23:
Goedemorgen Dennis,

Allereerst bedankt voor dit mooie stukje werk! Een kleine donatie komt er uiteraard nog aan...

Aangezien ik de laatste weken veelal aan het testen ben geweest, zijn de meeste metingen niet compleet. Komende dagen ben ik nog even bezig met wat laatste aanpassingen en dan wil ik vanaf 1 januari constant gaan meten. Vervelende dan wel is dat er oude records in staan die niet helemaal kloppen.

Is er een mogelijkheid om de database te resetten, zodat ik op 1 januari vanaf 0 kan gaan starten. Wellicht is een verse installatie een mogelijkheid, maar het liefst behoud ik mijn instellingen. Enkel de metingen wil ik graag verwijderen.

Ik hoor het in ieder geval graag van je! Fijne feestdagen alvast!

quadrophenianl, 2019-12-23:
Ik zie nu wel de mogelijkheid om de database te restoren. Eventueel kan ik de records eruit halen en dan een restore doen?

https://dsmr-reader.readthedocs.io/en/latest/installation/restore.html

Mocht het nog eenvoudiger kunnen, dan hoor ik het uiteraard graag!

dennissiemensma, 2019-12-23:
```
sudo su - postgres
psql dsmrreader

truncate table dsmr_consumption_electricityconsumption;
truncate table dsmr_consumption_gasconsumption;
truncate table dsmr_datalogger_dsmrreading;
truncate table dsmr_datalogger_meterstatistics;
truncate table dsmr_stats_daystatistics;
truncate table dsmr_stats_electricitystatistics;
truncate table dsmr_stats_hourstatistics;
```

quadrophenianl, 2019-12-24:
Perfect, dankjewel!

Ik ga even kijken hoe ik dit vanaf de eerste seconde van 2020 in een cron verwerkt krijg, of heb je hier wellicht al een scriptje voor liggen?

Nogmaals dank en fijne dagen gewenst!

dennissiemensma, 2019-12-24:
Je kunt het ook gewoon achteraf doen, ipv precies op de seconde:

```
sudo su - postgres
psql dsmrreader

delete from dsmr_datalogger_dsmrreading where timestamp::text < '2020-01-01 00:00:00';
delete from dsmr_consumption_electricityconsumption where read_at::text < '2020-01-01 00:00:00';
delete from dsmr_consumption_gasconsumption where read_at::text < '2020-01-01 00:00:00';
delete from dsmr_stats_hourstatistics where hour_start::text < '2020-01-01 00:00:00';
delete from dsmr_stats_daystatistics where day::text < '2020-01-01';

truncate table dsmr_stats_electricitystatistics;
truncate table dsmr_datalogger_meterstatistics;
```

closed 2019-12-24 by dennissiemensma