dsmrreader/dsmr-reader #811: Telegrams timestamp not correct with DSMR v5

remyderuysscher, 2019-12-25:
DSMR-Reader v2.12.1 (image: ualex73/dsmr-reader-docker:latest)
DSMR v5 meter

All telegrams get the same timestamp: date of  1-1-2000 (1 am) this is not correct. I have another docker container connected to a DSMR v4.2 meter which works correctly. Settings are 100% the same. 

I've checked the date inside the docker container. This is correct.

dennissiemensma, 2019-12-25:
Thanks for reporting. I'll need more details about the telegram, as the system time is not used for parsing telegrams. The problem could be with your meter, as I've seem many meters not specifying the right time or omitting it at all and send a factory default (2000, 2070).

Can you share one of the v5 telegrams here? You may remove the serials number(s) from them.

remyderuysscher, 2019-12-25:
/XMX5LGFXXXXXXX

1-3:0.2.8(50)
0-0:1.0.0(000101010000W)
0-0:96.1.1(4530303532303035333935353038303139)
1-0:1.8.1(000427.708*kWh)
1-0:1.8.2(000000.000*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00009)
0-0:96.7.9(00003)
1-0:99.97.0(2)(0-0:96.7.19)(000101010000W)(0000000253*s)(000101010000W)(0000000239*s)
1-0:32.32.0(00008)
1-0:52.32.0(00005)
1-0:72.32.0(00007)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(233.1*V)
1-0:52.7.0(229.4*V)
1-0:72.7.0(231.7*V)
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.000*kW)
1-0:41.7.0(00.000*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
!2E59

dennissiemensma, 2019-12-25:
It seems that your smart meter causes the issue. Its internal clock is not set properly.

This means year zero, and maps to 2000:
```
000101010000W
```

You should ask your the company providing the smart meter on how to solve this. When it's fixed in the telegrams, DSMR-reader will display the right timestamps again.

dennissiemensma, 2019-12-25:
FYI: It should look similar to (today/now):
```
0-0:1.0.0(191225142530W)
```

closed 2019-12-25 by dennissiemensma