dsmrreader/dsmr-reader #812: Foutmelding bij installatie op Ubuntu 18.04: supervisorctl

jrosier, 2019-12-25:
Ik krijg de foutmelding: The 'supervisor==3.3.1' distribution was not found .... enzovoort.
Maar met apt install supervisor is de versie 3.3.1-1.1 gewoon geïnstalleerd.
Wat kan dan het probleem zijn?

jrosier, 2019-12-25:
Aanvullend:
sudo ps faux | grep supervisord | grep -v grep 
geeft geen resultaat, ook niet na 
sudo service supervisor start

dennissiemensma, 2019-12-26:
Bedankt voor je melding. Waar krijg je de melding `The 'supervisor==3.3.1' distribution was not found` precies?

jrosier, 2019-12-26:
Ik krijg die melding als ik handmatig het commando "sudo supervisord" gebruik.
Het volgende geprobeerd:
- sudo apt remove supervisor / sudo pip install supervisor. Er wordt dan wel een andere versie geïnstalleerd, namelijk 4.1.0.
-De foutmelding is dan weg, er volgt geen melding / foutmelding bij "sudo supervisorctl reread"
en "sudo supervisorctl update".
- Er ontstaat nu een nieuw probleem: de website draait, maar zonder afbeeldingen:
[image]
Ik heb zelf het vermoeden dat er iets mis is met de python versie, ga daar nog even mee aan de slag.
Volgens de handleiding moet de map /var/www/dsmrreader eigendom zijn van de gebruiker dsmr, moet dat niet www-data zijn?

jrosier, 2019-12-26:
Ik krijg die melding als ik handmatig het commando "sudo supervisord" gebruik.
Het volgende geprobeerd:

    sudo apt remove supervisor / sudo pip install supervisor. Er wordt dan wel een andere versie geïnstalleerd, namelijk 4.1.0.
    -De foutmelding is dan weg, er volgt geen melding / foutmelding bij "sudo supervisorctl reread"
    en "sudo supervisorctl update".
    Er ontstaat nu een nieuw probleem: de website draait, maar zonder afbeeldingen:
 
[image]

Ik heb zelf het vermoeden dat er iets mis is met de python versie, ga daar nog even mee aan de slag.
Volgens de handleiding moet de map /var/www/dsmrreader eigendom zijn van de gebruiker dsmr, moet dat niet www-data zijn?

dennissiemensma, 2019-12-26:
De `dsmr` gebruiker moet eigenaar zijn omdat die de bestanden er heen kopieert (ook bij elke versie update), wanneer dit uitgevoerd wordt: 
```
./manage.py collectstatic --noinput
```

Je kunt kijken of er uberhaupt bestanden in die map staan.

jrosier, 2019-12-26:
Ik neem aan dat ik dit commando moet uitvoeren na sudo -su dsmr? En dan in de map dsmr-reader?
Dan krijg ik de volgende foutmeldingen, eindigend met "No module named 'solo'":
[image]

jrosier, 2019-12-26:
Bij gebruik van deploy.sh krijg ik de volgende meldingen:
[image]

Ik heb dsmrreader ook op een pi-2 geïnstalleerd, daar werkt het perfect maar dat ding loopt af en toe vast. Daarom probeer ik  dsmrreader nu op mijn ubuntu 18.04 server te installeren...

dennissiemensma, 2019-12-26:
Je hoeft dat command niet zomaar uit te voeren, gewoon de stappen uit de handleiding zouden afdoende moeten zijn.

De andere fout die je krijgt komt omdat de `dsmr` user niet automatisch de virtualenv laadt. In de handleiding staan een paar commands die voor de `.bashrc` nodig zijn. Als het goed ingesteld is zou je iets als `(dsmrreader)dsmr@server` moeten zien als prefix op je terminal, na een re-login als `dsmr` user.

jrosier, 2019-12-27:
Ik krijg nu tijdens de hele installatie geen foutmeldingen meer. Helaas zijn de grafische elementen nog steeds niet zichtbaar. Kan ik nog iets anders controleren?

jrosier, 2019-12-27:
Als ik in de firefox browser de volgende instelling verander: security.csp.enable = false (in plaats van true) is het probleem opgelost. Maar dat kan niet de bedoeling zijn neem ik aan? Ik had al een nginx installatie, misschien zit daar een blokkade voor onveilige content ingebouwd.

jrosier, 2019-12-27:
Na wat zoekwerk heb ik de volgende regel toegevoegd in het nginx configuratiebestand:
add_header Content-Security-Policy "default-src 'self';";
En nu werkt alles naar behoren op supervisord na: die wil niet automatisch laden na een reboot.
Dank voor het meedenken!

dennissiemensma, 2019-12-28:
Het is vreemd dat je CSP-meldingen krijgt, want voor zover ik weet doe ik daar niets mee omdat de applicatie doorgaans niet bereikbaar is op het Internet.

Wat betreft Supervisor, als die je niet via APT installeert maar via PIP, dan kan het zijn dat er geen autostart gedaan wordt in het systeem. Je kunt denk ik nog even proberen om het via APT te installeren, of uitzoeken hoe je de PIP-variant automatisch gestart krijgt:

http://supervisord.org/installing.html#installing-a-distribution-package
> A feature of distribution packages of Supervisor is that they will usually include integration into the service management infrastructure of the distribution, e.g. **allowing supervisord to automatically start when the system boots**.

closed 2019-12-31 by dennissiemensma