dsmrreader/dsmr-reader #814: MLOEA ondersteuning

LucienJoppe, 2019-12-27:
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.12.1`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `RaspberryPi 3.`
* DSMR-protocol slimme meter *(v4 of v5)*: `v5`

Beste Dennis,
Allereerst hulde voor je dit geweldige resutaat, echt super wat je neergezet hebt !!
Ik heb een MLOEA situatie (Meerdere Leveranciers Op Eén Aansluiting). Dit houdt in dat ik 2 slimme meters heb en zou graag beide uitlezen. 
Een paar vragen:
- Is het mogelijk om meerdere meters uit te lezen in 1 DSMR instantie?
- Aangenomen dat bovenstaande niet mogelijk is, dan zou ik 2 instanties van DSMR zou moeten draaien. Is dit mogelijk op 1 RPi, of moet ik naar docker om dit voor elkaar te krijgen? 
- Is het wel mogelijk om 2 dataloggers naast elkaar op 1 RPi te draaien, of ook middels docker? Dan zou ik bijvoorbeeld DSMR-Reader op de Synology kunnen draaien?
- Last resort is uiteraard 2 RPi's geheel separaat gebruiken, maar dat is minder leuk....

Ik hoop dat mijn vraag helder is en je me op weg kunt helpen. Docker vind ik nog erg lastig, maar daar kan ik mijn collega's mee lastig vallen.
Al vast veel dank,
Met vriendelijke groet, 
Lucien

dennissiemensma, 2019-12-28:
- Het is inderdaad niet mogelijk om twee meters met dezelfde installatie uit te lezen.
- Je zult dan inderdaad twee instanties moeten draaien.
- In principe zou het mogelijk moeten zijn om dat inderdaad via Docker te doen. Echter weet ik zelf niet zo 1-2-3 of ze het allemaal ondersteunen en wat je exact waar moet instellen voor Docker. Je kunt dan het beste even een issue aanmaken bij [een van de ontwikkelaars voor de Docker-containers](https://dsmr-reader.readthedocs.io/en/latest/installation/docker.html) (bij degene die je gebruikt) om te vragen wat mogelijk is.

dennissiemensma, 2019-12-31:
Mocht je nog meer vragen hebben, reageer gerust. Voor nu sluit ik deze.

closed 2019-12-31 by dennissiemensma