dsmrreader/dsmr-reader #815: "Is the server running on host "localhost" (::1)"

raoulteeuwen, 2019-12-29:
Gaat er iets fout, maar weet je niet wat? [Bekijk de logfiles](https://dsmr-reader.readthedocs.io/nl/v2/troubleshooting.html) als startpunt.

**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: laatste versie
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `RaspberryPi 4 4gb
* DSMR-protocol slimme meter *(v4 of v5)*: weet ik niet zeker

Ik heb een RPi4 gekocht. Wat er op de RPi3 liep wil ik dus werkend krijgen op de RPi4. Dat is Home Assistant en DSMR. Ik heb eea 1st opgebouwd op de RPi4: dat leek te werken. Ik was aangekomen op moment dat ik de RPi3 wilde vervangen door de RPi4 als 'hoofd machine'. Ik wilde dus de DSMR db overzetten. Ik heb DSMR en de db uit gezet cf instructies op https://dsmr-reader.readthedocs.io/en/latest/faq/database.html (alleen verse 11 van de db in de foldernamen).

Ik moest wat zoeken hoe ik de files van de RPi3 naar de RPi4 zou krijgen. Heb SCP gebruikt. Maar toen ik de files wilde kopieeren liep ik tegen permission denieds aan. Ik heb daarop op beide Pi's de Main-folder en alle files via chmod 777 gegeven, en daarmee lukte het me de files over te zetten.

Toen ik echter de Rpi3 afkoppelde en de Rpi4 boot-e, kwam HA wel op, maar DSMR niet. Ik krijg een melding

Is the server running on host "localhost" (::1) and accepting
        TCP/IP connections on port 5432?

Heb je enig idee hoe ik dit werkend kan krijgen? Ik zou hopen dat het een kleinigheid is. (maar als het teveel tijd kost, dan begin ik desnoods weer van voren af aan, evt zonder overzetten van de db)

raoulteeuwen, 2019-12-29:
Ben wat verder. Vond in postgresql-11-main.log in /var/log/postgresql:

2019-12-29 16:40:42.794 GMT [560] LOG:  shutting down
2019-12-29 16:40:42.881 GMT [556] LOG:  database system is shut down
2019-12-29 17:17:18.050 GMT [563] FATAL:  data directory "/var/lib/postgresql/11/main" has invalid permissions
2019-12-29 17:17:18.050 GMT [563] DETAIL:  Permissions should be u=rwx (0700) or u=rwx,g=rx (0750).
pg_ctl: could not start server
Examine the log output.

Na een "sudo chmod -R 750 main" zie ik in dat log

2019-12-29 20:16:51.548 GMT [561] LOG:  listening on IPv6 address "::1", port 5432
2019-12-29 20:16:51.550 GMT [561] LOG:  listening on IPv4 address "127.0.0.1", port 5432
2019-12-29 20:16:51.561 GMT [561] LOG:  listening on Unix socket "/var/run/postgresql/.s.PGSQL.5432"
2019-12-29 20:16:51.624 GMT [561] LOG:  could not open directory "base/pgsql_tmp": Permission denied
2019-12-29 20:16:51.632 GMT [561] LOG:  could not open directory "base/17473": Permission denied
2019-12-29 20:16:51.668 GMT [564] LOG:  database system was shut down at 2019-12-29 16:40:41 GMT
2019-12-29 20:16:51.670 GMT [564] PANIC:  could not open file "pg_wal/0000000100000002000000D6": Permission denied
2019-12-29 20:16:51.672 GMT [561] LOG:  startup process (PID 564) was terminated by signal 6: Aborted
2019-12-29 20:16:51.672 GMT [561] LOG:  aborting startup due to startup process failure
2019-12-29 20:16:51.677 GMT [561] LOG:  database system is shut down
pg_ctl: could not start server

Iets met rechten dus... Ik zoek verder.

raoulteeuwen, 2019-12-29:
Na een

sudo chown -R postgres:postgres 

lijkt het te werken ...

closed 2019-12-29 by raoulteeuwen

dennissiemensma, 2019-12-30:
Fijn dat je het zelf hebt kunnen oplossen!