dsmrreader/dsmr-reader #816: totalen voor bepaalde periode.

style2k, 2019-12-29:
hoi ik kan nu in dsmr dus zien kosten per maand per jaar en per dag.
nu heb ik bijv een contract voor de periode mei tot mei is het mogelijk om er een optie bij te hebben dat ik kan zeggen ik wil het totale verbruik en kosten  zien van mei tot mei zodat ik nu niet de maanden moet gaan optellen . maar dat ik dus in 1 keer zie wat het totale gebruik is voor mijn contract periode.?

en is het mogelijk om een knop in de interface toe te voegen waar ik een update kan uitvoeren van dsmr als er een update is ipv dat ik in de command line moet gaan ?

dennissiemensma, 2019-12-30:
Bedankt voor je verzoek. Wat betreft inzicht per contractduur, daarvoor staat al issue #617. Ik denk dat je daar tzt afdoende mee geholpen bent.

dennissiemensma, 2019-12-30:
Je andere vraag: Het updaten via de webinterface is niet mogelijk. Het kan namelijk zijn dat grote updates (of trage hardware) ervoor zorgt dat het langer duurt dan 60 seconden en dan krijgt de webinterface een timeout. Daarnaast moeten er processen herstart worden, waaronder de webinterface. Tot slot komen er regelmatig issues langs waarbij de exact output van de console nodig is, dus ook dat maakt het wat lastig debuggen.
Om dit alles robuust, werkend, veilig en betrouwbaar te maken, zijn een hoop workarounds nodig en ik denk persoonlijk niet dat dat opweegt tegen af en toe inloggen en een update command te draaien.

closed 2019-12-30 by dennissiemensma