dsmrreader/dsmr-reader #817: Grafieken niet zichtbaar in dashboard

Wouter1702, 2019-12-31:
Hallo Dennis,

Ik heb sinds de laatste update van DSMR het probleem dat de grafieken niet meer zichtbaar zijn in het dashboard Ik zie wel dat de metingen binnen komen en ook dat er metingen in de database zitten echter deze worden niet meer weergegeven in de grafieken.
Ik heb de update uitgevoerd met de cli commando en het deploy script.

Ik draaide versie 2.11.x en ik ben recent naar 2.12.1 gegaan. Sinds de update heb ik dit probleem.
Ik heb het zowel in Safari als in Chrome op twee verschillende OSX versies.

[image]

jrosier, 2019-12-31:
Ik had dit probleem ook, probeer ./deploy.sh opnieuw uit te voeren. Bij mij stond het filesysteem nog op readonly bij het uitvoeren va deploy.sh

Wouter1702, 2019-12-31:
Beste jrosier,

Bedankt voor de tip maar helaas zonder succes.
Ik heb zelfs eerst nog een fsck geforceerd, gereboot en vervolgens de deploy.sh gerunt (als dsmr user) maar het resultaat is nog hetzelfde :(

dennissiemensma, 2019-12-31:
@Wouter1702 ik vermoed dat je browser een foutmelding geeft. Kun je eens kijken in de webconsole van je browser (meestal F12 oid) en het dashboard refreshen?

Wouter1702, 2020-01-01:
Beste Dennis,

Aller eerst, de beste wensen voor het komend jaar.
Ik heb in chrome het webconsole gevonden en ik heb van het totaal een screenshot gemaakt.
[image]

Het zier er inderdaad naar uit dat er iets fout gaat in de uitvoering van de code. 

Wouter Iliohan

dennissiemensma, 2020-01-01:
Bedankt voor je terugkoppeling. Ik kijk er morgen inhoudelijk naar, want het lijkt een specifiek probleem bij je installatie of browser. Ik zie niet zo 1-2-3 de oorzaak.

De oudere Python-versie is niet de oorzaak van dit probleem, maar het is altijd goed om te upgraden. Nadat je de nieuwere Python-versie hebt geinstalleerd, zul je nog de virtualenv opnieuw moeten aanmaken met 3.7. Want die maak je bij de installatie eenmalig aan en is toen aangemaakt met 3.5. Zie hier voor instructies: https://github.com/dennissiemensma/dsmr-reader/issues/629#issuecomment-487041970

dennissiemensma, 2020-01-02:
Ik kan het helaas niet reproduceren.

- Krijg je ook fouten in het "Network" tabje naast de console?
- Wil je dit eens proberen:
```
sudo su - dsmr
./manage.py collectstatic --noinput
```

Wouter1702, 2020-01-03:
Hallo Dennis,

Nee in het netwerk tabje zie ik geen errors staan.

I weet nog wel dat ik voordat ik kon upgraden mijn filesystem in read-only mode stond. Ik heb toen een fsck moeten doen en hierbij is het nodige gerepareerd.
Vervolgens heb ik de upgrade gedaan en was vergeten te controleren of de applicatie het nog wel deed.

Ik maak als basis voor mijn VM’s gebruik van Debian 9 Jessie. Ik gebruik een turnkeylinux.org core appliance en daarin installeer ik telkens wat ik nodig heb voor een project.

Met vriendelijke groet,

Wouter Iliohan

dennissiemensma, 2020-01-03:
Heb je ook nog mijn tweede suggestie hierboven geprobeerd? Ik denk niet dat dat het oplost, maar het kan geen kwaad.

Wouter1702, 2020-01-03:
Hallo dessin,

Ik heb het commando uitgevoerd en ik zie op het dashboard geen verschil

Dit is de output van het commando

(dsmrreader) dsmr@DSMRreader ~/dsmr-reader$ ./manage.py collectstatic --noinput

0 static files copied to '/var/www/dsmrreader/static', 537 unmodified.
(dsmrreader) dsmr@DSMRreader ~/dsmr-reader$ 

Ik heb ook geprobeerd om de static directory even te hernomen en heb hem met het commando ook nog opnieuw aangemaakt.
Helaas het resultaat is nog steeds hetzelfde.

zou het kunnen zijn dat er een probleem in de database zit?

Met vriendelijke groet,

Wouter Iliohan

jaccorens, 2020-01-03:
Hoi Dennis,

Hier bracht het bovenstaande commando de oplossing voor hetzelfde probleem, maar pas na het toepassen van een `sudo chown -R dsmr:dsmr /var/www/dsmrreader/`. 

**Before**
```
./manage.py collectstatic --noinput
Copying '/home/dsmr/dsmr-reader/dsmr_frontend/static/dsmr_frontend/js/jquery-3.4.1.min.js'
Traceback (most recent call last):
<knip>
PermissionError: [Errno 13] Permission denied: 
```

**After**
```
./manage.py collectstatic --noinput
Copying '/home/dsmr/dsmr-reader/dsmr_frontend/static/dsmr_frontend/js/jquery-3.4.1.min.js'
Copying '/home/dsmr/dsmr-reader/dsmr_frontend/static/dsmr_frontend/js/moment-with-locales-2.24.0.min.js'
enz...
```

De grafiekjes laden nu weer prima.

Wouter1702, 2020-01-03:
Hallo Jacco,

Bedankt voor het meedenken. Ik heb jou oplossing geprobeerd maar nog steeds hetzelfde resultaat.
Er zit duidelijk iets anders dwars.

Met vriendelijke groet,

Wouter Iliohan

> Op 3 jan. 2020, om 20:14 heeft Jacco Rens <notifications@github.com> het volgende geschreven:
> 
> sudo chown -R dsmr:dsmr /var/www/dsmrreader/.

Wouter1702, 2020-01-03:
Dennis,

Ik heb nog even mijn browser cache geleegd en nadat ik de site opnieuw heb geladen, verschijnen mijn grafieken weer.
De foutieve modules zaten dus in cache.

Toch denk ik dat het opnieuw aanmaken van de statics zeker geholpen heeft. Want voordat ik dit had gedaan had ik het probleem ook met andere browsers op andere computers.

Met vriendelijke groet,

Wouter Iliohan

> Op 3 jan. 2020, om 20:20 heeft Wouter1702 <wouter1702@gmail.com> het volgende geschreven:
> 
> Hallo Jacco,
> 
> Bedankt voor het meedenken. Ik heb jou oplossing geprobeerd maar nog steeds hetzelfde resultaat.
> Er zit duidelijk iets anders dwars.
> 
> Met vriendelijke groet,
> 
> Wouter Iliohan
> 
> 
>> Op 3 jan. 2020, om 20:14 heeft Jacco Rens <notifications@github.com <mailto:notifications@github.com>> het volgende geschreven:
>> 
>> sudo chown -R dsmr:dsmr /var/www/dsmrreader/.
>

dennissiemensma, 2020-01-04:
Fijn om te horen!

closed 2020-01-04 by dennissiemensma