dsmrreader/dsmr-reader #820: [APIV2] Meer waardes oversturen via API v2

rubenverhoef, 2020-01-03:
Ik zou graag naast de standaard parameters alle extra parameters van het DSMR telegram oversturen via de API V2.
Dit zijn de standaard parameters in de API v2:
https://dsmr-reader.readthedocs.io/nl/latest/api.html#id4

Ik zou graag alle parameters waar DSMR-Reader wat mee doet over willen sturen via de V2 API.
(zoals DSMR versie, `Number of power failures in any phases`, etc.)

Momenteel is dat alleen mogelijk om het volledige telegram bericht over te sturen (APIV1) en niet als losse parameters zoals APIV2

dennissiemensma, 2020-01-03:
Bedankt voor je verzoek. Dat lijkt me opzich een prima toevoeging. Het zal wel even duren voordat ik hier aan toekom vermoedelijk.

rubenverhoef, 2020-01-05:
Oke, ik zie het wel een keer verschijnen bedankt! :)

dennissiemensma, 2020-01-15:
Ik heb hier nu een opzet voor werkend. 

Alleen zit ik nog even met de API-documentatie. Deze is nog vrij ouderwets, dus ik wil die nog eerst omzetten naar een OpenAPI-variant, met evt Postman. dat scheelt een hoop voorbeelden en eveneens vertalen (ik houd het dan bij Engels).

Tot die tijd kun je denk ik ondertussen eerst upgraden naar DSMR-reader v3.0. Deze feature komt waarschijnlijk in v3.1.

dennissiemensma, 2020-01-15:
Gemerged naar release-branch, docs via #839.

closed 2020-01-15 by dennissiemensma

rubenverhoef, 2020-01-16:
Bedankt!

Ik zal er eerdaags naar kijken. Klopt het dat ik alles van de [`PATCH`](https://github.com/dennissiemensma/dsmr-reader/blob/90c0d4b00b146b92840eefce137195b532c8ce8f/docs/api.rst#patch---dataloggermeter-statistics) moet gaan implementeren?

dennissiemensma, 2020-01-16:
Alle parameters zijn optioneel bij de PATCH, zodat je zelf kunt bepalen wat je wel en niet wilt/kan bijwerken.

dennissiemensma, 2020-01-17:
De release staat voor morgen gepland. Er zit ook een Postman-collectie bij zodat je werkende voorbeelden hebt.