dsmrreader/dsmr-reader #821: is update goed gegaan?

newtech2015, 2020-01-03:
hallo,

vandaag, 1/2 jaar na plaatsen nieuwe sd-kaart, blijkt de sd-kaart weer corrupt te zijn (zucht)..
zijn er mensen die ervaring hebben, dat bepaalde kaartjes langer dan 1/2 jaar meegaan met dsmr reader?

na backup terug te hebben gezet op een nieuwe sd kaart, maar gelijk geupdate..
op t einde zag ik dit voorbij komen: 
 * Reloading process: dsmr_webinterface (Gunicorn)   [??] PID file does not exist
 * Reloading process: dsmr_backend                kill: (4020): No such process
   [OK]
 * Reloading process: dsmr_datalogger                [??] PID file does not exist
 * Reloading process: dsmr_mqtt                   kill: (4017): No such process
   [OK]

 --- Clearing cache...

normaal, staat hier, als ik me niet vergis, in t groen "ok"

wat betekend dit, moet ik me zorgen maken, kan ik hier nog iets mee?

nadat ik dsmr weer gestart had, met supervisorctl start all.. werkte het programma weer gewoon, zover ik dat kan zien dan

gr ron

dennissiemensma, 2020-01-03:
Je zou nog even handmatig een post-deploy kunnen doen om te kijken of het nu wel goed gaat:
```
sudo su - dsmr
./post-deploy.sh
```

dennissiemensma, 2020-01-03:
Wat betreft SD-kaartjes, heb je toevallig een DSMR v5 meter? Zo ja dan zou je nog de frequentie van metingen kunnen verlagen, maar voor de rest heb ik er geen oplossing voor. Als ik jaren geleden had geweten dat SD-kaartjes hierdoor zo sluiten, had ik dit nooit voor de Pi gemaakt.

Je kunt ook nog overwegen om het ergens anders (mee) te hosten of een HDD te gebruiken.

newtech2015, 2020-01-03:
hoi,
ja, resultaat, is nu "ok", bedankt!

newtech2015, 2020-01-03:
hoi,
ik heb  een dsmr v5 meter, hoe verlaag ik het aantal metingen?heb gezocht,... bedoel je backend sleep, of datlogger sleep, of nog iets anders?

gr ron

dennissiemensma, 2020-01-04:
Ja klopt, het gaat om de datalogger sleep. Er vanuitgaande dat je gewoon DSMR-reader inclusief ingebouwde datalogger gebruikt.

Zie deze specifieke setting en bovenaan die pagina voor instructies: https://dsmr-reader.readthedocs.io/en/latest/settings.html#dsmrreader-datalogger-sleep

dennissiemensma, 2020-01-04:
Ik zou die sleep overigens zo hoog mogelijk zetten. Als je hem op 9 a 10 seconden zet, heb je 9 a 10 x zo weinig metingen en komt de doorvoer ongeveer overeen met een DSMR-v4 meter.

```
DSMRREADER_DATALOGGER_SLEEP = 10
```

Het hangt er ook vanaf waar je zelf natuurlijk waarde aan hecht. Het zijn je eigen gegevens, maar als je alleen dag/uur-totalen wilt bijhouden, zijn 6 telegrammen per minuut meer dan zat.

closed 2020-01-04 by dennissiemensma

newtech2015, 2020-01-04:
hoi,
bedankt, zal dit proberen, wanneer ie weer volledig in de lucht komt.dsmr is na backup vannacht weer gecrasht.(met nw sd-kaart)pi-image terug gezet, en wat verder gaan kijken, kom er nu achter, dat hier waarschijnlijk al iets niet goed zit:krijg als ik dsmr wil stoppen:
root@raspberrypi:~# sudo su - dsmr
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ supervisorctl stop all
error: <class 'socket.error'>, [Errno 13] Permission denied: file: /usr/lib/python2.7/socket.py line: 228

al hele tijd mee bezig en aan t googlen, maar krijg dit niet verholpen.jij een idee?
gr ron

dennissiemensma, 2020-01-04:
Je moet dat uitvoeren als root of sudo user en niet als `dsmr`. Als je `logout` typt en dan `sudo` ervoor zet (`sudo supervisorctl stop all`) zou het wel moeten werken.

newtech2015, 2020-01-04:
hoi,
ah, ok, dat was een denk fout van mij dan, bedankt!
voor de duidelijkheid:  ik werk met maandelijkse backups van mijn hele pi-image, en los hiervan met dagelijkse backups van dsmr via droppox...
als ik nu het pi-image terugzet van laten we als voorbeeld gebruiken, bv oktober dsmr v2.6, kan ik dan hierna de dsmr-data backup hierop terug zetten, waarvan de data inmiddels gemaakt is vanuit dsmr v2.11, of.. moet ik eerst 2.6 updaten naar 2.11, daarna data backup terug zetten, en dan eventueel updaten naar de nieuwste dsmr??

m.a.w:  moet de versie van dsmr hetzelfde zijn, als de versie die je had toen je (laatste) backup van dsmr is gemaakt, of blijft de datbase gelijk?
ik heb momenteel een probleem de dsmr data er terug in te krijgen, krijg hele waslijst foutmelingen en uiteindelijk heb ik nog steeds alleen data tot oktober~
wat mij al opvalt, is dat ik t commando om de datbase te clearen, al een fout geeft.zie de output die ik krijg:
s
root@raspberrypi:~# supervisorctl stop all
dsmr_mqtt: stopped
dsmr_webinterface: stopped
dsmr_datalogger: stopped
dsmr_backend: stopped
root@raspberrypi:~# sudo sudo -u postgres createdb -O dsmrreader dsmrreader
could not change directory to "/root": Permission denied
createdb: database creation failed: ERROR:  database "dsmrreader" already exists
root@raspberrypi:~# zcat /home/dsmr/dsmr-reader/backups/dsmrreader-postgresql-backup-Thursday.sql.gz | sudo sudo -u postgres psql dsmrreader
could not change directory to "/root": Permission denied
SET
SET
SET
SET
SET
 set_config
------------

(1 row)

SET
SET
SET
SET
CREATE EXTENSION
COMMENT
SET
SET
ERROR:  relation "auth_group" already exists
ALTER TABLE
ERROR:  relation "auth_group_id_seq" already exists
ALTER TABLE
ALTER SEQUENCE
ERROR:  relation "auth_group_permissions" already exists
ALTER TABLE
ERROR:  relation "auth_group_permissions_id_seq" already exists
ALTER TABLE
ALTER SEQUENCE
ERROR:  relation "auth_permission" already exists
ALTER TABLE
ERROR:  relation "auth_permission_id_seq" already exists
ALTER TABLE
ALTER SEQUENCE
ERROR:  relation "auth_user" already exists
ALTER TABLE
ERROR:  relation "auth_user_groups" already exists
ALTER TABLE
ERROR:  relation "auth_user_groups_id_seq" already exists
ALTER TABLE
ALTER SEQUENCE
ERROR:  relation "auth_user_id_seq" already exists
ALTER TABLE
ALTER SEQUENCE
ERROR:  relation "auth_user_user_permissions" already exists
ALTER TABLE
ERROR:  relation "auth_user_user_permissions_id_seq" already exists
ALTER TABLE
ALTER SEQUENCE
ERROR:  relation "django_admin_log" already exists
ALTER TABLE
ERROR:  relation "django_admin_log_id_seq" already exists
[... 693 lines cut ...]
ERROR:  relation "dsmr_backend_scheduledprocess_active_5ef7244b" already exists
ERROR:  relation "dsmr_backend_scheduledprocess_module_55cb894b_like" already exists
ERROR:  relation "dsmr_backend_scheduledprocess_planned_234388f1" already exists
ERROR:  relation "dsmr_consumption_electrici_currently_delivered_0f28e53a" already exists
ERROR:  relation "dsmr_consumption_electrici_currently_returned_b7a910c8" already exists
ERROR:  relation "dsmr_consumption_electrici_phase_currently_delivered__0315221e" already exists
ERROR:  relation "dsmr_consumption_electrici_phase_currently_delivered__21a515a7" already exists
ERROR:  relation "dsmr_consumption_electrici_phase_currently_delivered__9998ce55" already exists
ERROR:  relation "dsmr_consumption_electrici_phase_voltage_l1_72e3c8f8" already exists
ERROR:  relation "dsmr_consumption_electrici_phase_voltage_l2_917726c7" already exists
ERROR:  relation "dsmr_consumption_electrici_phase_voltage_l3_1a9d7c37" already exists
ERROR:  relation "dsmr_datalogger_dsmrreading_processed_add8a852" already exists
ERROR:  relation "dsmr_datalogger_dsmrreading_timestamp_1f7020d1" already exists
ERROR:  relation "dsmr_stats_note_day_83d05681" already exists
ERROR:  constraint "auth_group_permissio_permission_id_84c5c92e_fk_auth_perm" for relation "auth_group_permissions" already exists
ERROR:  constraint "auth_group_permissions_group_id_b120cbf9_fk_auth_group_id" for relation "auth_group_permissions" already exists
ERROR:  constraint "auth_permission_content_type_id_2f476e4b_fk_django_co" for relation "auth_permission" already exists
ERROR:  constraint "auth_user_groups_group_id_97559544_fk_auth_group_id" for relation "auth_user_groups" already exists
ERROR:  constraint "auth_user_groups_user_id_6a12ed8b_fk_auth_user_id" for relation "auth_user_groups" already exists
ERROR:  constraint "auth_user_user_permi_permission_id_1fbb5f2c_fk_auth_perm" for relation "auth_user_user_permissions" already exists
ERROR:  constraint "auth_user_user_permissions_user_id_a95ead1b_fk_auth_user_id" for relation "auth_user_user_permissions" already exists
ERROR:  constraint "django_admin_log_content_type_id_c4bce8eb_fk_django_co" for relation "django_admin_log" already exists
ERROR:  constraint "django_admin_log_user_id_c564eba6_fk_auth_user_id" for relation "django_admin_log" already exists

gr ron

dennissiemensma, 2020-01-04:
Je zult eerst de oude database moeten verwijderen, zie ook de melding `createdb: database creation failed: ERROR:  database "dsmrreader" already exists`

Je kunt dat doen met dit command, na het stoppen van de applicatie, en voordat je createdb doet::
```
sudo sudo -u postgres dropdb dsmrreader
```

Je melding `could not change directory to "/root": Permission denied` kun je negeren.

Het is goed om te lezen welke andere foutmeldingen je krijgt, voordat je uberhaupt geforceerd verder gaat. In een andere situatie kan je dat namelijk bijten.

newtech2015, 2020-01-04:
hoi,
ben weer verder, heb backup nu kunnen terugzetten, kwam in oude mails beschrijving tegen, dat ik eerste de oude database moet verwijderen, dan opnieuw aanmaken, en dan backup terug zetten.
wat ik net wel nog graag zou willen weten, of de door dsmr gemaakte backups versie-afhankelijk zijn
gr ron

dennissiemensma, 2020-01-04:
Je hoeft niet dezelfde versie te draaien. 

Het mag dezelfde of nieuwere zijn. Zodra je namelijk daarna een `./post-deploy.sh` doet, kijkt DSMR-reader zelf wat er bijgewerkt moet worden.

Het is wel beter om de applicatie pas te starten zodra dat gedaan is, want anders kun je wel fouten krijgen.

newtech2015, 2020-01-04:
hoi,
heb nu dsmr weer draaien, echter gezien dit nu al de 2e keer is dat mn sd kaartje corrupt is geraakt wil ik hiet wel wat mee gaan doen. 

zou het een manier zijn, om de database map, sudo sudo -u postgres createdb -O dsmrreader dsmrreadervia een mount point, op bv een harde schijf te gaan schrijven?maar belangrijkste:  ben ik dan af, van het extreem vaak schijven naar de sd-kaart, daar ik begrepen heb, dat "de database" niet zomaar 1 db file is, maar een hele verzameling van files is.
kun je mij een beetje de weg wijzen, hoe een methode kan zijn om dit te doen, en niet vervolgens bij een update erachter te komen dat t niet meer wekt, omdat de mappen niet meer standaard zijn?
ik werk al wat jaren met pi's, echter ben geen linux expert, heb nog redelijk basis kennis hiervan

gr ron

dennissiemensma, 2020-01-04:
In de meeste databases kun je sowieso vrij eenvoudig de datamap verplaatsen naar een andere locatie. Voor PostgreSQL staat hier een guide: https://dsmr-reader.readthedocs.io/en/latest/faq/database.html

newtech2015, 2020-01-08:
dennis,
ik ben nog op zoek, naar een oplossing om de database bv op een externe hdd te zetten, helaas is de pi hierdoor een paar keer gecrasht, en dan is telkens dsmr stuk gelopen.
ik blijf er dan telkens tegenaan lopen, dat de manier voorafgaand aan het terugzetten van de database, het verwijden en het opniew aanmaken, zonder inhoud, niet lukt.
wat doe ik hier toch niet goed??
wat ik dan probeer:# Als ROOT user
sudo supervisorctl stop all (stop DSMR reader)
  (stop de database)
sudo sudo -u postgres dropdb dsmrreader (verwijderen database)
sudo sudo -u postgres createdb -O dsmrreader dsmrreader (aanamken database zonder inhout)
zcat /home/dsmr/dsmr-reader/backups/dsmrreader-postgresql-backup-Tuesday.sql.gz | sudo sudo -u postgres psql dsmrreader (vullen database met inhoud uit backup file)

de commando's voorafgaand aan het terug zetten van de DB geeft dan:

root@raspberrypi:~# sudo sudo -u postgres dropdb dsmrreader                     could not change directory to "/root": Permission denied
root@raspberrypi:~# sudo sudo -u postgres createdb -O dsmrreader dsmrreader     could not change directory to "/root": Permission denied

waarom het ik geen rechten tot de DB?
gr ron

dennissiemensma, 2020-01-08:
Zoals eerder aangegeven, kun je de specifieke foutmeldingen `could not change directory to "/root": Permission denied` negeren. Als dat alle output is, ga ik er vanuit dat het goed is gegaan.

Je kunt het ook direct proberen als `postgres` user:

```
sudo cp /home/dsmr/dsmr-reader/backups/dsmrreader-postgresql-backup-Tuesday.sql.gz /var/lib/postgres/

sudo su - postgres
dropdb dsmrreader 
createdb -O dsmrreader dsmrreader
zcat dsmrreader-postgresql-backup-Tuesday.sql.gz | psql dsmrreader

psql --list
# De database zou er gewoon bij moeten staan dan.
```

newtech2015, 2020-01-09:
hoi,
vandaag heb ik de dsmr-database verplaatst naar een externe usb hdd
dit is gelukt, wel merk ik nu dat ik met regelmaat na een herstart van de pi, dsmr niet werkt, ik moet dan inloggen met putty en de databse nogmaals starten met:sudo systemctl start postgresql
dan werkt het weer.  
is dit een bekend probleem als je dsmr-database op een externe schijf zet?
2e probleem dat ik nu ondervindt, is dat ik nu niet meer dsmr kan updaten:
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ git checkout tags/v2.14.1
error: pathspec 'tags/v2.14.1' did not match any file(s) known to git.
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ git checkout tags/v2.12.0
error: pathspec 'tags/v2.12.0' did not match any file(s) known to git.

hij gaat met ./post-deploy.sh wel alles controleren, maar zie de versie nummers niet meer langskomen.
volgens de manual hoef je het versienummer helemaal niet mee te geven, maar als ik dat niet deed kwam er geen nieuwe versie na updaten (bleef op de oude staan).
ik deed dus altijd eerst:./post-deploy.shdan bv:git checkout tags/v2.12.0
en dan weer:./post-deploy.sh
dan ging ie de geselecteerde versie binnenhalen
vreemd, want, eigenlijk is alleen de database verplaatst en verder niets veranderd aan dsmr.
g ron

jeroenpeters1986, 2020-01-09:
Wat betreft de database; als software/gegevens zoals de database op een externe locatie staat kan het altijd even duren voordat iets beschikbaar is. 
Dat is niet specifiek zo voor DSMR-reader maar eigenlijk inherent aan deze constructie. Dus dit is 'normaal'. Je zou dan iets moeten verzinnen dat eerst checkt of de data er is (dus de schijf goed aangekoppeld) en dat dan pas postgres wordt opgestart.
Zelf heb ik de database in de cloud staan, dat gaat wel goed (tenzij internet eruit ligt natuurlijk).

dennissiemensma, 2020-01-09:
@newtech2015 je kunt het beste even checken of je wel op de juiste branch zit. Het beste is om `master` te gebruiken. Dan kun je gewoon updaten telkens.

```
sudo su - dsmr
git fetch
git checkout master
./deploy.sh
```

newtech2015, 2020-01-10:
hoi jeroen,

dat is eigenlijk een nog mooiere optie, database in de cloud.kun je mij een beschrijving geven, hoe je dat gedaan hebt?
gr ron

jeroenpeters1986, 2020-01-10:
Ja dat kan ik zeker, maar dat is voor de meeste mensen een no-go, omdat het via MySQL werkt. Daar is in v3 ook geen ondersteuning voor, dus als je ooit wilt upgraden zou ik mijn werkwijze zeker niet aanraden.
Ik heb bij een goedkope hostingpartij een pakket afgenomen met een database, en mijn DSMR-reader daar toegang toe gegeven. Je krijgt vrijwel overal MySQL, dus daar hoef je dan niets voor te installeren. Ook worden er (bij deze partij) automatisch backups gemaakt.

Een alternatief is om zelf een VPS ergens af te nemen, waarop je Postgres installeert en externe toegang openzet voor je DSMR-reader. Dan kun je ook mee naar v3, maar realiseer je dan wel dat je zelf verantwoordelijk bent voor de backups.

newtech2015, 2020-01-18:
Hoi,
al enige tijd, worden los van een totale database backup, ook partitial backups gemaakt.
in de beschrijving, vind ik alleen terug hoe je een complete backup moet terug zetten.
wil ik een partitial terugzetten, krijg ik foutmeldingen bv:
 zcat /home/dsmr/dsmr-reader/backups/dsmrreader-postgresql-partial-backup-2020-01-03.sql.gz | sudo -u postgres psql dsmrreader
could not change directory to "/root": Permission denied
psql: could not connect to server: No such file or directory
        Is the server running locally and accepting
        connections on Unix domain socket "/var/run/postgresql/.s.PGSQL.5432"?
hoe moet ik dit doen?
gr ron

dennissiemensma, 2020-01-18:
@newtech2015 de foutmelding `psql: could not connect to server: No such file or directory` geeft aan dat de databaseserver niet draait of niet bereikbaar is.

Heb je gewoon PostgreSQL geinstalleerd?

Wat betreft het restoren van een partial, ik heb dat nog niet gedocumenteerd omdat het meer een laatste redmiddel is. Heb je geen volledige backup meer?

newtech2015, 2020-01-18:
Jawel, maar mis paar dagen in de database, doordat kaartje stuk gegaan was. Dacht die dagen er even in te zetten. Maar, dat is dus niet zo simpel, begrijp ik. 
Groetjes ron 
Verzonden via Yahoo Mail op Android

dennissiemensma, 2020-01-18:
Je kunt geen data restoren zonder een lege database. Als je alleen de informatie wilt hebben van een paar dagen, kun je denk ik het beste de partial backup uitpakken en bekijken met een teksteditor.

Vervolgens kun je handmatig in de admin-interface de missende dagstatistieken aanmaken, met de gegevens uit de backup.