dsmrreader/dsmr-reader #826: dsmr_webinterface gunicorn issue bij opstarten

swerveshot, 2020-01-06:
### Wat gebruik je?
* DSMR-reader version: `v2.13b3`
* Hardware: `Odroid U3`
* DSMR protocol: `v4`

### Omschrijving
Sinds een upgrade naar een van de releases van de afgelopen maand is bij mij de dsmr_webinterface stuk gegaan. Deze meld nu bij bij opstarten:
```
[2020-01-06 16:31:58 +0100] [11423] [INFO] Starting gunicorn 19.7.0
[2020-01-06 16:31:58 +0100] [11423] [ERROR] Retrying in 1 second.
[2020-01-06 16:31:59 +0100] [11423] [ERROR] Retrying in 1 second.
[2020-01-06 16:32:00 +0100] [11423] [ERROR] Retrying in 1 second.
[2020-01-06 16:32:01 +0100] [11423] [ERROR] Retrying in 1 second.
[2020-01-06 16:32:02 +0100] [11423] [ERROR] Retrying in 1 second.
[2020-01-06 16:32:03 +0100] [11423] [ERROR] Can't connect to /var/tmp/gunicorn--dsmr_webinterface.socket
```

Deze melding krijg ik zowel bij gebruik van de supervisor als via de volgende commando's
```
sudo su - dsmr
/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn dsmrreader.wsgi
```

Wat dan weer wél werkt is
```
sudo su - dsmr
./manage.py runserver
```

De log files `/var/log/supervisor/dsmr_webinterface.log` en `/var/log/syslog` leveren geen nuttige informatie op. En ook andere log files zoals syslog of auth tonen niets relevants.

Tijdens issue #818 al enige troubleshooting gedaan, maar dit leverde niet echt iets op. En jammer genoeg ben ik sinds dien niet veel verder gekomen. Dus sorry @dennissiemensma, maar zou je me alsjeblieft iets verder willen helpen?

dennissiemensma, 2020-01-06:
Je kunt proberen om even een andere checkout + virtualenv aan te maken, dependencies opnieuw installeren, en dan weet Gunicorn proberen te draaien. Maar dan in de nieuwe, tijdelijke omgeving.

```
sudo su - dsmr
deactivate

git clone https://github.com/dennissiemensma/dsmr-reader.git /home/dsmr/temp-dsmr-reader
cd ~/temp-dsmr-reader
virtualenv /home/dsmr/.virtualenvs/temp-dsmrreader --no-site-packages --python python3
source ~/.virtualenvs/temp-dsmrreader/bin/activate

cp dsmrreader/provisioning/django/postgresql.py dsmrreader/settings.py
pip3 install -r dsmrreader/provisioning/requirements/base.txt -r dsmrreader/provisioning/requirements/postgresql.txt

gunicorn dsmrreader.wsgi
```

Als je na afloop de kopie wilt verwijderen:
```
deactivate
cd ~
rm -rf ~/.virtualenvs/temp-dsmrreader ~/temp-dsmr-reader
```

swerveshot, 2020-01-07:
🤯 Nooooooo!! Ook dit werkt niet.

```
(temp-dsmrreader) dsmr@odroid:~/temp-dsmr-reader$ gunicorn dsmrreader.wsgi
[2020-01-07 21:44:47 +0100] [21339] [INFO] Starting gunicorn 19.7.0
[2020-01-07 21:44:47 +0100] [21339] [ERROR] Retrying in 1 second.
[2020-01-07 21:44:48 +0100] [21339] [ERROR] Retrying in 1 second.
[2020-01-07 21:44:49 +0100] [21339] [ERROR] Retrying in 1 second.
[2020-01-07 21:44:50 +0100] [21339] [ERROR] Retrying in 1 second.
[2020-01-07 21:44:51 +0100] [21339] [ERROR] Retrying in 1 second.
[2020-01-07 21:44:52 +0100] [21339] [ERROR] Can't connect to ('127.0.0.1', 8000)
```

Wat ik me nog afvroeg is of het nog uitmaakt hoe mijn NGINX config eruit ziet. Die is een klein beetje getweaked om een subpath mogelijk te maken. Zo kan ik meer dan alleen DSRM Reader hosten door `http://hostname/dsmrreader` te gebruiken. Then again, dit heeft altijd gewoon gewerkt.

Still, voor de volledigheid post ik toch maar de configuratie die ik in `/etc/nginx/sites-available/dsmr-webinterface` heb staan.

```
upstream dsmr-webinterface {
    server unix:///var/tmp/gunicorn--dsmr_webinterface.socket fail_timeout=0;
}

server {
    listen      80;
    server_name _;  # No hostname available? Just use the underscore for wildcar                                                                                                             d matching. :]

##      # Uncomment these lines to force additional authentication on applicatio                                                                                                             n access.
##  # Do not forget to generate a password first in '/etc/nginx/htpasswd' and th                                                                                                             en RELOAD nginx!
##    auth_basic "Restricted application";
##    auth_basic_user_file /etc/nginx/htpasswd;

    location /dsmrreader/static {
        alias /var/www/dsmrreader/static;
    }

    location /dsmrreader/ {
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header Host $http_host;
        proxy_redirect off;
        proxy_pass http://dsmr-webinterface;
    }
}
```

Ik heb ook nog een revert gedaan naar de default config zonder subpath, maar dit gaf helaas hetzelfde resultaat.

swerveshot, 2020-01-07:
Dit is overigens ook niet het eerste issue dat ik met gunicorn heb; zie #418. Toen zat ik kennelijk op versie 19.7.1 terwijl ik nu vreemd genoeg op 19.7.0 zit. Weird huh? 

```
(dsmrreader) dsmr@odroid:~/dsmr-reader$ pip freeze | grep gunicorn
You are using pip version 9.0.1, however version 19.3.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
gunicorn==19.7.0
```

dennissiemensma, 2020-01-07:
Ik verwacht niet dat Nginx invloed heeft. Nginx praat alleen tegen de applicatieserver (Gunicorn), maar die wil dus uberhaupt niet starten.

Je kunt nog proberen om een nieuwere versie van Gunicorn te gebruiken:

```
sudo su - dsmr
pip3 install gunicorn --upgrade

gunicorn dsmrreader.wsgi
```

dennissiemensma, 2020-01-07:
En begrijp ik het goed dat de andere processen (datalogger, backend) wel zonder problemen draaien?

swerveshot, 2020-01-07:
Hey, dit gaat dus wél goed. En dat met de gunicorn die met alpine issues gaf.
```
(dsmrreader) dsmr@odroid:~/dsmr-reader$ gunicorn dsmrreader.wsgi
[2020-01-07 22:10:38 +0100] [23068] [INFO] Starting gunicorn 20.0.4
[2020-01-07 22:10:38 +0100] [23068] [INFO] Listening at: http://127.0.0.1:8000 (23068)
[2020-01-07 22:10:38 +0100] [23068] [INFO] Using worker: sync
[2020-01-07 22:10:38 +0100] [23072] [INFO] Booting worker with pid: 23072
```

swerveshot, 2020-01-07:
NGINX toont nu wel dit, dus daar moet ik nog even naar kijken.

```
Not Found

The requested resource was not found on this server.
```

Maar dit is vooruitgang. Vreesde dat dit een issue zou zijn waar het probleem tussen keyboard en chair zat.

dennissiemensma, 2020-01-07:
Ik zou dan lekker die Gunicorn-versie blijven gebruiken. In DSMR-reader v3 gebruik ik die versie namelijk ook. Gunicorn 19 was alleen nog maar in gebruik wegens #783. Het issue van destijds is in nieuwere minor versies van Gunicorn gefixt: https://github.com/benoitc/gunicorn/issues/2160

Je moet dan alleen wel na elke update van DSMR-reader even handmatig dat doen:
```
sudo su - dsmr
pip3 install gunicorn --upgrade
./reload.sh

# (Of herstarten via supervisor)
```

Maar waarschijnlijk hoef je dat nog maar 1 of 2 x te doen. DSMR-reader v2.14 is de laatste met toevoegingen voor v2. Er komt nu alleen nog v2.15, wat puur visueel de mededeling doet naar DSMR-reader v3.x straks.

dennissiemensma, 2020-01-07:
Je kunt nu trouwens die handmatige Gunicorn run gewoon afbreken en het draaien via Supervisor. Ik neem aan dat je dat al gedaan hebt nu?

En dan kijken in de errorlog van Nginx.

swerveshot, 2020-01-07:
Ja, supervisor doet het nu ook. En heb bij het troubleshooten waarschijnlijk een forward slash weggehaald op het eind van het pad in mijn proxy_pass parameter. Nu dat gefixed is krijg ik ook weer 'beeld'. Dus weer een issue dat gesloten kan worden.

Super bedankt voor het meedenken weer @dennissiemensma. 🙏

Off-topic; v3 wordt ook nog wel een dingetje. Want ik zit met mijn Odroid U3 vast aan Ubuntu 16.04 LTS en daarmee ook aan een antieke Python 3. Dus dat wordt waarschijnlijk een source install. Leuk!

swerveshot, 2020-01-07:
Oh nog 1 more thing. Enig idee hoe ik hier vanaf kom? Is dat een kwestie van tijd?
[image]

dennissiemensma, 2020-01-07:
In principe kun je gewoon DSMR-reader v2.x blijven gebruiken als Python upgraden te lastig is. Alleen krijgt v2 vanzelfsprekend geen updates, hoewel dat vooral een issue is wanneer je applicatie verbonden is aan het Internet. Als dat niet het geval is, maakt het (naar mijn mening voor zo'n projectje) niet uit en kun je heel lang v2 blijven draaien.

De source-install stelt trouwens weinig voor. Voor de v3 upgrade straks is dat namelijk ook de way to go. Zie hier voor de preview: https://dsmr-reader.readthedocs.io/en/v3/faq/v3_upgrade.html

dennissiemensma, 2020-01-07:
En dat hebt ik trouwens uitgevoerd op een `Ubuntu 16.04.6 LTS` server als test. Dus zit wel goed.

dennissiemensma, 2020-01-07:
Voor die melding over gas kun je het beste even in de debuglogs kijken: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging

Heb je een andere meter gekregen of iets?

closed 2020-01-07 by dennissiemensma

swerveshot, 2020-01-07:
> 
> 
> En dat hebt ik trouwens uitgevoerd op een `Ubuntu 16.04.6 LTS` server als test. Dus zit wel goed.

Old stable. Nice! 👊🏼

swerveshot, 2020-01-07:
> Voor die melding over gas kun je het beste even in de debuglogs kijken: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging
Zal ik doen.
> 
> Heb je een andere meter gekregen of iets?
Ja, andere woning. Dus andere meter.

dennissiemensma, 2020-01-07:
Ooh sorry, ik was alweer vergeten dat je #818 had gemaakt. Ik denk dat ik die statuspagina nog moet aanpassen, zodat die die melding niet toont als er geen gas is. Ik zal er een issue voor maken, maar het zal wel even duren voordat ik die oppakt. Niet eerder dan wanneer v3 klaar is.

swerveshot, 2020-01-07:
No worries. Besef me ook dat dit een beetje een edge-case is. Ben al blij dat het allemaal weer werkt joh.

dennissiemensma, 2020-01-07:
Laatste ding. Ik kan je trouwens aanraden om niet meer de `develop` branch te gebruiken. Op een gegeven moment switch ik die om naar `v3`, waardoor je bij een update mogelijk opeens een andere Python-versie nodig hebt.

Voor `master` ga ik dat niet doen. Die blijft altijd op `v2`, dus je kunt het beste `master` weer gebruiken. Mits je nu op v2.14 zit/wil.

swerveshot, 2020-01-09:
> De source-install stelt trouwens weinig voor. Voor de v3 upgrade straks is dat namelijk ook de way to go. Zie hier voor de preview: https://dsmr-reader.readthedocs.io/en/v3/faq/v3_upgrade.html

Zie op verschillende plekken in de instructie sudo voorkomen. Zou je niet liever een Python install onder de dsmr user willen doen zoals hier beschreven wordt? http://thelazylog.com/install-python-as-local-user-on-linux/

> Ik kan je trouwens aanraden om niet meer de develop branch te gebruiken.

Zal gaan switchen.

dennissiemensma, 2020-01-09:
Ik denk dat je het beste zelf kunt inschatten hoe je het liefste Python wilt installeren. Persoonlijk zie ik geen bezwaren om het systeembreed te doen, vooral omdat het een aparte binary wordt (dacht ik), maar de keuze is uiteraard aan jezelf.