dsmrreader/dsmr-reader #827: Mailbackup triggered elke halve minuut

vincedh, 2020-01-07:
**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.13`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `RaspberryPi 3`
* DSMR-protocol slimme meter *(v4 of v5)*: `v5`

Hoi Dennis,

Ten eerste, hartelijk dank voor je fijne applicatie. 
Het valt mij op dat sinds 14:40 vanmiddag (wanneer de DSMR-reader mail backup draait), er ongeveer elke halve minuut een nieuwe backup per mail gemaakt wordt:

> Laatst gedraaid: 8 seconden geleden 
> Volgende uitvoer: over 23 seconden

De mail zelf wordt niet verstuurd, want ik heb in m'n mailbox keurig 1 mail ontvangen van de DSMR-reader vandaag (om 14:40), maar het viel me op omdat ik in m'n DNS server erg vaak een request zie naar de mailserver vanaf m'n DSMR-pi.

Ik heb vanmorgen naar v2.13 ge-update, dus het lijkt wel in deze versie te zitten, hiervoor nergens last van.

Vriendelijke groet

dennissiemensma, 2020-01-07:
Bedankt voor je melding. Wellicht kun je even kort debug logging inschakelen, om te zien wat er exact gebeurt: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging

In de log van de backend kun je dan wellicht zien wat die doet.

vincedh, 2020-01-07:
Dank voor je snelle reactie, gelijk even gekeken en ik zie inderdaad wel wat misgaan, zie hieronder.

`[2020-01-07 18:33:04,054] INFO      - Creating new partial backup: /tmp/tmp5fxqxak1/dsmrreader-postgresql-partial-backup-2020-01-07.sql
[2020-01-07 18:33:04,962] DEBUG     - Backup exit code: 0
[2020-01-07 18:33:04,983] INFO      - Created and compressed statistics backup: /tmp/tmp5fxqxak1/dsmrreader-postgresql-partial-backup-2020-01-07.sql.gz
[2020-01-07 18:33:05,001] DEBUG    Email: Preparing to send email using mail server smtp.office365.com:587
[2020-01-07 18:33:05,003] DEBUG    Email backup: Sending an email to [PRIVATE] (DSMR-reader day/hour statistics backup)
[2020-01-07 18:33:06,465] ERROR    (554, b'5.2.0 STOREDRV.Submission.Exception:SendAsDeniedException.MapiExceptionSendAsDenied; Failed to process message due to a permanent exceptio
n with message Cannot submit message. 0.35250:1F004E32, 1.36674:7A000000, 1.61250:00000000, 1.45378:02000000, 1.44866:00140000, 1.36674:0A000000, 1.61250:00000000, 1.45378:13000000,
 1.44866:621E0000, 1.36674:0E000000, 1.61250:00000000, 1.45378:781E0000, 1.44866:EC010000, 16.55847:4B100000, 17.43559:000000000C020000000000000200000000000000, 20.52176:140F2A8A0A0
0101009000000, 20.50032:140F2A8A7A170000350A0000, 0.35180:0D000000, 255.23226:0A001D30, 255.27962:7A000000, 255.27962:0A000000, 255.27962:0E000000, 255.31418:1400883F, 0.35250:9E000
000, 1.36674:0A000000, 1.61250:00000000, 1.45378:02000000, 1.44866:5C000000, 1.36674:32000000, 1.61250:00000000, 1.45378:61000000, 1.44866:01000000, 16.55847:C8000000, 17.43559:0000
000000030000000000000C00000000000000, 20.52176:140F2A8A0A0000000300200E, 20.50032:140F2A8A7A1770200A004200, 0.35180:0A006C66, 255.23226:D9000000, 255.27962:0A000000, 255.27962:32000
000, 255.17082:DC040000, 0.27745:0A006500, 4.21921:DC040000, 255.27962:FA000000, 255.1494:0B00060C, 0.38698:0F010480, 0.37692:01000000, 0.37948:31303500, 5.33852:00000000534D5450004
33142, 7.36354:010000000000010900000000, 4.56248:DC040000, 7.40748:010000000000010B00000000, 7.57132:00000000000000000000322D, 1.63016:32000000, 4.39640:DC040000, 8.45434:162BF457DB
B3834B85ECB7F198AB0E7931313600, 5.10786:0000000031352E32302E323630322E3031303A444237503139334D42303236363A37386134626330662D346431312D346238322D393832652D346434383364376638646661005
D2D31313531, 7.51330:BE19ECA79793D70800000000, 255.1750:0300080E, 255.31418:03000B37, 0.22753:4E010000, 255.21817:DC040000, 4.60547:DC040000, 0.21966:56010000, 4.30158:DC040000 [Hos
tname=DB7P193MB0266.EURP193.PROD.OUTLOOK.COM]')
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_backend/services/schedule.py", line 30, in execute_scheduled_processes
    current.execute()
  File "/home/dsmr/dsmr-reader/dsmr_backend/models/schedule.py", line 50, in execute
    return service(self)
  File "/home/dsmr/dsmr-reader/dsmr_backup/services/email.py", line 40, in run
    attachment=backup_file
  File "/home/dsmr/dsmr-reader/dsmr_backend/services/email.py", line 43, in send
    message.send()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/mail/message.py", line 291, in send
    return self.get_connection(fail_silently).send_messages([self])
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/mail/backends/smtp.py", line 110, in send_messages
    sent = self._send(message)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/mail/backends/smtp.py", line 126, in _send
    self.connection.sendmail(from_email, recipients, message.as_bytes(linesep='\r\n'))
  File "/usr/lib/python3.5/smtplib.py", line 883, in sendmail
    raise SMTPDataError(code, resp)
smtplib.SMTPDataError: (554, b'5.2.0 STOREDRV.Submission.Exception:SendAsDeniedException.MapiExceptionSendAsDenied; Failed to process message due to a permanent exception with messa
ge Cannot submit message. 0.35250:1F004E32, 1.36674:7A000000, 1.61250:00000000, 1.45378:02000000, 1.44866:00140000, 1.36674:0A000000, 1.61250:00000000, 1.45378:13000000, 1.44866:621
E0000, 1.36674:0E000000, 1.61250:00000000, 1.45378:781E0000, 1.44866:EC010000, 16.55847:4B100000, 17.43559:000000000C020000000000000200000000000000, 20.52176:140F2A8A0A0010100900000
0, 20.50032:140F2A8A7A170000350A0000, 0.35180:0D000000, 255.23226:0A001D30, 255.27962:7A000000, 255.27962:0A000000, 255.27962:0E000000, 255.31418:1400883F, 0.35250:9E000000, 1.36674
:0A000000, 1.61250:00000000, 1.45378:02000000, 1.44866:5C000000, 1.36674:32000000, 1.61250:00000000, 1.45378:61000000, 1.44866:01000000, 16.55847:C8000000, 17.43559:0000000000030000000000000C00000000000000, 20.52176:140F2A8A0A0000000300200E, 20.50032:140F2A8A7A1770200A004200, 0.35180:0A006C66, 255.23226:D9000000, 255.27962:0A000000, 255.27962:32000000, 255.17082:DC040000, 0.27745:0A006500, 4.21921:DC040000, 255.27962:FA000000, 255.1494:0B00060C, 0.38698:0F010480, 0.37692:01000000, 0.37948:31303500, 5.33852:00000000534D545000433142, 7.36354:010000000000010900000000, 4.56248:DC040000, 7.40748:010000000000010B00000000, 7.57132:00000000000000000000322D, 1.63016:32000000, 4.39640:DC040000, 8.45434:162BF457DBB3834B85ECB7F198AB0E7931313600, 5.10786:0000000031352E32302E323630322E3031303A444237503139334D42303236363A37386134626330662D346431312D346238322D393832652D346434383364376638646661005D2D31313531, 7.51330:BE19ECA79793D70800000000, 255.1750:0300080E, 255.31418:03000B37, 0.22753:4E010000, 255.21817:DC040000, 4.60547:DC040000, 0.21966:56010000, 4.30158:DC040000 [Hostname=DB7P193MB0266.EURP193.PROD.OUTLOOK.COM]')
[2020-01-07 18:33:06,476] DEBUG    SP: Rescheduled "Backup per email" to 2020-01-07 18:33:36.467816+01:00 (ETA 0:00:29.991088)
`

dennissiemensma, 2020-01-07:
De melding zelf zegt me niet zo heel veel.

Het enige wat in de buurt komt is:
```
5.2.0 STOREDRV.Submission.Exception:SendAsDeniedException.MapiExceptionSendAsDenied; Failed to process message due to a permanent exception with message Cannot submit message.
``` 

Maar ik weet niet of dat een oorzaak of gevolg is. Wellicht is het niet toegestaan om via Office zomaar mail te versturen.

vincedh, 2020-01-07:
Ik zie nu dat er een veld bijgekomen is in de webinterface van DSMR-reader, 'Email From' 
Deze was leeg, hier heb ik m'n eigen e-mailadres ingevuld en nu werkt het weer, volgende backup over 23uur. 

Hartelijk dank voor het meedenken!

closed 2020-01-07 by vincedh

dennissiemensma, 2020-01-07:
Dat is goed om te horen. Bedankt voor het delen, want daar hebben we geen rekening mee gehouden. Ik zal dit ticket even pinnen.

Vergeet niet om debug logging weer uit te zetten.

dennissiemensma, 2020-01-07:
Ik zal er nog even een fix voor maken, want dit mag niet gebeuren.

dennissiemensma, 2020-01-07:
Gefixt in `v2.14.1`