dsmrreader/dsmr-reader #830: Vaste kosten voor een totaaloverzicht.

quadrophenianl, 2020-01-10:
Goedemorgen Dennis,

Is het wellicht (al) mogelijk om ergens de vaste maandelijkse kosten toe te voegen om zo een totaaloverzicht te krijgen van de totaal aan kosten? Hiermee doel ik op de leveringskosten en netbeheerderskosten.

Uiteraard snap ik dat dit in eerste instantie bedoeld is om het energieverbruik te monitoren, maar voor mij zou dit een hele mooie toevoeging zijn. Zeker als dit nog gekoppeld kan worden aan het voorschotbedrag.

Ik hoor het wel!

Groeten Mark

quadrophenianl, 2020-01-10:
Niet heel goed gezocht, want zag hier al een topic: https://github.com/dennissiemensma/dsmr-reader/issues/134 

Mocht hier nog nieuws over zijn hoor ik het graag!

dennissiemensma, 2020-01-10:
Ik denk dat het meer iets is voor #830. Die kun je dan het beste in de gaten houden.

closed 2020-01-10 by dennissiemensma