dsmrreader/dsmr-reader #831: Nieuwe installatie na stroomstoring - geen nieuwe metingen / toegang tot status pagina

raymond79nl, 2020-01-10:
**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2..14.1`
* Hardware: `RaspberryPi 3 Model B+`
* DSMR-protocol slimme meter *(v4 of v5)*: `v5`

Na stroomstoring volledig verse install gedaan (alle stappen gevolgd van [deze pagina](https://dsmr-reader.readthedocs.io/nl/latest/installation/quick.html)). Daarna meest recente backup teruggezet. Dat werkte alleen goed wanneer ik begon met een volledig lege database, dus na het volgen van volledige installatie de volgende commanda's uitgevoerd:
```
sudo supervisorctl stop all
sudo sudo -u postgres dropdb dsmrreader
sudo sudo -u postgres createdb -O dsmrreader dsmrreader
zcat /home/dsmr/dsmr-reader/backups/backupnaam.sql.gz | sudo sudo -u postgres psql dsmrreader
sudo su - dsmr
./deploy.sh
```
Als ik dan de Pi reboot kom ik gewoon in de web UI en zie ik ook al mijn oude data in archief (alle tabs (van bovenaf dashboard t/m energiecontracten werken). Ik kom echter niet in de status pagina (504 Gateway Time-out) en er worden slechts sporadisch nieuwe metingen gedaan.

Welke logfile kan het beste inzicht bieden in wat er hier misgaat?

dennissiemensma, 2020-01-10:
Je kunt het beste in alle drie de logsfiles kijken en eventueel logging op DEBUG zetten, mocht je niets kunnen vinden: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging

raymond79nl, 2020-01-10:
@dennissiemensma Thx voor je reactie!

Webinterface log geeft een critical worker timeout bij opvragen status pagina (hieronder 2x geprobeerd, om 19:36 en 19:44):
```
[2020-01-10 19:34:36 +0000] [6830] [INFO] Starting gunicorn 19.7.0
[2020-01-10 19:34:36 +0000] [6830] [INFO] Listening at: unix:/var/tmp/gunicorn--dsmr_webinterface.socket (6830)
[2020-01-10 19:34:36 +0000] [6830] [INFO] Using worker: sync
[2020-01-10 19:34:36 +0000] [6838] [INFO] Booting worker with pid: 6838
[2020-01-10 19:36:31 +0000] [6830] [CRITICAL] WORKER TIMEOUT (pid:6838)
[2020-01-10 19:36:32 +0000] [6874] [INFO] Booting worker with pid: 6874
[2020-01-10 19:44:43 +0000] [6830] [CRITICAL] WORKER TIMEOUT (pid:6874)
[2020-01-10 19:44:44 +0000] [6982] [INFO] Booting worker with pid: 6982
```
De backend log gaf melding van schrijfrechten op backup bestanden (die ik inderdaad vanuit mijn Linux install had gekopieerd naar de SD). Na verwijdering van de oude bestanden is daar niks geks meer te zien. Begint na boot meteen met opzetten nieuwe backup. Ook in andere logs niks geks te zien.

dennissiemensma, 2020-01-10:
Heb je toevallig heel veel telegrammen opgeslagen? Ook gezien je een v5-meter hebt. 

Het kan zijn dat er teveel telegrammen zijn voor de status-pagina. Ik moet dat nog een keertje op een andere manier maken.

----

Je kunt ofwel retentie inschakelen zodat telegrammen opgeschoond worden ofwel het laten voor wat het is. In de backend log kun je met DEBUG-level logging zien of de applicatie nog wel door blijft lopen. Zie `/var/log/supervisor/dsmr_backend.log`. 
Het kan zijn dat je een sleep van 2 a 3 seconden wilt aanzetten in de datalogger. Dat kan tegenwoordig via de admin-interface bij datalogger-configuratie. Je zult alsnog wel handmatig de processen moeten herstarten.

----

Een ander ding wat je kunt doen is kijken of de indexen op de telegram-tabel goed aangemaakt zijn. Er zijn gevallen bekend waarbij dat om een of andere reden niet goed is gegaan en daarmee de applicatie traag maakt.

```
sudo su - postgres
psql dsmrreader 
\dS dsmr_datalogger_dsmrreading;
```

Deze zouden er moeten zijn, onderaan de output:
```
Indexes:
    "dsmr_datalogger_dsmrreading_pkey" PRIMARY KEY, btree (id)
    "dsmr_datalogger_dsmrreading_9d090a40" btree (processed)
    "dsmr_datalogger_dsmrreading_timestamp_1f7020d1_uniq" btree ("timestamp")
```

raymond79nl, 2020-01-11:
Thx! Ik heb een database van grofweg 200 MB na 5 maanden. Heb nu de datalogger sleep op 2.5 seconden gezet om te testen, maar statuspagina opent nog steeds niet. De indexen lijken er te zijn, wel met iets andere naamgeving:
```
Indexes:
    "dsmr_datalogger_dsmrreading_pkey" PRIMARY KEY, btree (id)
    "dsmr_datalogger_dsmrreading_processed_add8a852" btree (processed)
    "dsmr_datalogger_dsmrreading_timestamp_1f7020d1" btree ("timestamp")
```
Ik moet trouwens toegeven dat ik uit mijn hoofd v5 heb opgeschreven bij slimme meter versie. Nu dacht ik, ik check dat even, maar kan het zo snel nergens terugvinden.

dennissiemensma, 2020-01-11:
Je hebt een v5-meter als je elke seconde een meting ontvangt. Anders is het elke 10 seconden.

Het verhogen van de sleep lost het niet direct voor je op, maar verlaagt de eventuele hoeveelheid telegrammen die je krijgt en daarmee ook de hoeveelheid data in de toekomst.

----

Wellicht moet je dan even handmatig kijken of je toevallig veel onverwerkte telegrammen hebt:

```
sudo su - postgres
psql dsmrreader 
select count(id) from dsmr_datalogger_dsmrreading where processed = false;
```

raymond79nl, 2020-01-12:
Hij loopt slechts 1 telegram achter, dus dat is het probleem ook niet. Ik dacht gisteren even, ik laat het er gewoon bij aangezien alles inmiddels redelijk leek te werken, op de status pagina na. Daarna is de Pi echter 2x vastgelopen, met enkele uren tussen de events (niet meer bereikbaar via webinterface of SSH). Een hard reboot werkt wel, maar ik zie geen melding in een van de loggings die we eerder raadpleegden (ook niet met DEBUG logging aan). Heb je een advies, anders dan nog eens een fresh install proberen?

dennissiemensma, 2020-01-12:
Als je DEBUG-logging aan hebt staan, moet je sowieso een heel uitgebreide output zien in de log van de dsmr_backend, met alles wat die doet. Bijvoorbeeld:

```
[2020-01-12 11:00:53,576] DEBUG    SP: 3 backend service(s) ready to run
[2020-01-12 11:00:53,576] DEBUG    SP: Running "Weather update" (dsmr_weather.services.run)
[2020-01-12 11:00:53,580] DEBUG    Buienradar: Reading temperature: https://data.buienradar.nl/2.0/feed/json
[2020-01-12 11:00:53,655] DEBUG    Buienradar: Read temperature: 6.0
[2020-01-12 11:00:53,677] DEBUG    SP: Rescheduled "Weather update" to 2020-01-12 12:00:00+01:00 (ETA 0:59:06.322336)
[2020-01-12 11:00:53,677] DEBUG    SP: Running "Generate day and hour statistics" (dsmr_stats.services.run)
[2020-01-12 11:00:53,721] DEBUG    Stats: Waiting for first gas reading on the next day...
[2020-01-12 11:00:53,722] DEBUG    SP: Rescheduled "Generate day and hour statistics" to 2020-01-12 11:05:53.721251+01:00 (ETA 0:04:59.998959)
[2020-01-12 11:00:53,722] DEBUG    SP: Running "Automatic update checker" (dsmr_backend.services.update_checker.run)
[2020-01-12 11:00:53,834] DEBUG    SP: Rescheduled "Automatic update checker" to 2020-01-19 11:00:53.819344+01:00 (ETA 6 days, 23:59:59.984819)
```

Als je dat niet ziet, dan staat het vermoedelijk niet goed. Vergeet ook niet de processen te herstarten. Zodra je het wel ziet, kun je op zoek naar dingen die relatief lang (meerdere seconden) duren. 

Een ander ding wat je nog kunt doen is [`atop` installeren/gebruiken](https://haydenjames.io/use-atop-linux-server-performance-analysis/). Deze maakt regelmatig een snapshot van alle processen en geheugengebruik, zodat die je na een crash kunt inzien en doorspoelen naar het moment vlak voor de crash. Het geeft dan vaak een goede hint welk proces de boosdoener was. Dat kan ook helpen in het zoeken naar de oorzaak.

raymond79nl, 2020-01-13:
Ik heb na een 3e crash (nog voordat ik naar logging settings kon kijken) toch maar een nieuwe install gedaan. Deze keer met een verse SSD. Tot nog toe draait het geheel 24+ uur stabiel (status pagina opent nog steeds niet, maar daar kan ik wel mee leven). Even aankijken...

dennissiemensma, 2020-01-13:
Fijn! Dus als ik het goed begrijp heb je een Pi met daaraan een SSD-schijf hangen? Ik ga er vanuit dat dat in ieder geval voor zorgt dat je SD-kaartje niet zo snel corrupt raakt.

raymond79nl, 2020-01-14:
Yes. In de vorige poging ook al (hoewel dat een traditionele laptop HDD was). Hij draait nu nog steeds stabiel, dus het lijkt erop dat we dit issue kunnen sluiten. Wellicht gewoon pech met de HDD. Bedankt voor je hulp!

closed 2020-01-14 by dennissiemensma