dsmrreader/dsmr-reader #832: Verbruik voorgaande jaren toevoegen(PRE DSMR Reader).

quadrophenianl, 2020-01-10:
Nogmaals een goedemorgen Dennis,

Excuus voor nog een nieuw topic, maar naar verwachting ben je dan van me af ;) Waarschijnlijk biedt dit project me dan alles wat ik nodig heb.

Aangezien ik van de vorige jaren gespecificeerde eindafrekeningen heb liggen, wil ik graag het totaalverbruik van dit jaren toevoegen aan de database. Zodoende kan ik afgesloten jaren met elkaar vergelijken. Aangezien ik deze reader pas sinds dit jaar gebruik. heb ik geen data ingeladen van dit jaren. Ik had al eens wat data toegevoegd aan de database, maar dit gaf niet het gewenste resultaat. 

Wel kan ik gewoon aan de frontend oude energiecontracten invoeren, maar helaas mis in de mogelijkheid om de jaar totalen toe te voegen.

Wellicht is dit op een of andere manier mogelijk? Eventueel wil ik ook wel sleutelen aan de database, maar dan moet ik even weten waar ik hem moet bewerken. Het toevoegen van items bij de dag en maand statistieken heeft me niet verder geholpen. De archief pagina leef namelijk leeg na inladen...

Ik hoor het graag en alvast een fijn weekend gewenst!

Groeten Mark

dennissiemensma, 2020-01-10:
Hoe heb je destijds de oude gegevens toegevoegd aan de database? Er zijn geen maandstatistieken, maar wel dagstatistieken. Bedoelde je wellicht uurstatistieken?

BlackDex, 2020-01-10:
Ik was hier toevallig net naar aan het zoeken waar ik dit kon doen.
Zo te zien, dus (nog) niet mogelijk op een makkelijke manier.

Ik kan in ieder geval aan rudimentaire data komen per dag via de greenchoice website.

dennissiemensma, 2020-01-10:
Je kunt dit handmatig doen via de admin-interface: `/admin/dsmr_stats/`

Het is echter oorspronkelijk niet bedoeld om daarmee zelf nieuwe gegevens toe te voegen (alleen herstellen na bijvoorbeeld meterwissel), maar ik zie geen reden in waarom dat niet zou kunnen werken. 
Je zou voor elke maand al het verbruik op 1 dag kunnen zetten, zodat het maand- en jaartotaal wel klopt.

En anders zul je het per dag moeten invullen. Als er behoefte is kan ik het ook nog een keertje via de API ontsluiten.

quadrophenianl, 2020-01-13:
Dank! Al kan ik niet zo snel de admin-interface vinden helaas... Heb wel nog niet heel veel tijd gehad om hier naar te kijken vanwege andere werkzaamheden.

dennissiemensma, 2020-01-13:
Het staat bij Dagstatistieken of Day statistics. Ergens bijna onderaan.

Je kunt daar nieuwe records aanmaken als het goed is.

quadrophenianl, 2020-01-14:
Werkt perfect, waarvoor dank!

closed 2020-01-14 by dennissiemensma