dsmrreader/dsmr-reader #833: Mqtt verbindt niet opnieuw na herstart mosquitto

duiffie, 2020-01-10:
**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.14.1`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `VirtualBox`
* DSMR-protocol slimme meter *(v4 of v5)*: `v5`

Als ik mosquitto herstart dan wordt de mqtt verbinding niet opnieuw gemaakt. Een herstart van DSMR reader lijkt nodig om de mqtt verbinding weer tot stand te brengen.

dennissiemensma, 2020-01-10:
Bedankt voor je melding. Jammer dat de client niet opnieuw verbind. Ik zal er in de v3.x versie een keer naar kijken. Voor nu zou je eventueel een hourly cronjob kunnen instellen die `dsmr_mqtt` telkens herstart oid.

dennissiemensma, 2020-01-14:
Ik heb inmiddels de oorzaak gevonden, maar helaas nog niet direct een oplossing.

De client detecteert wel goed wanneer de verbinding wegvalt. En er zit al langere tijd foutafhandeling in dat die op dat moment 1x opnieuw probeert te verbinden en anders het hele `dmsr_mqtt` proces stopt. Zodat Supervisor hem direct weer vers start en de client net zolang opnieuw probeert te verbinden totdat het weer lukt.

Echter, het blijkt nu dat die foutafhandeling deels werkt. Ik zie in de logs dat de wegvallende verbinding direct wordt gedecteerd, de reconnect faalt, mijn foutafhandeling gooit een exception om het proces te crashen, maar... blijkbaar wordt die exception weer door de client zelf afgevangen of iets.

```
...
[2020-01-14 19:27:10,939] DEBUG    dsmr_mqtt.management.commands.dsmr_mqtt: Sleeping 1.0s
[2020-01-14 19:27:12,051] DEBUG    dsmr_mqtt.management.commands.dsmr_mqtt: Sleeping 1.0s
[2020-01-14 19:27:13,158] DEBUG    dsmr_mqtt.management.commands.dsmr_mqtt: Sleeping 1.0s
[2020-01-14 19:27:14,161] DEBUG    MQTT: Paho client: on_disconnect(userdata, rc) None 1
[2020-01-14 19:27:14,161] WARNING  MQTT: --- Unexpected disconnect, re-connecting...
[2020-01-14 19:27:14,161] ERROR    MQTT: Failed to re-connect to broker: [Errno 111] Connection refused
[2020-01-14 19:27:14,162] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 8 Caught exception in on_disconnect: 
[2020-01-14 19:27:14,166] DEBUG    dsmr_mqtt.management.commands.dsmr_mqtt: Sleeping 1.0s
[2020-01-14 19:27:15,178] DEBUG    dsmr_mqtt.management.commands.dsmr_mqtt: Sleeping 1.0s
...
```

Ik zoek nog even verder. In het ergste geval moet ik het proces zichzelf een `kill -SIGHUP` laten geven.

dennissiemensma, 2020-01-14:
Ik heb een workaround kunnen vinden. In v3.1 komt de fix beschikbaar.

closed 2020-01-14 by dennissiemensma