dsmrreader/dsmr-reader #834: Cannot delete notifications

lukevredeveld, 2020-01-10:
So, last week I was on holiday for a whole week, while my internet connection was interupted. When I got home, I got an enormous amount of notifications of Buienradar api. After clicking for 10 minutes to delete them all, the notifications didn't go away anymore. Instead, the green icon keeps spinning forever. Although I have managed to delete most of them, I still have several (~15) left and the  notifications are adding up (also not able to delete those).

Any change for me to delete those nasty buggers ;-) ?

[image]

dennissiemensma, 2020-01-10:
Bedankt voor je melding. Ik neem aan dat je niet de laatste versie van DSMR-reader draait? De laatste versie heeft het in bulk sluiten van notificaties mogelijk gemaakt. 

Zie hier voor een screenshot: https://github.com/dennissiemensma/dsmr-reader/issues/793#issuecomment-570944735

Wil je eens kijken of dat het voor je oplost?

closed 2020-01-14 by dennissiemensma

lukevredeveld, 2020-01-14:
Ik liep inderdaad 1 release achter. Heb met de nieuwste release alles kunnen verwijderen!