dsmrreader/dsmr-reader #835: Geen Data uit DSMR met RS485

ilhanu, 2020-01-10:
**Wat gebruik je?**
* DSMR-reader versie: `v2.14.1`
* Hardware : `RaspberryPi 3`
* DSMR-protocol slimme meter: `v2_3` 

**Verhaal:**

Meter die ik probeer uit te lezen is industrieel en vandaar dat ik niet de DSMR versie weet. [Landys-Gyr E650](https://www.landisgyr.eu/webfoo/wp-content/uploads/2012/09/D000030108-E650-ZMD300xT-Series-3-User-Manual-en-k.pdf).  Het gaat om een meter die alleen aangesloten is op zonnepanelen.

Naast de PI, gebruik ik volgende onderdelen.
- A RS485 kabel van de smart meter, geinstaleerd door de netwerk beheerder. 
- The RS485 kabel naar USB converter (ch341-uart converter)

Ik zie de usb op `/dev/ttyUSB0` , alleen tot nu toe niet succesvol geweest om data eruit te krijgen. Kan de front-end volledig runnen, ik zie amper iets terug in de logs dat er iets fout gaat. 

Wat kan ik nog meer proberen?

dennissiemensma, 2020-01-10:
Heb je toevallig als handmatig op de commandline uitlezen via `cu` geprobeerd? Dat is de beste plek om te beginnen, ook om uit te zoeken welke connectieparameters (baud rate etc) nodig zijn.

Zie voor een voorbeeld: https://dsmr-reader.readthedocs.io/en/latest/installation/explained.html#your-first-reading-optional

ilhanu, 2020-01-13:
Ik krijg connected eruit, maar er komt geen data. 
Heb gezocht heb ook de volgend [Gurux.DLMS.Python](https://github.com/gurux/gurux.dlms.python) repo geprobeerd en krijg dat het geen verbinding kan maken met de data.

dennissiemensma, 2020-01-13:
Ik denk dat je wellicht je situatie op het Tweakers-forum kan plaatsen. Vooral in dit topic is een hoop kennis aanwezig over allerlei meters, kabels en combinaties: https://gathering.tweakers.net/forum/list_message/61153878

Ik heb zelf alleen een oudere DSMR-v4 meter, dus ik kan je helaas niet inhoudelijk helpen met uitzoeken hoe je meter werkt.

closed 2020-01-15 by dennissiemensma