dsmrreader/dsmr-reader #840: ""

raoulteeuwen, 2020-01-15:
Gaat er iets fout, maar weet je niet wat? [Bekijk de logfiles](https://dsmr-reader.readthedocs.io/nl/v2/troubleshooting.html) als startpunt.

**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.xxx (weet ik niet: ben dsmr aan het upgraden)
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `RaspberryPi 4 4GB
* DSMR-protocol slimme meter *(v4 of v5)*: `v...` (weet ik niet)

Ik heb https://dsmr-reader.readthedocs.io/en/v3/faq/v3_upgrade.html gevolgd. Alles ging mi goed, tot de laatste stap. Ik krijg dan:

"
(dsmrreader) dsmr@RPi4:~/dsmr-reader $ ./deploy.sh

 --- You are currently running version: 
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/dsmr/dsmr-reader/dsmrreader/__init__.py", line 1, in <module>
    from django.utils.version import get_version
ModuleNotFoundError: No module named 'django'

     [!] FAILED to call Django (did you activate the 'dsmrreader' VirtualEnv before running?)

[!] Halted in pre-deployment: Either user ABORTED or script FAILED

"

raoulteeuwen, 2020-01-15:
pip3 list levert

(dsmrreader) dsmr@RPi4:~/dsmr-reader $ pip3 list
Package         Version
--------------- -------
pip             19.3.1 
pkg-resources   0.0.0  
psycopg2-binary 2.8.4  
setuptools      45.0.0 
wheel           0.33.6

dennissiemensma, 2020-01-15:
Wil je eens kijken of dit het oplost:

```
sudo su - dsmr
pip3 install -r dsmrreader/provisioning/requirements/base.txt
```

En dan weer verder gaan met waar je was.

raoulteeuwen, 2020-01-15:
Yes, nu loopt hij door, dank!

dennissiemensma, 2020-01-15:
Jij bedankt voor het melden. Kleine check waar ik geen rekening mee gehouden had.

Het is inmiddels aangepast in de documentatie.

closed 2020-01-15 by dennissiemensma

Djjvb, 2020-02-11:
Hallo,

Ik heb ook dit issue na de install gevolgd te hebben. Met de volgende melding:

 --- You are currently running version: 
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/dsmr/dsmr-reader/dsmrreader/__init__.py", line 1, in <module>
    from django.utils.version import get_version
ModuleNotFoundError: No module named 'django'

     [!] FAILED to call Django (did you activate the 'dsmrreader' VirtualEnv before running?)

[!] Halted in pre-deployment: Either user ABORTED or script FAILED

Heb toen de oplossing hierbij toegepast:
pip3 install -r dsmrreader/provisioning/requirements/base.txt

Toen kreeg ik de volgende melding:
pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Collecting https://qpypi3.qpython.org/repository/1745924/PyCRC-1.21.tar.gz (from -r dsmrreader/provisioning/requirements/base.txt (line 19))
Exception:
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 423, in send
    timeout=timeout
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 582, in urlopen
    conn = self._get_conn(timeout=pool_timeout)
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 255, in _get_conn
    return conn or self._new_conn()
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 814, in _new_conn
    raise SSLError("Can't connect to HTTPS URL because the SSL "
requests.packages.urllib3.exceptions.SSLError: Can't connect to HTTPS URL because the SSL module is not available.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/req/req_set.py", line 620, in _prepare_file
    session=self.session, hashes=hashes)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/download.py", line 821, in unpack_url
    hashes=hashes
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/download.py", line 659, in unpack_http_url
    hashes)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/download.py", line 853, in _download_http_url
    stream=True,
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 501, in get
    return self.request('GET', url, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 488, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 497, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: Can't connect to HTTPS URL because the SSL module is not available.

Kortom: help! Nu doet niks het meer en heb geen idee wat ik aan het doen ben. 

Alvast bedankt!

dennissiemensma, 2020-02-11:
@Djjvb je kunt hier het beste even een apart issue voor maken, om te voorkomen dat iedereen die ooit dit topic heeft gevolgd er ook allemaal mail van krijgt.

https://github.com/dennissiemensma/dsmr-reader/issues/new?assignees=&labels=&template=help.md&title=