dsmrreader/dsmr-reader #841: Dropbox sync: OAuth token fout: foutmelding en gedrag kan beter

tomkooij, 2020-01-18:
(Dit is DSMRreader v2.12.1 in ualex docker: niet getest in v3)

Ik heb het verkeerde Dropbox OAuth token ingevoerd in het admin panel... Stom.

Er was geen foutmelding in de webinterface. Alleen "laatste dropbox sync" datum bleef staan op de eerste poging en in Dropbox zelf gebeurde er niets. Geen nieuwe map in `/Apps` enz.

In de `supervisor.log` van dsmr_backend staat **elke seconde**:
``` 
[2020-01-18 06:58:25,981] ERROR    BadInputError('<token>', 'Error in call to API fun
ction "files/get_metadata": The given OAuth 2 access token is malformed.')
NoneType: None
```
Deze foutmelding is ook niet overduidelijk van de dropbox module... (het module veld is leeg)

Ik zou graag willen: 
 - Een duidelijke foutmelding "API token fout" oid in de webinterface.
 - Niet elke seconde opnieuw proberen.
 - module naam in logfiles

Ik ben bereid een PR te maken. Laat maar weten.

dennissiemensma, 2020-01-18:
Bedankt voor je melding. Dit was eerder een bekend probleem en zou in #264 (1.21) en #584 (2.6) moeten opgelost zijn.

Bij ongeldige credentials zou er een melding in de interface moeten komen en worden de credentials verwijderd, zodat die het niet blijft proberen: https://github.com/dennissiemensma/dsmr-reader/blob/v2.6.0/dsmr_dropbox/services.py#L128

Het kan echter zijn dat je tegen een andere foutmelding aanloopt die niet afgevangen wordt (of het is bij Dropbox veranderd). Ik heb er namelijk alleen unittests voor en geen integratietests.

Ik zal kijken of ik het kan reproduceren.

dennissiemensma, 2020-01-18:
Het lijkt er inderdaad op dat het is veranderd. Ik zal de foutafhandeling aanpassen.

dennissiemensma, 2020-01-18:
Opgelost voor v3.2

closed 2020-01-18 by dennissiemensma