dsmrreader/dsmr-reader #842: Gasgrafiek als staafdiagram

mkruiver, 2020-01-18:
DSMR draait nog steeds goed, dank daarvoor!

Ik heb een verzoekje: ik zou graag de gasgrafiek als staafdiagram kunnen weergeven, zodat ik bijvoorbeeld kan zien dat er in een bepaald uur 0.6m3 verbruikt is. De huidige representatie met m3/h op de y-as vind ik erg lastig te lezen.

dennissiemensma, 2020-01-18:
Ik heb even voor je geprobeerd of het daadwerkelijk verschilt qua weergave en dat is inderdaad zo. Ik zal het in v3.2 standaard aanpassen naar staaf. Als er veel klachten over komen dan maak ik het wel instelbaar.

Voorbeelden:
[image]

[image]

dennissiemensma, 2020-01-18:
@mkruiver je kunt het trouwens voor jezelf aanpassen.

`/var/www/dsmrreader/static/dsmr_frontend/js/dsmr-reader/dashboard/gas.js` op regel `59`:
```
type: 'line',
```

Naar:
```
type: 'bar',
```

Idem voor `/var/www/dsmrreader/static/dsmr_frontend/js/dsmr-reader/archive/gas.js` op regel `54`.

Bij de eerste volgende deploy die je doet wordt het weer overschreven met de repo standaard.

closed 2020-01-18 by dennissiemensma

mkruiver, 2020-01-18:
Gaaf, dat was snel! Het werkt inderdaad, mooi, bedankt!

Kan ik nog ergens instellen dat ik meer dan 5min per staafje wil zien, bijv 1 staaf per 30 of 60min ipv 5?

[image]

dennissiemensma, 2020-01-18:
Daarvoor moet eerst iets gemaakt worden dat de metingen kan groeperen. Voor nu is het afhankelijk van of je een v4 of v5 meter hebt.

FutureCow, 2020-01-31:
Zou het ook mogelijk zijn om in de nieuwe staafdiagram een gedeelte te selecteren, zodat je kunt zien hoeveel er in dat tijdbestek is verbruikt?

Mixpower, 2020-02-01:
Ik zou het toch erg prettig vinden als ik dit instelbaar is, ik neb namelijk veel liever een line om de hele flow te zien dat is nu niet meer mogelijk en zit ik met een staafdiagram.

dennissiemensma, 2020-02-01:
@Mixpower bedankt voor je feedback. Dan zal ik daar wat voor maken.

Voor nu kun je dit uitvoeren: https://github.com/dennissiemensma/dsmr-reader/issues/842#issuecomment-575893210

En van `bar` weer `line` maken.

dennissiemensma, 2020-02-01:
Ik heb er  #860 voor gemaakt

dennissiemensma, 2020-02-01:
@FutureCow ik denk het niet. Wat je wel kunt doen is het weer terugzetten naar lijn of het gas laten groeperen per uur. Dat kan al iets helpen.

mkruiver, 2020-02-01:
Waar kun je eigenlijk groeperen per uur? En is dat dan alleen voor de
frontend of ook voor de database?

Mixpower, 2020-02-01:
> Waar kun je eigenlijk groeperen per uur? En is dat dan alleen voor de frontend of ook voor de database?
> […](#)

Configuratie > Verbruiksconfiguratie > Gasmetingen groeperen

dennissiemensma, 2020-02-02:
> En is dat dan alleen voor de frontend of ook voor de database?
Voor de database.

Helloitsme, 2020-02-15:
is het ook mogelijk om ook voor stroom er een (stacked) bar van te maken? vooral in het maand overzicht zou dit handig zijn want nu heb ik 2 lijn grafieken: voor laag en hoog tarief, maar daardoor kan je het totaalverbruik van een dag niet makkelijk zien. met een stacked bar kan je dat wel

dennissiemensma, 2020-02-16:
@Helloitsme ik denk dat je hier het beste een apart issue voor kan maken. Dan kijk ik tzt wat mogelijk is.