dsmrreader/dsmr-reader #846: Teruglevering in Belgie

thijscrombeen, 2020-01-20:
In Belgie is het zo dat je niet betaald krijgt voor de teruglevering van elektriciteit op het net.
Je betaalt namelijk voor het verbruik tussen 2 officiele meetpunten (bijv van 01/05/2018 tot 30/04/2019). Is je verbruik tussen die periodes negatief dan betaal je niets. Het cijfer op je teller na het meetpunt (op 01/05/2019) wordt dan weer gezien als 0-punt en dan begint het weer opnieuw.

Is het mogelijk om iets te bouwen dat hier mee om kan gaan? Met andere woorden dat vanaf een bepaalde datum alles wat wordt teruggeleverd wordt opgeteld. Hetgeen wordt verbruikt wordt hier dan van afgehaald. Enkel als je meer verbruikt dan dat je oplevert zou dit als kost moeten gezien worden. Ik heb al enkele topics hierover gezien en volgens mij brengt dat volgende problemen mee.
Voor zover ik zie moet je een teller bijhouden met netto verbruik.

- als je start op 01/05 dan gaat deze teller heel de zomer meer negatief worden aangezien je doorgaans veel meer oplevert dan verbruikt. In de winter zal deze netto teller stilletjes naar boven gaan. In het geval van de dagelijkse kost bereking gaat het volgens mij wel goed.
- als je start op 01/12 dan zal deze teller doorgaans oplopen omdat je in de winter veel meer verbruikt dan je opbrengt. Na de winter zal deze teller naar 0 of kleiner gaan. In dit geval gaat de dagelijkse kostberekening niet kloppen.

Mogelijks zou dit een feature kunnen zijn waar je deze logica voor zou kunnen berekenen binnen een periode. Dus dat je in de UI bijvoorbeeld 01/05/2018 en 30/04/2019 op kunt geven en deze dan je verbruik - oplevering toont en daar de kost van berekent. (Indien je meer teruglevert dan verbruikt zou het 0€ moeten opleveren)

Een andere optie kan zijn om enkel het teruglever bedrag te tellen voor het aantal kwh dat je ook die dag verbruikt. Maw als je 1kwh verbruikt en er 2 teruglevert dan zou het bedrag 0€ moeten zijn voor die dag. Dit zou per dag toch al een iets realistischere waarde moeten geven ook al is over een periode niet correct.

Laat maar weten wat je hierover denkt en bedankt voor deze fantastische applicatie!

dennissiemensma, 2020-01-20:
Bedankt voor je verzoek. Het is een heel verhaal, maar begrijp ik het goed dat het afdoende is wanneer je simpelweg zelf een periode (zoals van 01/05/2018 tot 30/04/2019) kan selecteren en daar de totalen van ziet? 
Dan hoef je ook geen tellers of iets bij te houden. Er is namelijk al een soortgelijk verzoek om zoiets te doen voor een contractperiode en dit is een variant erop.

thijscrombeen, 2020-01-21:
Als het voor die periode de teruglevering in rekening houdt tot aan 0 dan lijkt me dat afdoende.
Bedankt voor je snelle reactie

dennissiemensma, 2021-03-07:
Hier heb hier opnieuw naar gekeken. Inmiddels is #1084 ook aangemaakt in het najaar en lijkt dat het meeste op wat nodig is qua  wijzigingen voor teruglevering in Belgie. 
Ik weet niet of ik dat er daadwerkelijk ook goed in ga krijgen, maar hoe dan ook zal ik die feature eerst proberen.

closed 2021-03-07 by dennissiemensma