dsmrreader/dsmr-reader #848: Het uitlezen van de P1 stopt na een willekeurige tijd maar altijd minder dan 24uur

Pendora, 2020-01-22:
Goedendag, 

Ik heb een probleem waarmee ik hoop dat je me kunt helpen en hierbij alle informatie:

Hardware; MacMini (OS 10.14.6) —> Virtualbox (6.1) —> Debian (10) 
P1 lezer: FTDI FT232R USB UART [0600]
DSMR 3.0 (zelfde probleem met 2.X)

Probleemomschrijving;
Het uitlezen van de P1 stopt na een willekeurige tijd maar altijd minder dan 24uur. Een herstart van de VM is het enige dat werkt om het uitlezen weer te starten. 

Wat heb ik geprobeerd:
Dit is mijn eerste poging met DSMR dus alles is nog nieuw maar ik heb sowieso de standaard dingen zoals herinstallatie, opnieuw starten ook geprobeerd. 

De P1 werd eerst uitgelezen in een andere VirtualBox met daarop Home-Assistant. Hier had ik nooit problemen. 

Daarna een extra VM aangemaakt puur voor DSMR en werkt perfect tot het uitlezen stopt. 

Doormiddel van CU heb ik de poort uitgelezen voor tweemaal 48 uur en hierbij geen enkele storing. 

Datalogger proces sleep verhoogd naar 2 seconden

Data logger:

Start met:

[2020-01-21 08:43:06,046] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Starting infinite command loop...

Daarna alle telegrammen van de P1 reader tot het volgende:

[2020-01-21 09:42:35,839] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 0.5s
[2020-01-21 09:42:46,195] INFO
/ISk5\2MT382-1003

0-0:96.1.1(5A424556303035303933323137373132)
1-0:1.8.1(11011.320*kWh)
1-0:1.8.2(12353.821*kWh)
1-0:2.8.1(00553.010*kWh)
1-0:2.8.2(01408.229*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(0000.24*kW)
1-0:2.7.0(0000.00*kW)
0-0:17.0.0(0999.00*kW)
0-0:96.3.10(1)
0-0:96.13.1()
0-0:96.13.0()
0-1:24.1.0(3)
0-1:96.1.0(3238303131303031323133353334303132)
0-1:24.3.0(200121090000)(00)(60)(1)(0-1:24.2.1)(m3)
(07012.202)
0-1:24.4.0(1)
!

[2020-01-21 09:42:46,239] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 0.5s

Als ik vervolgens met CU handmatig probeer de data binnen te halen krijg ik alleen:

cu -l /dev/ttyUSB0 -s 9600 --parity=none
Connected.

sudo supervisorctl
dsmr_backend                     RUNNING   pid 351, uptime 4:03:54
dsmr_datalogger                  RUNNING   pid 350, uptime 4:03:54
dsmr_webinterface                RUNNING   pid 349, uptime 4:03:54

Herstarten van deze services maakt niks uit. 

DSMR_backend.log:

[2020-01-21 09:42:50,580] DEBUG    SP: 0 backend service(s) ready to run
[2020-01-21 09:42:50,587] DEBUG    Compact: Waiting for newer readings before grouping data...
[2020-01-21 09:42:50,591] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s

En dit is hetzelfde ervoor het stoppen van het lezen als erna

supervisord.log
Geen enkele informatie rond die tijd. 

dsmr_webinterface.log
Geen enkele informatie rond die tijd.

dennissiemensma, 2020-01-22:
Bedankt voor je melding. Dus als ik het goed begrijp:
- Je leest de P1 poort met max 1 applicatie tegelijkertijd uit?
- De datalogger stopt zonder foutmelding in `/var/log/supervisor/dsmr_datalogger.log`?
- Alleen de VM herstarten lost het op? Of is een herstart van het datalogger-proces ook afdoende?

Ik ben benieuwd:
- Welke DSMR-versie heeft je meter? In het telegram zie ik geen versie staan, is het een oudere meter?
- Werkt `cu` wel of niet? En onder welke omstandigheden dan wel of niet?

Wat je nog kunt doen is kijken welke systemcalls gedaan worden zodra de datalogger hangt:

- `sudo ps faux | grep datalogger | grep -v grep`
- `sudo strace -fv -p PROCESSID`

Bijvoorbeeld:
```
sudo ps faux | grep datalogger | grep -v grep
dsmr      8608  2.1  2.7  68856 55248 ?        SN   Jan18 127:15  \_ /home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger
```

```
sudo strace -fv -p 8608
strace: Process 8608 attached
_newselect(11, [9 10], [], [], {tv_sec=19, tv_usec=305937}
```

Je hoeft niet alle output hier te plakken, maar je kunt wel zien of die ergens op hangt. Ik heb een v4-meter die elke 10s wat doet, dus ik zie dat die vaak even hangt, maar na 10s opeens een shitload aan output geeft.

Pendora, 2020-01-23:
Per vraag een antwoord:

- Je leest de P1 poort met max 1 applicatie tegelijkertijd uit?
Ja. Alleen DSMR. 

- De datalogger stopt zonder foutmelding in /var/log/supervisor/dsmr_datalogger.log?
Correct. 

- Alleen de VM herstarten lost het op? Of is een herstart van het datalogger-proces ook afdoende?
Alleen de herstart lost het probleem op. 

- Welke DSMR-versie heeft je meter? In het telegram zie ik geen versie staan, is het een oudere meter?
2 het is een oude meter geplaatst in 2012. 

- Werkt cu wel of niet? En onder welke omstandigheden dan wel of niet?
Als er nog geen storing is en ik het CU commando uitvoer dan stopt DSMR uiteraard maar het CU commando blijft uitlezen. Tweemaal 48 uur uitgeprobeerd en daarbij geen enkele storing. Als er al wel een storing is dan krijg ik met het CU commando ook geen metingen meer. 

- sudo ps faux | grep datalogger | grep -v grep
dsmr       363  0.4 16.2 127608 80192 ?        SN   07:41   0:30  \_ /home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger

- sudo strace -fv -p PROCESSID
strace: Process 363 attached
select(11, [9 10], [], [], {tv_sec=7, tv_usec=55197}) = 0 (Timeout)
select(11, [9 10], [], [], {tv_sec=19, tv_usec=999992}) = 0 (Timeout)
select(11, [9 10], [], [], {tv_sec=19, tv_usec=999993}) = 0 (Timeout)
select(11, [9 10], [], [], {tv_sec=19, tv_usec=999992}) = 0 (Timeout)
select(11, [9 10], [], [], {tv_sec=19, tv_usec=999993}) = 0 (Timeout)
Etc. 

Momenteel laat ik de VM elk uur herstarten. Dan mis ik 3 metingen maar verder blijft die wel altijd werken.

dennissiemensma, 2020-01-23:
Ik vermoed dat het ergens in de mapping van de seriele poort via de virtualisatie zit, omdat de poort voor het hele OS niet meer te gebruiken is zodra die vastloopt.

Alleen heb ik hier geen ervaring mee en weet ik ook niet wat het veroorzaakt. Je zou nog kunnen kijken in `/var/log/syslog` binnen de VM, maar ik verwacht niet dat daar iets in staat.

dennissiemensma, 2020-01-28:
@Pendora je zou trouwens nog je issue kunnen voorleggen in https://gathering.tweakers.net/forum/list_messages/1578510. 

Wellicht dat anderen een soortgelijk issue hebben gehad met virtualisatie en/of DSMR-reader.

Pendora, 2020-01-29:
Ik ben er nog verder ingedoken ook met de algemene logs maar ik ben bang inderdaad dat dit een vaag probleem van de VM is. Wel vreemd dat HomeAssistant nooit deze problemen had. Ik zal even kijken naar Tweakers.

dennissiemensma, 2020-01-30:
Mocht ik je nog op ander gebied kunnen assisteren dan hoor ik het graag. Voor nu laat ik dit dan voor wat het is.

closed 2020-01-30 by dennissiemensma