dsmrreader/dsmr-reader #850: Fresh install Ubuntu 18.04: No module named 'django'

Milithor, 2020-01-24:
Gaat er iets fout, maar weet je niet wat? [Bekijk de logfiles](https://dsmr-reader.readthedocs.io/nl/v2/troubleshooting.html) als startpunt.

**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.xxx`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `RaspberryPi 3B+ voor logger, Ubuntu 18.04 voor de dsmreader`
* DSMR-protocol slimme meter *(v4 of v5)*: `v5`

Ik probeer de dsmreader te installeren op mijn Ubuntu server. Wanneer ik de volgende guide volg: https://dsmr-reader.readthedocs.io/en/latest/installation/quick.html stuit ik bij de volgende lijn op een error.

lijn: 

> sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/dsmr-reader/manage.py migrate

Error: 

> Traceback (most recent call last):
>   File "/home/dsmr/dsmr-reader/manage.py", line 8, in <module>
>     from django.core.management import execute_from_command_line
> ModuleNotFoundError: No module named 'django'

Milithor, 2020-01-24:
> ERROR: Could not find a version that satisfies the requirement PyCRC==1.21 (from -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt (line 12)) (from versions: 0.9.2)
> ERROR: No matching distribution found for PyCRC==1.21 (from -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt (line 12))

Bovenstaande ook gezien bij volgend commando: 

> sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/postgresql.txt

Ivan86NL, 2020-01-25:
Same issue here with a fresh install on Ubuntu 19.10.

When running the "manual"-steps it goes wrong on line (logged in as dsmr user (sudo su - dsmr)):
`virtualenv ~/.virtualenvs/dsmrreader --no-site-packages --python python3`

With the following error:
> Running virtualenv with interpreter /home/dsmr/.virtualenvs/dsmrreader/bin/python3
> Please use the *system* python to run this script
> Traceback (most recent call last):
>   File "/usr/lib/python3/dist-packages/virtualenv.py", line 2379, in <module>
>     main()
>   File "/usr/lib/python3/dist-packages/virtualenv.py", line 724, in main
>     symlink=options.symlink)
>   File "/usr/lib/python3/dist-packages/virtualenv.py", line 950, in create_environment
>     site_packages=site_packages, clear=clear, symlink=symlink))
>   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/posixpath.py", line 378, in abspath
>     path = os.fspath(path)
> TypeError: expected str, bytes or os.PathLike object, not NoneType

Milithor, 2020-01-25:
Hetzelfde gaat fout op een clean install met een raspberry pi 3B+.

dennissiemensma, 2020-01-25:
@Milithor bedankt voor het melden. De foutmelding van Django krijg je omdat je eerder een andere foutmelding krijgt. 

Het lijkt erop dat een van de externe packages (`PyCRC`) is veranderd of overgedragen of iets anders, want de gepinde release is opeens niet beschikbaar. Ik zal hier verder naar moeten kijken.

dennissiemensma, 2020-01-25:
@Ivan86NL ik denk dat je foutmelding ongerelateerd is, omdat zowel de fout als de distro verschilt. je kunt daar het beste een los issue voor maken.

meizenga, 2020-01-25:
PyCRC is niet te vinden in de wheels repo. Run dit voor dat je `./deply` uitvoert.

`pip3 install http://qpypi3.qpython.org/repository/1745924/PyCRC-1.21.tar.gz`

dennissiemensma, 2020-01-25:
Het lijkt erop dat de oude auteur ermee gekapt is. De oude repo (leek) te draaien op: https://github.com/cristianav/pycrc/

Ik heb nog wel een kopie van code van de virtualenv staan. Ik kijk even of ik dat ofwel toevoeg aan de repo, of bovenstaande tar toevoeg aan de repo en de requirements file.

dennissiemensma, 2020-01-25:
En op Pypi staat weer een andere repo gelinked: https://pypi.org/project/pycrc

dennissiemensma, 2020-01-25:
Ik heb het nu voor v3 opgelost. Voor v2 doe ik later nog een release.

Milithor, 2020-01-25:
@dennissiemensma Dank hiervoor. Ik heb de install gedaan die  @meizenga heeft voorgesteld en toen werkte het. De DSMReader is hierbij geïnstalleerd.

Ivan86NL, 2020-01-25:
> @Ivan86NL ik denk dat je foutmelding ongerelateerd is, omdat zowel de fout als de distro verschilt. je kunt daar het beste een los issue voor maken.

@dennissiemensma de oplossing welke je voor PyCRC hebt toegepast in v3.1.1 is ook voor mij de oplossing geweest. De snelle installer gaat nu geheel goed (op Ubuntu 19.10) en DSMR werkt zoals verwacht.

dennissiemensma, 2020-01-26:
Nu ook gefixt voor v2

closed 2020-01-26 by dennissiemensma

mkruiver, 2020-01-30:
Begrijp ik goed dat dit alleen een issue is voor nieuwe installaties?

dennissiemensma, 2020-01-30:
Vermoedelijk wel.

dennissiemensma, 2020-01-31:
Ik heb voor toekomstige release v3.3 de fix uit https://github.com/ndokter/dsmr_parser/pull/48 toegepast via 0400ec45118986c91461ca27744917f3b2615cf9 - credits aan @dupondje

Dit verwijdert ook de onnodige afhankelijkheid van de externe beschikbaarheid van `PyCRC-1.21.tar.gz` (als workaround). Ik had niet door dat `dsmr_parser` uiteindelijk maar een klein deel van PyCRC gebruikte.