dsmrreader/dsmr-reader #851: power outages

marco2152, 2020-01-25:
Hi - is it possible that DSMR start logging power outages? We had that issue in the past, and would have loved a logger to show my power company the number and duration of outages. Thnx!

Regards,
Marco

ps Love DSMR, thanks for your hard work

dennissiemensma, 2020-01-25:
Bedankt voor je verzoek. Ik denk niet dat dit mogelijk is, want om te loggen is er stroom nodig en als er geen stroom is, werkt de installatie (in de meeste gevallen) niet.

Ik denk dat je het beste iets van een scriptje op je Pi kan draaien wat elke minuut en reboot iets logt, zodat je kunt zien wanneer, hoevaak en hoelang de stroom er uit ligt.

closed 2020-01-25 by dennissiemensma

marco2152, 2020-01-25:
maar het verschil tussen de laatste keer loggen en de eerstvolgende keer
loggen is toch je power outage?

Groeten,
Marco

dennissiemensma, 2020-01-25:
Klopt, maar dat is geen garantie. Telegrammen wordt soms ook via het netwerk verstuurd en als die niet aankomen is het dus niet zeker of het een netwerk-probleem is of een stroomstoring of iets anders.

Ik denk tevens dat je voor je specifieke verzoek met een scriptje veel sneller en betrouwbaarder af bent.

marco2152, 2020-01-25:
👍🏻