dsmrreader/dsmr-reader #852: Na terugzetten back-up foutmelding: dsmr_datalogger: ERROR (spawn error)

Ivan86NL, 2020-01-26:
Gaat er iets fout, maar weet je niet wat? [Bekijk de logfiles](https://dsmr-reader.readthedocs.io/nl/v2/troubleshooting.html) als startpunt.

**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v3.1.1`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `ESXi VM - Ubuntu 19.10`
* DSMR-protocol slimme meter *(v4 of v5)*: `v5`

Op dit moment ben ik bezig mijn Raspberry Pi 3 DSMR-Reader om te zetten naar een Ubuntu 19.10 VM op VMware ESXi. Hiervoor heb ik de RPi eerste ge-update naar versie 3.1.1. Deze zelfde versie draait op Ununtu 19.10 (clean install). Hierna heb ik een back-up gemaakt op de RPi en deze kopieerd naar de VM. 

Het inlezen van de back-up lijkt zonder fouten te verlopen. Echter wanneer ik hierna `sudo supervisorctl start all` draai krijg ik de output dat dsmr_datalogger een ERROR geeft (dit was voor de import van de back-up niet het geval), zie onderstaande:
>  dsmr_backend: started
dsmr_datalogger: ERROR (spawn error)
dsmr_webinterface: started

Wanneer in de log van dsmr_datalogger bekijk zie ik alleen maar de volgende meldingen:

> The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v2/troubleshooting.html#logging
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v2/troubleshooting.html#logging

De meldingen veranderen niet wanneer ik `LOGGING['loggers']['commands']['level'] = 'DEBUG'` inschakel.

Hoe kan ik achterhalen wat de reden is dat de dsmr_datalogger niet goed wil starten? Zou graag de eerdere meetpunten mee overnemen :)

Ivan86NL, 2020-01-26:
Na nog wat verder in de code graven ben ik er achtergekomen dat de dsmr_datalogger met onderstaand command gestart wordt:
`/usr/bin/nice -n 5 /home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger`

Wanneer ik deze draai onder `sudo su - dsmr` krijg ik onderstaande (fout)melding:
> The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v2/troubleshooting.html#logging
[2020-01-26 08:26:31,866] ERROR    dsmr_datalogger.management.commands.dsmr_datalogger: [!] Exception raised. Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/serial/serialposix.py", line 265, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyUSB0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 90, in run_once
    self.run(data=self.data, **options)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 23, in run
    telegram = dsmr_datalogger.services.read_telegram()
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", line 73, in read_telegram
    serial_handle.open()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'

Blijkbaar probeert hij verbinding te maken met `/dev/ttyUSB0`. Dit was inderdaad de poort op de RPi waarmee deze verbinding had met de slimme meter. Echter heeft de VM deze poort niet. 
Wanneer ik de poort probeer te verwijderen via de webinterface krijg ik de melding dat dit ingevuld moet zijn. Zie onderstaand screenshot.
[image]

Hoe kan ik de poort toch uit de configuratie halen zodat dsmr_datalogger hier niet over struikelt?

dennissiemensma, 2020-01-26:
Bedankt voor je vraag. Wellicht heb ik er overheen gelezen, maar wat wil je precies bereiken? Wil je de datalogger wel of niet gebruiken in je nieuwe VM? 

Als je deze niet wilt gebruiken, kun je het datalogger proces in Supervisor uitschakelen. Afhankelijk van wanneer je geinstalleerd hebt, staat dit in ofwel `/etc/supervisor/conf.d/dsmr-reader.conf` ofwel `/etc/supervisor/conf.d/dsmr_datalogger.conf`. 
Je kunt het proces daarin uitcommenten en daarna `sudo supervisorctl reread` en `sudo supervisorctl update` uitvoeren.

Ivan86NL, 2020-01-26:
Ik heb een RPi als datalogger in de meterkast hangen. Deze verstuurd het via de API naar de VM.
Op de VM heb ik nu de stappen uitgevoerd welk je aangaf. Hiermee is het opgelost! Dankjewel!

closed 2020-01-26 by Ivan86NL