dsmrreader/dsmr-reader #853: NotNullViolation in datalogger, probably due to missing current consumption information

froggi123, 2020-01-26:
* DSMR-reader versie
[dsmr_datalogger-stderr---supervisor-yt433V.log](https://github.com/dennissiemensma/dsmr-reader/files/4113671/dsmr_datalogger-stderr---supervisor-yt433V.log)
: v3.1.1
* Hardware: RaspberryPi 3
* DSMR-protocol slimme meter *(v4 of v5)*

I do not get any dsmr readings, although serial connection seems to be fine.
From what I understand it looks like my meter does not send out current consumption information in Watts. (But there is e.g. Voltage and Current for all 3 phases and state of energy consumed/supplied on both different rates. Which would basically allow to calculate current power, maybe with limited accuracy.)

I get an error-log from dsmr-datalogger (dsmr_datalogger-stderr---supervisor-yt433V.log). I just take out two readings in the attached log.

Is this something which can be fixed or will just not work with the missing values?

dennissiemensma, 2020-01-27:
Hi, thanks for sharing. I think the best solution is to use the v2 API and manually add some (bogus) data for the fields omitted by your smart meter.

You could have a look at a combination of these two:
- https://github.com/ndokter/dsmr_parser
- The v2 DSMR-reader API: `POST /api/v2/datalogger/dsmrreading`

For more info about the latter, view the local API-documentation in your DSMR-reader installation, Visit the web address of your DSMR-reader and append `/docs/api#operation/v2_datalogger_dsmrreading_post` to the URL.

The API docs contain an example:

```
import requests
import json

# Don't forget to replace 'YOUR-DSMR-URL' and 'YOUR-API-KEY' with your own values.
response = requests.post(
    'http://YOUR-DSMR-URL/api/v2/datalogger/dsmrreading',
    headers={'Authorization': 'Token YOUR-API-KEY'},
    data={
        'electricity_currently_delivered': 1.500,
        'electricity_currently_returned': 0.025,
        'electricity_delivered_1': 2000,
        'electricity_delivered_2': 3000,
        'electricity_returned_1': 0,
        'electricity_returned_2': 0,
        'timestamp': '2017-04-15T00:00:00+02',
    }
)

if response.status_code != 201:
    print('Error: {}'.format(response.text))
else:
    print('Created: {}'.format(json.loads(response.text)))
```

----

So you can try to user dsmr-parser to read and parse the telegram from your meter, and then (in the same script) send the values read through the API, using bogus values for the `electricity_currently_delivered` and `electricity_currently_returned`.

dennissiemensma, 2020-01-27:
I could make the values optional, but I'm quite hessistant with changing the specs to anything non-DSMR, as that is the core of this project.

I'd rather not build in any exceptions for some situations that 'should' not happen. Or well, of which I know of at least.

froggi123, 2020-01-27:
Thank you for answering. As my meter is located in Switzerland, it probably does not behave exactly according the DSMR specs.
Actually I would prefer to have the `electicity_currently_delivered/returned` values in the protocol, instead of calculating it. But I guess there is not much I can do about it. So I will try with your suggested solution.

I absolutely agree with you, that this is a non-standard case, which is not in scope of the project.

closed 2020-01-27 by froggi123

dennissiemensma, 2020-01-28:
You're welcome. 

If you're unable to figure the script out I could assist a bit eventually, but it's somewhat hard to simulate your environment here for me anyway.