dsmrreader/dsmr-reader #855: Geen telegram data

HeyItsMeNobody, 2020-01-28:
**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v3.1`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `RaspberryPi 3B`
* DSMR-protocol slimme meter *(v4 of v5)*: `v5`

Ik probeer al een week of twee dit te laten werken en heb gisteren een P1 naar usb kabel besteld en ontvangen nadat ik zelf heb geprobeert een kabel te maken. Ik heb nu deze usb kabel aangesloten aan de pi maar ik krijg nog steeds geen data binnen.

cu en dsmr-datalogger zeggen alleen Connected/Starting

Geen idee wat ik nu moet doen.

dennissiemensma, 2020-01-28:
Bedankt voor je melding. Als je echter in `cu` ook geen telegrammen ziet, weet ik niet wat er mis kan zijn.

Mijn advies is om op Tweakers in https://gathering.tweakers.net/forum/list_messages/1578510 je situatie, hardware, slimme meter, kabel en pogingen te beschrijven. 
Wellicht is iemand tegen hetzelfde aangelopen bij hetzelfde type meter (die je hier niet vermeld hebt).

Zodra je telegrammen ziet in `cu`, maar DSMR-reader nog steeds niet werkt, kan ik je daadwerkelijk helpen.

dennissiemensma, 2020-01-30:
Voor nu sluit ik deze. Heropen gerust als je weer verder kunt maar ergens anders tegenaan loopt.

closed 2020-01-30 by dennissiemensma