dsmrreader/dsmr-reader #856: Piek / Dal tellers omgedraaid (Fluvius)

Milithor, 2020-01-29:
**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v3.1`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `RaspberryPi 3B+`
* DSMR-protocol slimme meter *(v4 of v5)*: `v5`

Ik heb de datalogger op een RBP 3B+ staan, en de dsmreader zelf op een VM, dit voor database retentie en performantie. (Ik wil mijn RBP ontlasten).
Zelf heb ik een meter van Fluvius en merk dat mijn Piek en Daluren omgekeerd worden gegeven.

[image]

De optie voor de datalogger om dit op Fluvius België te zetten is reeds gevonden, al heeft dit geen effect. Wellicht omdat de logger op een ander platform staat.

Kan je toelichten waar deze zaken aangepast moeten worden in dit geval?

Bedankt !

dennissiemensma, 2020-01-29:
Bedankt voor je vraag. Het hangt er vanaf op welke plek het telegram wordt omgezet.

Maar eerst: heb je alle DSMR-reader processen in de VM herstart nadat je bovenstaande instelling hebt doorgevoerd?

Zo ja, dan ligt het er aan wat je gebruikt om de metingen door te sturen. Is dat het standaard datalogger-script dat de v1-API gebruikt?

Milithor, 2020-01-29:
Ik heb volgende guide gevolgd om op de raspberry de logger te plaatsen: https://dsmr-reader.readthedocs.io/en/latest/installation/datalogger.html

Daarna heb ik om mijn VM de dsmreader gezet. En heb de VM meerdere keren herstart.
Maar ik denk omdat het een gesplitste setup is, dat het niet gaat met die instelling? 
De MQTT heb ik ook moeten verplaatsen van de logger naar de reader in deze opstelling.

dennissiemensma, 2020-01-29:
Je kunt de remote logger het beste zien als een verlengstuk van je seriele kabel. Voor DSMR-reader is het dan net alsof de applicatie direct aan de kabel hangt.

De reden dat ik het vraag is omdat het ook mogelijk is om de v2-API te gebruiken. Maar die accepteert alleen gegevens die al geparsed zijn. De v1-API gebruikt het ruwe telegram, dus alle instellingen zouden daar ook voor moeten werken. 

Wellicht kun je hier een telegram delen (of mailen naar github@ dennissiemensma.nl). Je kunt het laatste telegram vinden in je DSMR-reader op: `/admin/dsmr_datalogger/meterstatistics/1/change/` (onderaan de pagina).

Milithor, 2020-01-29:
Bij deze:

/FLU5\253769484_A

0-0:96.1.4(50213)
0-0:96.1.1(3153414733313030303530313836)
0-0:1.0.0(200129213012W)
1-0:1.8.1(000055.686*kWh)
1-0:1.8.2(000061.068*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.376*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.615*kW)
1-0:2.7.0(00.000*kW)
1-0:32.7.0(232.5*V)
1-0:52.7.0(229.8*V)
1-0:72.7.0(230.3*V)
1-0:31.7.0(000*A)
1-0:51.7.0(002*A)
1-0:71.7.0(000*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(37464C4F32313139303830313935)
0-1:24.4.0(1)
0-1:24.2.3(200129212505W)(00062.898*m3)
!38D5
Het meest recente uitgelezen telegram. N.B.: Alleen het laatste telegram wordt opgeslagen

dennissiemensma, 2020-01-29:
Als ik hem in een test laat parsen met Fluvius, maakt die er dit van:

```
timestamp                       2020-01-29 20:30:12+00:00
electricity_delivered_1         55.686
electricity_returned_1          0.000
electricity_delivered_2         61.068
electricity_returned_2          0.376
electricity_currently_delivered 0.615
electricity_currently_returned  0.000
phase_currently_delivered_l1    None
phase_currently_delivered_l2    None
phase_currently_delivered_l3    None
extra_device_timestamp          2020-01-29 20:25:05+00:00
extra_device_delivered          62.898
phase_currently_returned_l1     None
phase_currently_returned_l2     None
phase_currently_returned_l3     None
phase_voltage_l1                232.5
phase_voltage_l2                229.8
phase_voltage_l3                230.3
```

Als ik hem op DSMR v4/5 zet:
```
timestamp                       2020-01-29 20:30:12+00:00
electricity_delivered_1         55.686
electricity_returned_1          0.000
electricity_delivered_2         61.068
electricity_returned_2          0.376
electricity_currently_delivered 0.615
electricity_currently_returned  0.000
phase_currently_delivered_l1    None
phase_currently_delivered_l2    None
phase_currently_delivered_l3    None
extra_device_timestamp          None
extra_device_delivered          None
phase_currently_returned_l1     None
phase_currently_returned_l2     None
phase_currently_returned_l3     None
phase_voltage_l1                232.5
phase_voltage_l2                229.8
phase_voltage_l3                230.3
```

De gas-info is dan leeg bij bovenstaande (en dat komt door het verschil met DSMR v4/5). Dus de keuze voor Fluvius lijkt in ieder geval goed.

----

Toen ben ik weer wat verder gaan zoeken en weet ik weet wat het was. Het is puur een tekstueel verschil tussen NL en BE:
- 1: https://github.com/dennissiemensma/dsmr-reader/issues/750#issuecomment-549138360
- 2: https://github.com/dennissiemensma/dsmr-reader/issues/750#issuecomment-549036424

Ik kan het ook niet meer omdraaien omdat anderen er dan last van hebben. 

Als er genoeg animo voor is kan ik er hooguit een instelling van maakt die visueel omwisselt of iets. Het parsen zelf gaat dus gewoon goed. Het is puur dat in NL de tarieven een andere naam hebben dan in BE.

Milithor, 2020-01-29:
Oh right, wat lekker dat het weer zo goed gestandardiseerd is hiero ;-)
Dus eigenlijk als ik voor mezelf piek als dal hanteer en andersom, is het opgelost voor mezelf?

dennissiemensma, 2020-01-29:
Je kunt wellicht dubbelchecken of het tariefnummer overeen komt met de teller die nu oploopt. Hoe dan ook lijkt het me eerder een visueel iets.

Milithor, 2020-01-29:
1.8.1 is hier DAG tarief
1.8.2 is hier NACHT tarief

Voor jullie is dit andersom. Misschien ooit een idee om een country code mee te nemen, of gewoon waar je letterlijk de koppeling maakt met dag / nacht (piek / dal).

dennissiemensma, 2020-01-30:
Ik denk dat de makkelijkste oplossing uiteindelijk een variabele tekst is, waar iedereen zelf de namen van de tarieven kan aanpassen. Het is namelijk een tekstuele voorkeur/verschil.

Ik zal er een los issue van maken.

closed 2020-01-30 by dennissiemensma

dennissiemensma, 2020-02-18:
@Milithor ter info, in release v3.4 komt ondersteuning om zelf de namen in te stellen voor de tarieven (#858)