dsmrreader/dsmr-reader #857: InfluxDB integration

wasperen, 2020-01-30:
Hey --

Just wanted to check if you guys have ever thought about an integration with InfluxDB? Just like the MQTT tool, this would send every reading into an instance of InfluxDB.

I've now done the integration using MQTT and Node-RED but having a native integration is much cleaner.

If there is any interest, happy to dig into it and see if I can add something like that.

dennissiemensma, 2020-01-30:
I have no experience with Influx. What kind of integration would you expect? Simply duplicating the DSMR-readings using the default data structure of DSMR-reader?

I see there is a Python client: https://github.com/influxdata/influxdb-python
Usage seems simple enough:
```
>>> from influxdb import InfluxDBClient

>>> json_body = [
    {
        "measurement": "cpu_load_short",
        "tags": {
            "host": "server01",
            "region": "us-west"
        },
        "time": "2009-11-10T23:00:00Z",
        "fields": {
            "value": 0.64
        }
    }
]

>>> client = InfluxDBClient('localhost', 8086, 'root', 'root', 'example')

>>> client.create_database('example')

>>> client.write_points(json_body)
```

For reference, you could expect these kinds of fields and values provider by DSMR-reader:
```
timestamp                       2020-01-29 20:30:12+00:00
electricity_delivered_1         55.686
electricity_returned_1          0.000
electricity_delivered_2         61.068
electricity_returned_2          0.376
electricity_currently_delivered 0.615
electricity_currently_returned  0.000
phase_currently_delivered_l1    None
phase_currently_delivered_l2    None
phase_currently_delivered_l3    None
extra_device_timestamp          2020-01-29 20:25:05+00:00
extra_device_delivered          62.898
phase_currently_returned_l1     None
phase_currently_returned_l2     None
phase_currently_returned_l3     None
phase_voltage_l1                232.5
phase_voltage_l2                229.8
phase_voltage_l3                230.3
```

wasperen, 2020-01-30:
Yeah, it is pretty straight forward. Cool thing is that you then can use something like Grafana to do time series vizualizations and dashboards.If you want I can have a look to see what it would entice to implement such a thing.

>>> from influxdb import InfluxDBClient

>>> json_body = [
    {
        "measurement": "cpu_load_short",
        "tags": {
            "host": "server01",
            "region": "us-west"
        },
        "time": "2009-11-10T23:00:00Z",
        "fields": {
            "value": 0.64
        }
    }
]

>>> client = InfluxDBClient('localhost', 8086, 'root', 'root', 'example')

>>> client.create_database('example')

>>> client.write_points(json_body)

For reference, you could expect these kinds of fields and values provider by DSMR-reader:
timestamp                       2020-01-29 20:30:12+00:00
electricity_delivered_1         55.686
electricity_returned_1          0.000
electricity_delivered_2         61.068
electricity_returned_2          0.376
electricity_currently_delivered 0.615
electricity_currently_returned  0.000
phase_currently_delivered_l1    None
phase_currently_delivered_l2    None
phase_currently_delivered_l3    None
extra_device_timestamp          2020-01-29 20:25:05+00:00
extra_device_delivered          62.898
phase_currently_returned_l1     None
phase_currently_returned_l2     None
phase_currently_returned_l3     None
phase_voltage_l1                232.5
phase_voltage_l2                229.8
phase_voltage_l3                230.3

dennissiemensma, 2020-01-30:
You can try using a plugin in DSMR-reader first:

----

Install client:
```
sudo su - dsmr
pip3 install influxdb
```

Create file `dsmr_plugins/modules/forward_json_dsmrreading_to_influxdb.py` with content:
```
import json

from django.dispatch import receiver
from django.core import serializers
from django.utils import timezone
import django.db.models.signals
from influxdb import InfluxDBClient

from dsmr_datalogger.models.reading import DsmrReading

@receiver(django.db.models.signals.post_save, sender=DsmrReading)
def handle_forward_json_dsmrreading_to_influxdb(sender, instance, created, raw, **kwargs):
    if not created or raw:
        return

    instance.timestamp = timezone.localtime(instance.timestamp)

    if instance.extra_device_timestamp:
        instance.extra_device_timestamp = timezone.localtime(instance.extra_device_timestamp)

    serialized_reading = json.loads(serializers.serialize('json', [instance]))
    serialized_reading = serialized_reading[0]['fields']

    # ALTER THESE FOR YOUR LOCAL SETUP
    INFLUX_HOST = 'localhost'
    INFLUX_PORT = 8086
    INFLUX_USER = 'root'
    INFLUX_PASSWORD = 'pass'
    INFLUX_DATABASE = 'example'

    influx_body = [
        {
            "measurement": "dsmr_reading",
            "time": serialized_reading['timestamp'],
            "fields": serialized_reading,
        }
    ]

    # from pprint import pprint
    # pprint(influx_body, indent=4)

    try:
        client = InfluxDBClient(INFLUX_HOST, INFLUX_PORT, INFLUX_USER, INFLUX_PASSWORD, INFLUX_DATABASE)
        client.create_database(INFLUX_DATABASE)
        client.write_points(influx_body)
    except Exception as error:
        print('forward_json_dsmrreading_to_influxdb:', error)

```

Don't forget to alter these in the code above:
```
    INFLUX_HOST = 'localhost'
    INFLUX_PORT = 8086
    INFLUX_USER = 'root'
    INFLUX_PASSWORD = 'pass'
    INFLUX_DATABASE = 'example'
```

Add to dsmrreader/settings.py:
```
DSMRREADER_PLUGINS = [
    'dsmr_plugins.modules.forward_json_dsmrreading_to_influxdb',
]
```

Restart processes to apply changes:

```
sudo su - dsmr
./reload.sh
```

----

Data sent to Influx (see `pprint(influx_body, indent=4)`):

```
[   {   'fields': {   'electricity_currently_delivered': '2.300',
                      'electricity_currently_returned': '0.000',
                      'electricity_delivered_1': '16034.364',
                      'electricity_delivered_2': '9620.619',
                      'electricity_returned_1': '0.000',
                      'electricity_returned_2': 0,
                      'extra_device_delivered': None,
                      'extra_device_timestamp': None,
                      'phase_currently_delivered_l1': '0.081',
                      'phase_currently_delivered_l2': '0.871',
                      'phase_currently_delivered_l3': '1.348',
                      'phase_currently_returned_l1': '0.000',
                      'phase_currently_returned_l2': '0.000',
                      'phase_currently_returned_l3': '0.000',
                      'phase_voltage_l1': '233.0',
                      'phase_voltage_l2': '231.1',
                      'phase_voltage_l3': '230.2',
                      'processed': False,
                      'timestamp': '2020-01-30T20:53:55+01:00'},
        'measurement': 'dsmr_reading',
        'time': '2020-01-30T20:53:55+01:00'}]
```

I'm not sure if the structure above is sufficient. I don't have Influxdb running locally so I can't test.

---

I'm not sure if `client.create_database(INFLUX_DATABASE)` is needed, you can try removing the line if the database already exists.

wasperen, 2020-01-31:
This looks as if it would work. Couple of points.

The values are now posted to InfluxDB as strings. That is one thing I do in my current MQTT->NodeRED->InfluxDB flow: parse these strings into proper float / int values.

Another thing is that it creates a new InfluxDB client for every single instance, yet without properly closing it. With login and other handshakes that may be quite a lot of overhead. With a plug-in, is there a way to create a singleton object at the start of the process?

Finally, I would like to think about splitting the stream into different measurements. Something like:

```
electricity_current:
  electricity_currently_delivered,
  electricity_currently_returned,
  phase_currently_delivered_l1,
  phase_currently_delivered_l2,
  phase_currently_delivered_l3,
  phase_currently_returned_l1,
  phase_currently_returned_l2,
  phase_currently_returned_l3

electricity_present:
  electricity_delivered_1,
  electricity_delivered_2,
  electricity_returned_1,
  electricity_returned_2

electricity_voltage:
  phase_voltage_l1.
  phase_voltage_l2,
  phase_voltage_l3
```

Would `extra_device_delivered` be gas?

dennissiemensma, 2020-01-31:
Thanks for the response. I should explain that the plugin is just a proof of concept and nothing definitive.

If passing the values into Influx work, I could look into a more fitting integration, probably similar to MQTT in DSMR-reader.

Regarding splitting the messurements, I could probably use a similar config/mapping as currently used in MQTT.

wasperen, 2020-01-31:
I understand. Happy to give it a go. Will revert back.

wasperen, 2020-02-15:
This works like a charm.

There is a number of things that I would like to do.
* send the data in the respective data type (rather than just string)
* add an indicator low/high to show if the measurement was against low or high tariff
* add the monetary value of the consumption
* gas readings in a separate measurement

Happy to put some effort into this. Just let me know what works for you.

dennissiemensma, 2020-02-15:
Thanks for trying and  the feedback. Any help in testing, later on, would be great.

dfkt80, 2020-02-19:
I would be very interested in this as well! Also using Influxdb & Grafana voor data visualization. Is it possible to run this with just the dsmr datalogger?

dennissiemensma, 2020-02-19:
The datalogger only reads telegrams and does not parse them.

dennissiemensma, 2020-07-14:
Het heeft even geduurd, maar in ben inmiddels bezig met de integratie voor InfluxDB. Het komt dan samen met MQTT (wanneer ingeschakeld) te draaien in een apart proces. 

Wanneer ik iets heb dat getest kan worden kom ik er op terug. Voorwaarde is sowieso dat het pas vanaf DSMR-reader v4.x mogelijk is, dus die upgrade is dan daarmee wel verplicht.

wasperen, 2020-07-14:
Top. Helemaal goed. Ben benieuwd.

dennissiemensma, 2020-07-14:
In de kern werkt het nu en ook met behoud van de native types. Het instellen en mappen van de measurements kan straks geheel naar eigen wens. De default is zoiets:

```
### [measurement_name]
### DSMR-reader field 1 = InfluxDB field 1
### DSMR-reader field 2 = InfluxDB field 2

[electricity_live]
electricity_currently_delivered = currently_delivered
electricity_currently_returned = currently_returned

[electricity_positions]
electricity_delivered_1 = delivered_1
electricity_returned_1 = returned_1
electricity_delivered_2 = delivered_2
electricity_returned_2 = returned_2

[electricity_voltage]
phase_voltage_l1 = phase_l1
phase_voltage_l2 = phase_l2
phase_voltage_l3 = phase_l3

[electricity_phases]
phase_currently_delivered_l1 = currently_delivered_l1
phase_currently_delivered_l2 = currently_delivered_l2
phase_currently_delivered_l3 = currently_delivered_l3
phase_currently_returned_l1 = currently_returned_l1
phase_currently_returned_l2 = currently_returned_l2
phase_currently_returned_l3 = currently_returned_l3

[electricity_power]
phase_power_current_l1 = current_l1
phase_power_current_l2 = current_l2
phase_power_current_l3 = current_l3

[gas_positions]
extra_device_delivered = delivered
```

Een test met fictieve nep-data maakt er dan dit van:
```
> select * from gas_positions
name: gas_positions
time                delivered
----                ---------
1594763236000000000 5240.786
1594763237000000000 5240.786
1594763238000000000 5240.786
1594763880000000000 5240.805
1594763881000000000 5240.805
1594763882000000000 5240.805
1594763883000000000 5240.805
> select * from electricity_power
name: electricity_power
time                current_l1 current_l2 current_l3
----                ---------- ---------- ----------
1594763236000000000 5          3          1
1594763237000000000 7          2          1
1594763238000000000 5          2          0
1594763880000000000 5          2          1
1594763881000000000 5          2          0
1594763882000000000 6          2          0
1594763883000000000 8          3          0
> select * from electricity_phases
name: electricity_phases
time                currently_delivered_l1 currently_delivered_l2 currently_delivered_l3 currently_returned_l1 currently_returned_l2 currently_returned_l3
----                ---------------------- ---------------------- ---------------------- --------------------- --------------------- ---------------------
1594763236000000000 1.297                  1.375                  0.664                  0                     0                     0
1594763237000000000 1.401                  1.23                   0.48                   0                     0                     0
1594763238000000000 0.09                   1.375                  0.878                  0                     0                     0
1594763880000000000 0                      0                      0                      0.881                 0.465                 0.964
1594763881000000000 0                      0                      0                      0.306                 1.135                 0.319
1594763882000000000 0                      0                      0                      0.097                 1.264                 0.639
1594763883000000000 0                      0                      0                      0.867                 0.087                 0.602
> select * from electricity_voltage
name: electricity_voltage
time                phase_l1 phase_l2 phase_l3
----                -------- -------- --------
1594763236000000000 231      228.1    232.2
1594763237000000000 230      229.1    231.2
1594763238000000000 231      228.1    229.2
1594763880000000000 230      231.1    231.2
1594763881000000000 232      228.1    232.2
1594763882000000000 228      230.1    232.2
1594763883000000000 230      228.1    230.2
> select * from electricity_positions
name: electricity_positions
time                delivered_1 delivered_2 returned_1 returned_2
----                ----------- ----------- ---------- ----------
1594763236000000000 17469.284   10481.571   0          0
1594763237000000000 17469.284   10481.571   0          0
1594763238000000000 17469.284   10481.571   0          0
1594763880000000000 17469.348   10481.609   0          0
1594763881000000000 17469.349   10481.609   0          0
1594763882000000000 17469.349   10481.609   0          0
1594763883000000000 17469.349   10481.609   0          0
> select * from electricity_live
name: electricity_live
time                currently_delivered currently_returned
----                ------------------- ------------------
1594743473472203000 1.234               0
1594743474480947000 1.234               0
1594743475489910000 1.234               0
1594757797506441044 1.234               0
1594757809847693880 1.234               0
1594757810855500111 1.234               0
1594757811862520890 1.234               0
1594763236000000000 3.336               0
1594763237000000000 3.109               0
1594763238000000000 2.343               0
1594763880000000000 0                   2.31
1594763881000000000 0                   1.76
1594763882000000000 0                   1.999
1594763883000000000 0                   1.555
```

wasperen, 2020-07-15:
Gaaf -- is er al code die ik kan draaien?

> select * from gas_positions
name: gas_positions
time                delivered
----                ---------
1594763236000000000 5240.786
1594763237000000000 5240.786
1594763238000000000 5240.786
1594763880000000000 5240.805
1594763881000000000 5240.805
1594763882000000000 5240.805
1594763883000000000 5240.805
> select * from electricity_power
name: electricity_power
time                current_l1 current_l2 current_l3
----                ---------- ---------- ----------
1594763236000000000 5          3          1
1594763237000000000 7          2          1
1594763238000000000 5          2          0
1594763880000000000 5          2          1
1594763881000000000 5          2          0
1594763882000000000 6          2          0
1594763883000000000 8          3          0
> select * from electricity_phases
name: electricity_phases
time                currently_delivered_l1 currently_delivered_l2 currently_delivered_l3 currently_returned_l1 currently_returned_l2 currently_returned_l3
----                ---------------------- ---------------------- ---------------------- --------------------- --------------------- ---------------------
1594763236000000000 1.297                  1.375                  0.664                  0                     0                     0
1594763237000000000 1.401                  1.23                   0.48                   0                     0                     0
1594763238000000000 0.09                   1.375                  0.878                  0                     0                     0
1594763880000000000 0                      0                      0                      0.881                 0.465                 0.964
1594763881000000000 0                      0                      0                      0.306                 1.135                 0.319
1594763882000000000 0                      0                      0                      0.097                 1.264                 0.639
1594763883000000000 0                      0                      0                      0.867                 0.087                 0.602
> select * from electricity_voltage
name: electricity_voltage
time                phase_l1 phase_l2 phase_l3
----                -------- -------- --------
1594763236000000000 231      228.1    232.2
1594763237000000000 230      229.1    231.2
1594763238000000000 231      228.1    229.2
1594763880000000000 230      231.1    231.2
1594763881000000000 232      228.1    232.2
1594763882000000000 228      230.1    232.2
1594763883000000000 230      228.1    230.2
> select * from electricity_positions
name: electricity_positions
time                delivered_1 delivered_2 returned_1 returned_2
----                ----------- ----------- ---------- ----------
1594763236000000000 17469.284   10481.571   0          0
1594763237000000000 17469.284   10481.571   0          0
1594763238000000000 17469.284   10481.571   0          0
1594763880000000000 17469.348   10481.609   0          0
1594763881000000000 17469.349   10481.609   0          0
1594763882000000000 17469.349   10481.609   0          0
1594763883000000000 17469.349   10481.609   0          0
> select * from electricity_live
name: electricity_live
time                currently_delivered currently_returned
----                ------------------- ------------------
1594743473472203000 1.234               0
1594743474480947000 1.234               0
1594743475489910000 1.234               0
1594757797506441044 1.234               0
1594757809847693880 1.234               0
1594757810855500111 1.234               0
1594757811862520890 1.234               0
1594763236000000000 3.336               0
1594763237000000000 3.109               0
1594763238000000000 2.343               0
1594763880000000000 0                   2.31
1594763881000000000 0                   1.76
1594763882000000000 0                   1.999
1594763883000000000 0                   1.555

dennissiemensma, 2020-07-15:
Wellicht in de loop van volgende week, maar ik kom daar nog wel op terug.

dennissiemensma, 2020-07-21:
@wasperen (e.a.) wat mij betreft kun je het proberen. 

Het is echter niet een kwestie van alleen een 'branch wisselen' zoals ik dat soms wel voor andere features kon doen. De InfluxDB-integratie is onderdeel van DSMR-reader v4.0, die nog niet officieel uit is, maar sinds vanavond wel als een schaduw-release. 
Deze draai ik zelf al enkele dagen en het lijkt allemaal prima (vooral de bestaande functionaliteit), maar ik ben geen gebruiker van InfluxDB, dus het is voor mij lastig zeggen of het afdoende is. 

Als je InfluxDB wilt testen, zul je dus moeten upgraden naar DSMR-reader v4 en daar zit (wat) eenmalig handwerk bij.

N.B.:
- Zorg dat je een (recente) databasebackup hebt.
- Upgrade-stappen staan hier: https://dsmr-reader.readthedocs.io/nl/v4/faq/v4_upgrade.html
- Je kunt dit uitvoeren en vervolgens test wanneer je wilt, maar ik kan je alleen in de avonduren assisteren bij issues of feedback.
- Hoewel de InfluxDB-integratie goed lijkt te werken, kan het zijn dat foutafhandeling nog niet perfect is. Een eerste plek om te kijken bij fouten zijn de logs, zoals beschreven in: https://dsmr-reader.readthedocs.io/nl/v4/troubleshooting.html
- Na upgrade naar v4(rc1) kun je de instellingen beheren in de admin als je daar (of in de tussenpagina) zoekt op 'influxdb'.

wasperen, 2020-07-22:
Top. Weekendklusje!

Zorg dat je een (recente) databasebackup hebt.Upgrade-stappen staan hier: https://dsmr-reader.readthedocs.io/nl/v4/faq/v4_upgrade.htmlJe kunt dit uitvoeren en vervolgens test wanneer je wilt, maar ik kan je alleen in de avonduren assisteren bij issues of feedback.Hoewel de InfluxDB-integratie goed lijkt te werken, kan het zijn dat foutafhandeling nog niet perfect is. Een eerste plek om te kijken bij fouten zijn de logs, zoals beschreven in: https://dsmr-reader.readthedocs.io/nl/v4/troubleshooting.htmlNa upgrade naar v4(rc1) kun je de instellingen beheren in de admin als je daar (of in de tussenpagina) zoekt op 'influxdb'.

slashw, 2020-07-23:
Dennis,
Ik bevestigen dat deze migratie werkt. 
Volgens duidelijke procedure gemigreerd naar v4.0rc1, daarna Influx geactiveerd.
Nu kijk ik naar een werkend Grafana met data van DSMR-reader
DSMR-reader werkt ook verder naar behoren, alle data van 3.11 is mee gemigreerd.
Alleen jammer dat Grafana alleen vanaf nu pas registreerd.
Top, dankjewel voor deze migratie naar Infuxdb.

Met vriendelijke groet,
Wim

dennissiemensma, 2020-07-23:
@slashw bedankt voor het proberen en de terugkoppeling!

Voor bestaande data zal ik #1055 toevoegen. Ik kan niet beloven dat het goed werkt in de praktijk, maar het is een relatief kleine moeite en als het toch werkt, dan is dat mooi meegenomen. Volg dus #1055 even verder.

dennissiemensma, 2020-07-27:
https://dsmr-reader.readthedocs.io/en/v4/faq/v4_upgrade.html

dennissiemensma, 2020-07-27:
Zit nu in v4. Bugs en aanvullingen kunnen het beste via een vervolg-issue.

closed 2020-07-27 by dennissiemensma