dsmrreader/dsmr-reader #859: Kan vanuit home assistant op raspberry DSMR niet gebruiken.

ovromans, 2020-01-31:
Ik gebruik een RaspberryPi 3 met een schone installatie van raspbian en home assistant en gebruikt
DSMR-protocol slimme meter v5.

Met deze RaspberryPi maak ik een verbinding met een andere RaspberryPI die aangesloten is op de slimme meter en deze dus uitleest.

Hieronder heb ik het stukje uit het configuraton bestand:
sensor:
  - platform: dsmr
    host: 192.168.178.23
    port: 2001
    dsmr_version: 5

Ik krijg geen gegevens te zien en in de logging staat:
Unable to install package dsmr_parser==0.12: Could not find a version that satisfies the requirement PyCRC<2,>=1.2 (from dsmr_parser==0.12) (from versions: 0.9.2)
No matching distribution found for PyCRC<2,>=1.2 (from dsmr_parser==0.12)

Ik heb een tijdje het os gebruikt van home assistant zelf gebruikt, en met dit os werkte het uitlezen van de slimme meter wel. Ook gebruik ik hetzelfde configuratie bestand.

dennissiemensma, 2020-01-31:
Bedankt voor je vraag. Kan het zijn dat je bij het verkeerde project zit? Ik lees nergens in je verhaal iets over DSMR-reader, alleen over Home Assistant en een plugin daarvan die 'DSMR' heet.

Wellicht moet je hier zijn: https://github.com/home-assistant/home-assistant

ovromans, 2020-02-01:
Je hebt helemaal gelijk, ik zit in het verkeerde plek. Mijn vraag gaat inderdaad over de plugin 'DSMR' en niet over de DSRM-reader. Heb zelf niet goed gezocht.

closed 2020-02-01 by ovromans