dsmrreader/dsmr-reader #861: exited: dsmr_mqtt (exit status 0; expected)

SaturnusDJ, 2020-02-02:
Sinds v3.1.0 de log vol met:
```
2020-02-02 10:36:39,119 INFO success: dsmr_mqtt entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-02-02 10:36:46,046 INFO exited: dsmr_mqtt (exit status 0; expected)
2020-02-02 10:36:46,056 INFO spawned: 'dsmr_mqtt' with pid 42
```

Of is dit de workaround voor https://github.com/dennissiemensma/dsmr-reader/issues/833 in werking?

dennissiemensma, 2020-02-02:
Het hangt er denk ik vanaf of je uberhaupt MQTT gebruikt. Is het in jouw situatie wel of niet de bedoeling dat het proces draait?

SaturnusDJ, 2020-02-02:
Wel de bedoeling, ja. MQTT lijkt ook prima te werken als ik het vanuit Home Assistant bekijk.

dennissiemensma, 2020-02-02:
Ik denk dat je dan het beste even debug-logging aan kan zetten en kijken wat die exact doet: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging

SaturnusDJ, 2020-02-02:
```
[2020-02-02 18:36:08,735] DEBUG    MQTT: Queued message for dsmr/reading/electricity_currently_delivered
2020-02-02 17:36:08,857 DEBG 'dsmr_datalogger' stderr output:
[2020-02-02 18:36:08,855] DEBUG    MQTT: Queued message for dsmr/day-consumption/electricity_merged
2020-02-02 17:36:08,884 DEBG 'dsmr_datalogger' stderr output:
[2020-02-02 18:36:08,882] DEBUG    MQTT: Queued message for dsmr/day-consumption/electricity_cost_merged
2020-02-02 17:36:08,897 DEBG 'dsmr_datalogger' stderr output:
[2020-02-02 18:36:08,895] DEBUG    MQTT: Ignoring dsmr/day-consumption/gas due to cache
2020-02-02 17:36:08,902 DEBG 'dsmr_datalogger' stderr output:
[2020-02-02 18:36:08,901] DEBUG    MQTT: Ignoring dsmr/day-consumption/gas_cost due to cache
2020-02-02 17:36:08,951 DEBG 'dsmr_datalogger' stderr output:
[2020-02-02 18:36:08,948] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 30.0s
2020-02-02 17:36:09,031 DEBG 'dsmr_mqtt' stderr output:
[2020-02-02 18:36:09,030] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending DISCONNECT
2020-02-02 17:36:09,033 DEBG 'dsmr_mqtt' stderr output:
[2020-02-02 18:36:09,031] DEBUG    MQTT: Paho client: on_disconnect(userdata, rc) None 0
2020-02-02 17:36:09,034 DEBG 'dsmr_mqtt' stderr output:
[2020-02-02 18:36:09,032] DEBUG    dsmr_mqtt.management.commands.dsmr_mqtt: Exited
2020-02-02 17:36:09,649 DEBG fd 15 closed, stopped monitoring <POutputDispatcher at 1979923520 for <Subprocess at 1979923496 with name dsmr_mqtt in state RUNNING> (stdout)>
2020-02-02 17:36:09,650 DEBG fd 19 closed, stopped monitoring <POutputDispatcher at 1979511544 for <Subprocess at 1979923496 with name dsmr_mqtt in state RUNNING> (stderr)>
2020-02-02 17:36:09,651 INFO exited: dsmr_mqtt (exit status 0; expected)
2020-02-02 17:36:09,653 DEBG received SIGCHLD indicating a child quit
2020-02-02 17:36:10,663 INFO spawned: 'dsmr_mqtt' with pid 36
2020-02-02 17:36:11,668 INFO success: dsmr_mqtt entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-02-02 17:36:16,827 DEBG 'dsmr_mqtt' stderr output:
[2020-02-02 18:36:16,826] DEBUG    MQTT: Initializing insecure connection (no TLS)
2020-02-02 17:36:16,830 DEBG 'dsmr_mqtt' stderr output:
[2020-02-02 18:36:16,829] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b'DSMR-reader'
2020-02-02 17:36:16,833 DEBG 'dsmr_mqtt' stderr output:
[2020-02-02 18:36:16,832] DEBUG    dsmr_mqtt.management.commands.dsmr_mqtt: Starting infinite command loop...
2020-02-02 17:36:16,852 DEBG 'dsmr_mqtt' stderr output:
[2020-02-02 18:36:16,851] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m1), 'b'dsmr/day-consumption/electricity_cost_merged'', ... (4 bytes)
2020-02-02 17:36:16,855 DEBG 'dsmr_mqtt' stderr output:
[2020-02-02 18:36:16,852] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 1
2020-02-02 17:36:16,921 DEBG 'dsmr_mqtt' stderr output:
[2020-02-02 18:36:16,920] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m2), 'b'dsmr/day-consumption/electricity_merged'', ... (5 bytes)
2020-02-02 17:36:16,924 DEBG 'dsmr_mqtt' stderr output:
[2020-02-02 18:36:16,921] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 2
2020-02-02 17:36:16,935 DEBG 'dsmr_mqtt' stderr output:
[2020-02-02 18:36:16,934] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m3), 'b'dsmr/reading/electricity_currently_delivered'', ... (5 bytes)
2020-02-02 17:36:16,936 DEBG 'dsmr_mqtt' stderr output:
[2020-02-02 18:36:16,935] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 3
2020-02-02 17:36:16,950 DEBG 'dsmr_mqtt' stderr output:
[2020-02-02 18:36:16,949] INFO     dsmr_mqtt.management.commands.dsmr_mqtt: [i] Detected StopInfiniteRun exception
2020-02-02 17:36:16,952 DEBG 'dsmr_mqtt' stderr output:
[2020-02-02 18:36:16,950] INFO     dsmr_mqtt.management.commands.dsmr_mqtt: Exiting on next run...
[2020-02-02 18:36:16,950] DEBUG    dsmr_mqtt.management.commands.dsmr_mqtt: Sleeping 30.0s
2020-02-02 17:36:39,388 DEBG 'dsmr_datalogger' stderr output:
[2020-02-02 18:36:39,387] INFO     
```
`--- telegram verwijderd ---`
```
2020-02-02 17:36:39,542 DEBG 'dsmr_datalogger' stderr output:
[2020-02-02 18:36:39,541] DEBUG    MQTT: Queued message for dsmr/reading/electricity_currently_delivered
2020-02-02 17:36:39,646 DEBG 'dsmr_datalogger' stderr output:
[2020-02-02 18:36:39,645] DEBUG    MQTT: Ignoring dsmr/day-consumption/electricity_merged due to cache
2020-02-02 17:36:39,652 DEBG 'dsmr_datalogger' stderr output:
[2020-02-02 18:36:39,651] DEBUG    MQTT: Ignoring dsmr/day-consumption/electricity_cost_merged due to cache
2020-02-02 17:36:39,657 DEBG 'dsmr_datalogger' stderr output:
[2020-02-02 18:36:39,656] DEBUG    MQTT: Ignoring dsmr/day-consumption/gas due to cache
2020-02-02 17:36:39,666 DEBG 'dsmr_datalogger' stderr output:
[2020-02-02 18:36:39,661] DEBUG    MQTT: Ignoring dsmr/day-consumption/gas_cost due to cache
2020-02-02 17:36:39,694 DEBG 'dsmr_datalogger' stderr output:
[2020-02-02 18:36:39,692] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 30.0s
2020-02-02 17:36:47,053 DEBG 'dsmr_mqtt' stderr output:
[2020-02-02 18:36:47,051] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending DISCONNECT
2020-02-02 17:36:47,055 DEBG 'dsmr_mqtt' stderr output:
[... cut ...]
2020-02-02 17:36:47,708 DEBG fd 21 closed, stopped monitoring <POutputDispatcher at 1979511496 for <Subprocess at 1979923496 with name dsmr_mqtt in state RUNNING> (stderr)>
2020-02-02 17:36:47,710 INFO exited: dsmr_mqtt (exit status 0; expected)
2020-02-02 17:36:47,712 DEBG received SIGCHLD indicating a child quit
2020-02-02 17:36:48,721 INFO spawned: 'dsmr_mqtt' with pid 38
2020-02-02 17:36:49,725 INFO success: dsmr_mqtt entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
```

dennissiemensma, 2020-02-03:
Bedankt voor het delen. Ik heb er geen directe verklaring voor.

In v3.1 is een check toegevoegd die telkens kijkt of de MQTT-client nog een verbinding heeft. Zo niet dan zorgt die ervoor dat het proces zichzelf herstart.

Ik kan wel even een branch voor je maken met wat extra debugging output om er zeker van te zijn dat dat het is.

dennissiemensma, 2020-02-03:
Ik heb afgetakt van v3.2, dus als je die versie nog niet hebt, kun je het beste eerst upgraden. Als je dat niet wil, dan kan ik eventueel een aftakking maken van v3.1.

Als je op v3.2 zit, kun je dit doen:
```
sudo su - dsmr
git fetch
git checkout -b debug/861-mqtt-debug origin/debug/861-mqtt-debug
./post-deploy.sh
```

Ik ben vervolgens benieuwd of je het volgende in de debug log ziet:
```
mqtt_client.is_connected() is False, restarting process...
```

SaturnusDJ, 2020-02-03:
Dank. Ik bedacht mij nog dat de in HA geïntegreerde MQTT de oorzaak zou kunnen zijn. Die is namelijk behoorlijk brak. Ik ga eerst Mosquitto werkend zien te krijgen (was sowieso een item op de todo list) om te zien of dat een oplossing is.

SaturnusDJ, 2020-02-04:
Wel...het lijkt inderdaad de brakke MQTT in HA te zijn geweest. Sinds het verbinden met Mosquitto (40 min. terug) geen exits meer.

closed 2020-02-04 by SaturnusDJ

dennissiemensma, 2020-02-05:
Bedankt voor het terugkoppelen en mooi dat dat het oplost!