dsmrreader/dsmr-reader #863: Exchange of electricity and/or gas meeter.

Piet-PvdZ, 2020-02-03:
Recently my electricity meter was changed du to bad RF connections to the Stedin network.
The new meter starts at 00000 but DSMR dit not detect the 0000 value and calculated a huge negative value. The graphs are destroyed du to the large negative number.
A suggestion:
Check the meter serial number from the telegram data and if this is changed ignore the huge negative usage. Or check for negative results and handle accordingly.
Running now DSMR V3.1.0

dennissiemensma, 2020-02-03:
Bedankt voor je melding. Dit is een bekend issue en er staat al een ticket voor om het ooit op te lossen: #507

Voor nu kun je het beste na afloop van de dag handmatig de dagtotalen bijwerken. Dit hoef je maar eenmalig te doen als het goed is. Zie bovenstaand issue voor meer info.

closed 2020-02-03 by dennissiemensma

Piet-PvdZ, 2020-02-03:
Dennis.Dank voor je reactie.Had het ticket #507 niet opgemerkt.Ik wacht een mogelijke oplossing wel af. Hoop dat Stedin niet nogeens komt voor vervanging.Groet.  Piet van der Zanden.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/dennissiemensma/dsmr-reader/issues/863?email_source=notifications\u0026email_token=AOAM7UDP7R34F753LMCFL7TRBBWI3A5CNFSM4KPHYQS2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEKVDEOI#issuecomment-581579321",
"url": "https://github.com/dennissiemensma/dsmr-reader/issues/863?email_source=notifications\u0026email_token=AOAM7UDP7R34F753LMCFL7TRBBWI3A5CNFSM4KPHYQS2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEKVDEOI#issuecomment-581579321",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]