dsmrreader/dsmr-reader #865: Upgrade tot v3.2 segfault

Spelvoud, 2020-02-04:
**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v3.2`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: ` NUC Ubuntu 19.10`
* DSMR-protocol slimme meter *(v4 of v5)*: `v5`

Latest update to V3.2 geeft een segfault aan het einde van de update (lijkt allemaal wel te werken)
Als ik de laaste regel uit het post-process script uitvoer krijg ik dezelfde foutmelding:

_(dsmrreader) dsmr@server:~/dsmr-reader$ ./manage.py dsmr_frontend_clear_cache
Segmentation fault (core dumped)_

dennissiemensma, 2020-02-04:
Bedankt voor je melding. Welke Python-versie gebruik je?

```
sudo su - dsmr
python3 --version
```

Spelvoud, 2020-02-04:
Python 3.7.3

dennissiemensma, 2020-02-05:
Dat is jammer, want ik draai dezelfde versie op m'n Pi en daar crasht die helaas niet.

Kun je eens kijken met `strace` wat de laatste system calls zijn voordat die crasht?

```
sudo su - dsmr
strace -vf ./manage.py dsmr_frontend_clear_cache
```

Spelvoud, 2020-02-05:
Hier de laatste regels:

clock_getres(CLOCK_MONOTONIC, {tv_sec=0, tv_nsec=1}) = 0
epoll_create1(EPOLL_CLOEXEC)            = 5
socketpair(AF_UNIX, SOCK_STREAM|SOCK_CLOEXEC, 0, [6, 7]) = 0
ioctl(6, FIONBIO, [1])                  = 0
ioctl(7, FIONBIO, [1])                  = 0
epoll_ctl(5, EPOLL_CTL_ADD, 6, {EPOLLIN, {u32=6, u64=139981574111238}}) = 0
close(4)                                = 0
close(3)                                = 0
rt_sigaction(SIGINT, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER,
sa_restorer=0x7f509f9d9470}, {sa_handler=0x5f8820, sa_mask=[],
sa_flags=SA_RESTORER, sa_restorer=0x7f509f9d9470}, 8) = 0mmap(NULL, 262144,
PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f5099465000
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x8} ---
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault (core dumped)

dennissiemensma, 2020-02-06:
Ik moet toegeven dat het mij toch helaas te weinig zegt. Ik had er op gehoopt dat het eindige met een systemcall naar een specifieke Python-file of iets gerelateerd aan cache, maar hier heb ik te weinig ervaring mee.

Begrijp ik het goed dat de applicatie verder wel gewoon goed draait? En bij eerdere versies ging het dus niet fout?

Spelvoud, 2020-02-06:
Ik draai al een jaar je prachtige programma en volg altijd netjes de
updates, laatst naar de V3 branch geüpdate en nu bij de tweede keer updaten
deze melding, daarvoor nooit een probleem.
Er is ook geen gevolg volg het functioneren, programma draait perfect, maar
het leek me netjes om te melden.

dennissiemensma, 2020-02-06:
Bedankt voor de info. Dan ben ik geneigd om dit even te laten voor wat het is. 

Segfaults zijn lastig op te lossen, helemaal wanneer ze niet makkelijk te reproduceren zijn (voor mij). Dan zul je namelijk oudere releases moeten draaien om te kijken wat wel en wat niet werkt en daarna eventuele branches waar telkens iets in of uitgeschakeld is. Als de applicatie verder goed werkt, doe ik je dat niet aan.

Houd het vooral in de gaten of je nog andere issues hebt. Dan kunnen we altijd alsnog debuggen.

closed 2020-02-06 by dennissiemensma