dsmrreader/dsmr-reader #866: DETAIL:  Key (id)=(118) already exists.

kropveld, 2020-02-04:
Ik heb problemen gehad met de upgrade, en uiteindelijk een de-installatie gedaan, en vervolgend weer een (her-)installatie. Maar toch krijg ik deze melding:

`sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/dsmr-reader/manage.py migrate
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backend, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_frontend, dsmr_mindergas, dsmr_mqtt, dsmr_notification, dsmr_pvoutput, dsmr_stats, dsmr_weather, sessions
Running migrations:
  Applying dsmr_backend.0009_update_email_settings_mail_from...Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 86, in _execute
    return self.cursor.execute(sql, params)
psycopg2.errors.UniqueViolation: duplicate key value violates unique constraint "django_migrations_pkey"
DETAIL:  Key (id)=(118) already exists.
(...)
`
Hoe verder, tips? Meer informatie?

dennissiemensma, 2020-02-04:
Bedankt voor je melding, wat is hiervan de output?

```
sudo su - dsmr
./manage.py showmigrations
```

kropveld, 2020-02-05:
Het draait op een Pi3B+ onder Raspbian GNU/Linux 10 (buster).  (met recente updates)

```
admin
 [X] 0001_initial
 [X] 0002_logentry_remove_auto_add
 [X] 0003_logentry_add_action_flag_choices
auth
 [X] 0001_initial
 [X] 0002_alter_permission_name_max_length
 [X] 0003_alter_user_email_max_length
 [X] 0004_alter_user_username_opts
 [X] 0005_alter_user_last_login_null
 [X] 0006_require_contenttypes_0002
 [X] 0007_alter_validators_add_error_messages
 [X] 0008_alter_user_username_max_length
 [X] 0009_alter_user_last_name_max_length
 [X] 0010_alter_group_name_max_length
 [X] 0011_update_proxy_permissions
contenttypes
 [X] 0001_initial
 [X] 0002_remove_content_type_name
dsmr_api
 [X] 0001_api_settings
 [X] 0002_generate_random_auth_key
 [X] 0003_create_api_user
dsmr_backend
 [X] 0001_backend_language
 [X] 0002_email_settings
 [X] 0003_scheduled_processes
 [X] 0004_backend_schedule_process_active_flag
 [X] 0005_schedule_auto_update_check
 [X] 0006_backend_auto_update_check
 [X] 0007_schedule_stats_generator
 [X] 0008_scheduled_process_fields
 [ ] 0009_update_email_settings_mail_from
 [ ] 0010_process_sleep
 [ ] 0011_disable_capabilities
 [ ] 0012_fix_email_from
dsmr_backup
 [X] 0001_initial
 [X] 0002_settings_documentation
 [X] 0003_dropbox_sync_field
 [X] 0004_custom_backup_path
 [X] 0005_email_backup_settings
 [X] 0006_scheduled_email_backup
 [X] 0007_remove_backupsettings_compress
dsmr_consumption
 [X] 0001_squashed_0004_recalculate_gas_consumption
 [X] 0002_verbose_text
 [X] 0003_electricity_consumption_indexes
 [X] 0004_merge_electricity_tariffs
 [X] 0005_phase_currently_delivered
 [X] 0006_dsmr_firmware_v5
 [X] 0007_settings_documentation
 [X] 0008_energysupplier_return_prices
 [X] 0009_consumption_settings_text
 [X] 0010_phases_currently_returned
 [X] 0011_track_phase_voltages
 [X] 0012_add_voltage_indexes
 [ ] 0013_schedule_generate_consumption
 [ ] 0014_gas_grouping
dsmr_datalogger
 [X] 0001_squashed_0005_optional_gas_readings
 [X] 0002_meta_names
 [X] 0003_telegram_checksum
 [X] 0004_phase_currently_delivered
 [X] 0005_verify_telegram_crc_setting
 [X] 0006_dsmr_firmware_v5
 [X] 0007_dsmrreading_timestamp_index
 [X] 0008_dsmrreading_help_text
 [X] 0009_retention_settings
 [X] 0010_phases_currently_returned
 [X] 0011_raw_telegram_insight
 [X] 0012_track_phase_voltages
 [X] 0013_remove_datalogger_track_settings
 [X] 0014_remove_optional_crc_check
 [X] 0015_datalogger_foreign_countries
 [ ] 0016_process_sleep
 [ ] 0017_log_telegrams
 [ ] 0018_meterstatistics_no_autonow
dsmr_frontend
 [X] 0001_initial
 [X] 0002_recent_history_weeks
 [X] 0003_drop_history_page
 [X] 0004_chart_colors
 [X] 0005_notifications
 [X] 0006_notifications_initial
 [X] 0007_merge_electricity_tariffs
 [X] 0008_merge_electricity_tariffs_notification
 [X] 0009_docs_no_reverse_match
 [X] 0010_drop_reverse_dashboard_graphs_setting
 [X] 0011_phase_currently_delivered
 [X] 0012_frontendsettings_dashboard_graph_width
 [X] 0013_echart_graphs_message
 [X] 0014_phases_currently_returned
 [X] 0015_notification_meta
 [X] 0016_v200_release
 [X] 0017_v201_release
 [X] 0018_v202_release
 [X] 0019_v210_release
 [X] 0020_v230_release
 [X] 0021_v260_release
 [X] 0022_v270_release
 [X] 0023_v290_release
 [X] 0024_v2100_release
 [ ] 0025_increase_max_dashboard_graph_width
 [ ] 0026_v2140_release
 [ ] 0027_v215_release
 [ ] 0028_v3_0_fix_upgrade_redirect
dsmr_mindergas
 [X] 0001_mindergas
 [X] 0002_mindergas_notification
 [X] 0003_mindergas_next_export_datetime
 [X] 0004_mindergas_latest_sync
 [X] 0005_schedule_mindergas_export
dsmr_mqtt
 [X] 0001_mqtt_settings
 [X] 0002_daytotalsmqttsettings
 [X] 0003_splittopicmeterstatisticsmqttsettings
 [X] 0004_phases_currently_returned
 [X] 0005_mqtt_use_local_timezone
 [X] 0006_mqtt_refactoring
 [X] 0007_mqtts
 [X] 0008_mqtt_null_payload
 [X] 0009_remove_optional_debug
 [X] 0010_mqtt_gas_consumption
 [X] 0011_mqtt_meta_description
 [X] 0012_mqtt_voltage_defaults
 [ ] 0013_process_sleep
dsmr_notification
 [X] 0001_initial
 [X] 0002_notificationsetting_next_notification
 [X] 0003_notification_notification
 [X] 0004_statusnotificationsetting
 [X] 0005_notify_my_android_ended_support
 [X] 0006_support_for_pushover
 [X] 0007_support_for_telegram
dsmr_pvoutput
 [X] 0001_pvoutput_settings
 [X] 0002_pvoutput_latest_sync
dsmr_stats
 [X] 0001_squashed_0016_drop_stats_settings
 [X] 0002_regenerate_missing_gas_data
 [X] 0003_hour_statistics_gas_default
 [X] 0004_hour_statistics_gas_default_retroactive
 [X] 0005_statistics_exportverbose_names
 [X] 0006_min_max_temperature_statistics
 [X] 0007_min_max_temperature_statistics_retroactive
 [X] 0008_meta_names
 [X] 0009_statistics_editable
 [X] 0010_statistics_models_index
 [X] 0011_note_model_index
 [X] 0012_electricity_statistics
 [X] 0013_all_time_low
dsmr_weather
 [X] 0001_weather_models
 [X] 0002_meta
 [X] 0003_next_sync_setting
 [X] 0004_next_sync_setting_retroactive
 [X] 0005_weather_refactor_scheduling
 [X] 0006_schedule_weather_update
sessions
 [X] 0001_initial
```

dennissiemensma, 2020-02-05:
Je kunt dit nog proberen:

```
sudo su - dsmr

./manage.py sqlsequencereset dsmr_backend dsmr_consumption dsmr_datalogger dsmr_frontend dsmr_mqtt --no-color
```

**Kopieer de output (queries) en bewaar die.**

Doe daarna:
```
logout
sudo su - postgres
psql dsmrreader

# Plak nu alle eerder output (queries) hier in de terminal
```

Probeer nogmaals de migrate:
```
CTRL+D
CTRL+D

sudo su - dsmr
./manage.py migrate
```

kropveld, 2020-02-07:
Kennelijk was dit een update van de database structuur. Maar helaas was dit niet de oplossing:

```
pi@metrix:~ $ sudo su - dsmr
(dsmrreader) dsmr@metrix:~/dsmr-reader $ ./manage.py migrate
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backend, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_frontend, dsmr_mindergas, dsmr_mqtt, dsmr_notification, dsmr_pvoutput, dsmr_stats, dsmr_weather, sessions
Running migrations:
  Applying dsmr_backend.0009_update_email_settings_mail_from...Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 86, in _execute
    return self.cursor.execute(sql, params)
psycopg2.errors.UniqueViolation: duplicate key value violates unique constraint "django_migrations_pkey"
DETAIL:  Key (id)=(119) already exists.

The above exception was the direct cause of the following exception:
(...)
```
Misschien gewoon met een leeg SD-kaartje opnieuw installeren??

dennissiemensma, 2020-02-07:
Mijn vorige suggestie miste nog een query voor de tabel waar het over gaat. Kun je dit proberen:

```
sudo su - postgres
psql dsmrreader

BEGIN;
SELECT setval(pg_get_serial_sequence('"django_migrations"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "django_migrations";
COMMIT;
```

Daarna nogmaals `migrate` proberen.

kropveld, 2020-02-09:
OK dat was het. :)
Het was nog verder een heel gedoe om het weer werkend te krijgen omdat de vorige herinstallatie afgebroken was, maar na veel gehannes kreeg ik het weer werkend. De geschiedenis is miraculeus ook bewaard gebleven!

closed 2020-02-09 by kropveld