dsmrreader/dsmr-reader #867: Many unprocessed telegrams after update to v3.2

MMXInside, 2020-02-06:
Na een update naar v3.2 eerder deze week lijkt mijn dsmr_datalogger niet meer te werken en lopen de unprocessed telegrams erg op (226.223).

Bij het controleren van de supervisorctl krijg ik het volgende te zien,

> dsmr_backend                     RUNNING   pid 1054, uptime 0:12:12
dsmr_datalogger                  BACKOFF   Exited too quickly (process log may have details)
dsmr_mqtt                        STOPPED   Not started
dsmr_webinterface                RUNNING   pid 783, uptime 0:00:00

Dit staat in de log in /home/dsmr/dsmr-reader/logs/dsmrreader.log

> [2020-02-02 11:17:19,523] WARNING  @ services | Rejected telegram (Invalid telegram CRC. The calculated checksum '40424' (9DE8) does not match the telegram checksum '19999' (4E1F)) (base64 encoded): b'L0lTSzVcMk01NTBFLTEwMTINCg0KMS0zOjAuMi44KDUwKQ0KMC0wOjEuMC4wKDIwMDIwMjExMTcxM1cpDQowLTA6OTYuMS4xKDQ1MzAzMDM0MzMzMDMwMzYzODM2MzMzODM2MzYzNDMxMzcpDQoxLTA6MS44LjEoMDA1ODM5Ljk1MSprV2gpDQoxLTA6MS44LjIoMDA1MDMyLjg3NSprV2gpDQoxLTA6Mi44LjEoMDAwMDAwLjAwMCprV2gpDQoxLTA6Mi44LjIoMDAwMDAwLjAwMCprV2gpDQowLTA6OTYuMTQuMCgwMDAxKQ0KMS0wOjEuNy4wKDAwLjY1NyprVw0KMS0wOjIyLjcuMCgwMC4wMDAqa1cpDQowLTE6MjQuMS4wKDAwMykNCjAtMTo5Ni4xLjAoNDczMDMwMzMzOTMwMzAzMTM3MzAzNDMzMzAzMjM5MzEzNykNCjAtMToyNC4yLjEoMjAwMjAyMTExNTA4VykoMDMxMTguNzg2Km0zKQ0KITRFMUY='
[2020-02-02 11:17:19,524] ERROR    @ services | Invalid telegram CRC. The calculated checksum '40424' (9DE8) does not match the telegram checksum '19999' (4E1F)
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", line 127, in telegram_to_reading
    parsed_telegram = parser.parse(data)
  File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 43, in parse
    self.validate_checksum(telegram_data)
  File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 89, in validate_checksum
    expected_crc_hex
dsmr_parser.exceptions.InvalidChecksumError: Invalid telegram CRC. The calculated checksum '40424' (9DE8) does not match the telegram checksum '19999' (4E1F)

Enig idee hoe ik dit kan verhelpen? :-)

dennissiemensma, 2020-02-06:
Bedankt voor je melding. Ik denk dat er twee dingen door elkaar lopen hier:
- Telegrammen worden geweigerd wegens verkeerde checksum
- Telegrammen die niet worden geweigerd worden niet verwerkt

Het eerste is geen probleem, zolang er maar wel genoeg telegrammen worden geaccepteerd. Gezien je aangeeft dat de teller oploopt, lijkt dat dus wel zo.

Het tweede hangt waarschijnlijk ergens op. Je kunt het beste debug logging inschakelen en dan in de log van `dsmr_backend` kijken waar die op wacht.

- Voor inschakelen debug-log: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging
- Bekijken logfile: `/var/log/supervisor/dsmr_backend.log`

MMXInside, 2020-02-06:
Bedankt voor de reactie. 
Excuses voor de verwarring ik dacht dat de twee punten met elkaar te maken hadden.

Ik heb de logging naar DEBUG gezet en krijg nu het volgende te zien in `/var/log/supervisor/dsmr_backend.log`

>[2020-02-06 23:11:52,933] DEBUG    SP: 0 backend service(s) ready to run
[2020-02-06 23:11:52,938] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-02-06 23:11:53,942] DEBUG    SP: 0 backend service(s) ready to run
[2020-02-06 23:11:53,947] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-02-06 23:11:54,954] DEBUG    SP: 0 backend service(s) ready to run
[2020-02-06 23:11:54,958] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-02-06 23:11:55,964] DEBUG    SP: 0 backend service(s) ready to run
[2020-02-06 23:11:55,969] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-02-06 23:11:56,973] DEBUG    SP: 0 backend service(s) ready to run
[2020-02-06 23:11:56,977] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s

dennissiemensma, 2020-02-07:
Zie je hetzelfde als je ook wat verder scrolt? Het zou kunnen zijn dat die slechts 1x per minuut de telegrammen probeert te verwerken en dat je net die logregels mist.

En wat krijg je hier te zien:
```
sudo su - postgres
psql dsmrreader
select * from dsmr_backend_scheduledprocess;
```

MMXInside, 2020-02-07:
Nee helaas krijg ik niks anders te zien dan:

>root@DSMR:/home/*USER*# tail -f /var/log/supervisor/dsmr_backend.log
[2020-02-07 22:31:33,884] DEBUG    SP: 0 backend service(s) ready to run
[2020-02-07 22:31:33,888] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-02-07 22:31:34,895] DEBUG    SP: 0 backend service(s) ready to run
[2020-02-07 22:31:34,900] DEBUG 

Bij het uitvoeren van je tweede vraag krijg ik het volgende te zien:

>postgres@DSMR:~$ psql dsmrreader
psql (11.5 (Debian 11.5-1+deb10u1))
Type "help" for help.

>dsmrreader=# select * from dsmr_backend_scheduledprocess;
 id |                   name                    |                  module                  |            planned            | active |       last_executed_at
----+-------------------------------------------+------------------------------------------+-------------------------------+--------+-------------------------------
  4 | Upload gas meter position to MinderGas.nl | dsmr_mindergas.services.run              | 2020-01-08 23:35:53.240613+01 | f      |
  2 | Generate day and hour statistics          | dsmr_stats.services.run                  | 2020-02-07 22:33:18.391369+01 | t      | 2020-02-07 22:28:18.305553+01
  5 | Weather update                            | dsmr_weather.services.run                | 2020-02-07 23:00:00+01        | t      | 2020-02-07 22:00:00.790162+01
  1 | Automatic update checker                  | dsmr_backend.services.update_checker.run | 2020-02-14 00:03:31.392618+01 | t      | 2020-02-07 00:03:31.291942+01
  3 | Backup per email                          | dsmr_backup.services.email.run           | 2020-02-08 00:03:31.415536+01 | t      | 2020-02-07 00:03:31.408513+01
(5 rows)
dsmrreader=#

Excuses, dit ^^^ is misschien wat onhandig lezen ;-)

dennissiemensma, 2020-02-08:
Ik denk dat het veroorzaakt wordt doordat er een proces mist in de lijst. Wat krijg je hierbij te zien:
```
sudo su - dsmr
./manage.py showmigrations
```

MMXInside, 2020-02-08:
```
(dsmrreader) dsmr@DSMR:~/dsmr-reader$ ./manage.py showmigrations
admin
 [X] 0001_initial
 [X] 0002_logentry_remove_auto_add
 [X] 0003_logentry_add_action_flag_choices
auth
 [X] 0001_initial
 [X] 0002_alter_permission_name_max_length
 [X] 0003_alter_user_email_max_length
 [X] 0004_alter_user_username_opts
 [X] 0005_alter_user_last_login_null
 [X] 0006_require_contenttypes_0002
 [X] 0007_alter_validators_add_error_messages
 [X] 0008_alter_user_username_max_length
 [X] 0009_alter_user_last_name_max_length
 [X] 0010_alter_group_name_max_length
 [X] 0011_update_proxy_permissions
contenttypes
 [X] 0001_initial
 [X] 0002_remove_content_type_name
dsmr_api
 [X] 0001_api_settings
 [X] 0002_generate_random_auth_key
 [X] 0003_create_api_user
dsmr_backend
 [X] 0001_backend_language
 [X] 0002_email_settings
 [X] 0003_scheduled_processes
 [X] 0004_backend_schedule_process_active_flag
 [X] 0005_schedule_auto_update_check
 [X] 0006_backend_auto_update_check
 [X] 0007_schedule_stats_generator
 [X] 0008_scheduled_process_fields
 [X] 0009_update_email_settings_mail_from
 [X] 0010_process_sleep
 [X] 0011_disable_capabilities
 [X] 0012_fix_email_from
dsmr_backup
 [X] 0001_initial
 [X] 0002_settings_documentation
 [X] 0003_dropbox_sync_field
 [X] 0004_custom_backup_path
 [X] 0005_email_backup_settings
 [X] 0006_scheduled_email_backup
 [X] 0007_remove_backupsettings_compress
dsmr_consumption
 [X] 0001_squashed_0004_recalculate_gas_consumption
 [X] 0002_verbose_text
 [X] 0003_electricity_consumption_indexes
 [X] 0004_merge_electricity_tariffs
 [X] 0005_phase_currently_delivered
 [X] 0006_dsmr_firmware_v5
 [X] 0007_settings_documentation
 [X] 0008_energysupplier_return_prices
 [X] 0009_consumption_settings_text
 [X] 0010_phases_currently_returned
 [X] 0011_track_phase_voltages
 [X] 0012_add_voltage_indexes
 [ ] 0013_schedule_generate_consumption
 [ ] 0014_gas_grouping
dsmr_datalogger
 [X] 0001_squashed_0005_optional_gas_readings
 [X] 0002_meta_names
 [X] 0003_telegram_checksum
 [X] 0004_phase_currently_delivered
 [X] 0005_verify_telegram_crc_setting
 [X] 0006_dsmr_firmware_v5
 [X] 0007_dsmrreading_timestamp_index
 [X] 0008_dsmrreading_help_text
 [X] 0009_retention_settings
 [X] 0010_phases_currently_returned
 [X] 0011_raw_telegram_insight
 [X] 0012_track_phase_voltages
 [X] 0013_remove_datalogger_track_settings
 [X] 0014_remove_optional_crc_check
 [X] 0015_datalogger_foreign_countries
 [X] 0016_process_sleep
 [X] 0017_log_telegrams
 [ ] 0018_meterstatistics_no_autonow
dsmr_frontend
 [X] 0001_initial
 [X] 0002_recent_history_weeks
 [X] 0003_drop_history_page
 [X] 0004_chart_colors
 [X] 0005_notifications
 [X] 0006_notifications_initial
 [X] 0007_merge_electricity_tariffs
 [X] 0008_merge_electricity_tariffs_notification
 [X] 0009_docs_no_reverse_match
 [X] 0010_drop_reverse_dashboard_graphs_setting
 [X] 0011_phase_currently_delivered
 [X] 0012_frontendsettings_dashboard_graph_width
 [X] 0013_echart_graphs_message
 [X] 0014_phases_currently_returned
 [X] 0015_notification_meta
 [X] 0016_v200_release
 [X] 0017_v201_release
 [X] 0018_v202_release
 [X] 0019_v210_release
 [X] 0020_v230_release
 [X] 0021_v260_release
 [X] 0022_v270_release
 [X] 0023_v290_release
 [X] 0024_v2100_release
 [X] 0025_increase_max_dashboard_graph_width
 [X] 0026_v2140_release
 [ ] 0027_v215_release
 [ ] 0028_v3_0_fix_upgrade_redirect
dsmr_mindergas
 [X] 0001_mindergas
 [X] 0002_mindergas_notification
 [X] 0003_mindergas_next_export_datetime
 [X] 0004_mindergas_latest_sync
 [X] 0005_schedule_mindergas_export
dsmr_mqtt
 [X] 0001_mqtt_settings
 [X] 0002_daytotalsmqttsettings
 [X] 0003_splittopicmeterstatisticsmqttsettings
 [X] 0004_phases_currently_returned
 [X] 0005_mqtt_use_local_timezone
 [X] 0006_mqtt_refactoring
 [X] 0007_mqtts
 [X] 0008_mqtt_null_payload
 [X] 0009_remove_optional_debug
 [X] 0010_mqtt_gas_consumption
 [X] 0011_mqtt_meta_description
 [X] 0012_mqtt_voltage_defaults
 [X] 0013_process_sleep
dsmr_notification
 [X] 0001_initial
 [X] 0002_notificationsetting_next_notification
 [X] 0003_notification_notification
 [X] 0004_statusnotificationsetting
 [X] 0005_notify_my_android_ended_support
 [X] 0006_support_for_pushover
 [X] 0007_support_for_telegram
dsmr_pvoutput
 [X] 0001_pvoutput_settings
 [X] 0002_pvoutput_latest_sync
dsmr_stats
 [X] 0001_squashed_0016_drop_stats_settings
 [X] 0002_regenerate_missing_gas_data
 [X] 0003_hour_statistics_gas_default
 [X] 0004_hour_statistics_gas_default_retroactive
 [X] 0005_statistics_exportverbose_names
 [X] 0006_min_max_temperature_statistics
 [X] 0007_min_max_temperature_statistics_retroactive
 [X] 0008_meta_names
 [X] 0009_statistics_editable
 [X] 0010_statistics_models_index
 [X] 0011_note_model_index
 [X] 0012_electricity_statistics
 [X] 0013_all_time_low
dsmr_weather
 [X] 0001_weather_models
 [X] 0002_meta
 [X] 0003_next_sync_setting
 [X] 0004_next_sync_setting_retroactive
 [X] 0005_weather_refactor_scheduling
 [X] 0006_schedule_weather_update
sessions
 [X] 0001_initial
```

dennissiemensma, 2020-02-08:
De upgrade is dan niet volledig afgerond. Je kunt het nogaals proberen met:

```
sudo su - dsmr
./deploy.sh
```

En als dat niet werkt:
```
sudo su - dsmr
./manage.py migrate
```

MMXInside, 2020-02-08:
Met beide commando’s heb ik geen succes :-(

```
(dsmrreader) dsmr@DSMR:~/dsmr-reader$ ./deploy.sh

 --- You are currently running version: 
3.2

 --- Checking Python version.
Running Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.3.0]

 --- Pulling remote repository for new commits...

 --- Merging/updating checkout.
Already up to date.

 >>> Running post-deployment script. <<<

 --- Checking whether VirtualEnv is activated.

 --- Checking Python version.
Running Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.3.0]

 --- Checking & synchronizing base requirements for changes.
Collecting https://qpypi3.qpython.org/repository/1745924/PyCRC-1.21.tar.gz (from -r dsmrreader/provisioning/requirements/base.txt (line 19))
  Downloading https://qpypi3.qpython.org/repository/1745924/PyCRC-1.21.tar.gz (27 kB)
Requirement already up-to-date: coreapi==2.3.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 1)) (2.3.3)
Requirement already up-to-date: crcmod==1.7 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 2)) (1.7)
Requirement already up-to-date: django==3.0.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 3)) (3.0.2)
Requirement already up-to-date: django-colorfield==0.1.15 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 4)) (0.1.15)
Requirement already up-to-date: django-debug-toolbar==2.1 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 5)) (2.1)
Requirement already up-to-date: django-filter==2.2.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 6)) (2.2.0)
Requirement already up-to-date: djangorestframework==3.11 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 7)) (3.11.0)
Requirement already up-to-date: django-solo==1.1.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 8)) (1.1.3)
Requirement already up-to-date: dropbox==9.4.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 9)) (9.4.0)
Requirement already up-to-date: gunicorn==20.0.4 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 10)) (20.0.4)
Requirement already up-to-date: paho-mqtt==1.5.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 11)) (1.5.0)
Requirement already up-to-date: pyserial==3.4 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 12)) (3.4)
Requirement already up-to-date: python-dateutil==2.8.1 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 13)) (2.8.1)
Requirement already up-to-date: pytz==2019.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 14)) (2019.3)
Requirement already up-to-date: requests==2.22.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 15)) (2.22.0)
Requirement already up-to-date: urllib3==1.25.8 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 16)) (1.25.8)
Requirement already satisfied, skipping upgrade: coreschema in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (0.0.4)
Requirement already satisfied, skipping upgrade: uritemplate in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (3.0.1)
Requirement already satisfied, skipping upgrade: itypes in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (1.1.0)
Requirement already satisfied, skipping upgrade: asgiref~=3.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from django==3.0.2->-r dsmrreader/provisioning/requirements/base.txt (line 3)) (3.2.3)
Requirement already satisfied, skipping upgrade: sqlparse>=0.2.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from django==3.0.2->-r dsmrreader/provisioning/requirements/base.txt (line 3)) (0.3.0)
Requirement already satisfied, skipping upgrade: six>=1.3.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from dropbox==9.4.0->-r dsmrreader/provisioning/requirements/base.txt (line 9)) (1.13.0)
Requirement already satisfied, skipping upgrade: setuptools>=3.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from gunicorn==20.0.4->-r dsmrreader/provisioning/requirements/base.txt (line 10)) (44.0.0)
[... cut ...]
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/utils.py", line 90, in __exit__    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 86, in _execute
    return self.cursor.execute(sql, params)
django.db.utils.IntegrityError: duplicate key value violates unique constraint "django_migrations_pkey"
DETAIL:  Key (id)=(116) already exists.

(dsmrreader) dsmr@DSMR:~/dsmr-reader$
```

dennissiemensma, 2020-02-08:
Die foutmelding zou je dan ook al bij de upgrade gehad moeten hebben.

Je kunt kijken of dit het oplost:
```
sudo su - postgres
psql dsmrreader

BEGIN;
SELECT setval(pg_get_serial_sequence('"django_migrations"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "django_migrations";
COMMIT;
```

En daarna weer
```
sudo su - dsmr
./manage.py migrate
```

MMXInside, 2020-02-10:
Super, dit was de oplossing.
Alle unprocessed telegrams worden op dit moment verwerkt.
En de grafieken zijn ook weer zichtbaar.  :-D

dennissiemensma, 2020-02-10:
Top, fijn om te horen!

closed 2020-02-10 by dennissiemensma