dsmrreader/dsmr-reader #868: MQTT process start niet bij verse installatie

PlatoNL, 2020-02-06:
Ik heb DSMR Reader vandaag geïnstalleerd op een RPi4, maar ik krijg de MQTT functionaliteit niet aan de gan g.

Wanneer ik reload.sh aanroep krijg ik de volgende output:
(```
dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ ./reload.sh

 * Reloading process: dsmr_webinterface (Gunicorn)   [OK]
 * Reloading process: dsmr_backend                   [OK]
 * Reloading process: dsmr_datalogger                [OK]
 * Reloading process: dsmr_mqtt                      [??] PID file does not exist
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $
```

Ik heb de instructies voor het opnieuw installeren van alle requirements gevolgd volgens deze thread:
[](https://github.com/dennissiemensma/dsmr-reader/issues/570)

Maar dat heeft niet geholpen.

Verder zie ik geen dsmr_mqtt.conf bestand in /etc/supervisor/conf.d/ maar ik weet niet of deze er überhaupt zou moeten staan.

PlatoNL, 2020-02-07:
Zelf aan de gang gekregen door dsmr_mqtt.conf aan te maken.

closed 2020-02-07 by PlatoNL

dennissiemensma, 2020-02-07:
Fijn dat je het zelf hebt gevonden. Het staat wellicht niet prominent gelinked in de applicatie zelf, maar hier staat meer informatie: https://dsmr-reader.readthedocs.io/nl/v3/mqtt.html