dsmrreader/dsmr-reader #873: Update naar v3 loopt vast

Djjvb, 2020-02-11:
Gaat er iets fout, maar weet je niet wat? [Bekijk de logfiles](https://dsmr-reader.readthedocs.io/nl/v2/troubleshooting.html) als startpunt.

**Wat gebruik je?**
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.12/v3`
* Hardware *(RaspberryPi 3, 4 of iets anders)*: `RaspberryPi 3b`
* DSMR-protocol slimme meter *(v4 of v5)*: `v4`

Heb het  upgrade bestand gevolgd https://dsmr-reader.readthedocs.io/nl/v3/faq/v3_upgrade.html

En python is bijgewerkt:
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ python3.6 --version
Python 3.6.9

Alleen krijg bij de laatste stap de volgende melding:

(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ ./deploy.sh

 --- You are currently running version: 
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/dsmr/dsmr-reader/dsmrreader/__init__.py", line 1, in <module>
    from django.utils.version import get_version
ModuleNotFoundError: No module named 'django'

     [!] FAILED to call Django (did you activate the 'dsmrreader' VirtualEnv before running?)

[!] Halted in pre-deployment: Either user ABORTED or script FAILED

Nu zag ik daar ook al een eerder issue van bij #840, met de oplossing om het volgende te doen:

sudo su - dsmr
pip3 install -r dsmrreader/provisioning/requirements/base.txt

Dit heb ik uitgevoerd, maar toe kreeg ik de volgende meldingen:

pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Collecting https://qpypi3.qpython.org/repository/1745924/PyCRC-1.21.tar.gz (from -r dsmrreader/provisioning/requirements/base.txt (line 19))
Exception:
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 423, in send
    timeout=timeout
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 582, in urlopen
    conn = self._get_conn(timeout=pool_timeout)
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 255, in _get_conn
    return conn or self._new_conn()
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 814, in _new_conn
    raise SSLError("Can't connect to HTTPS URL because the SSL "
requests.packages.urllib3.exceptions.SSLError: Can't connect to HTTPS URL because the SSL module is not available.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/req/req_set.py", line 620, in _prepare_file
    session=self.session, hashes=hashes)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/download.py", line 821, in unpack_url
    hashes=hashes
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/download.py", line 659, in unpack_http_url
    hashes)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/download.py", line 853, in _download_http_url
    stream=True,
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 501, in get
    return self.request('GET', url, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 488, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/home/dsmr/.virtualenvs/dsmrreader/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 497, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: Can't connect to HTTPS URL because the SSL module is not available.

Daarna heb ik geprobeerd de code van stap 6 nogmaals uit te voeren. Hierin zit ook bovenstaande, maar ook nog de volgende issues:

(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ pip3 install -r dsmrreader/provisioning/requirements/postgresql.txt
pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Collecting psycopg2-binary==2.8.4 (from -r dsmrreader/provisioning/requirements/postgresql.txt (line 1))
  Cache entry deserialization failed, entry ignored
  Could not find a version that satisfies the requirement psycopg2-binary==2.8.4 (from -r dsmrreader/provisioning/requirements/postgresql.txt (line 1)) (from versions: )
No matching distribution found for psycopg2-binary==2.8.4 (from -r dsmrreader/provisioning/requirements/postgresql.txt (line 1))
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ ./deploy.sh

En nu doet niks het meer. Ook als ik de code invoer om het terug te zetten, krijg ik hem niet meer aan de praat. Met een 502 bad gateway melding als resultaat. 

Alvast bedankt voor je reactie en al het werk dat je hierin stopt!

dennissiemensma, 2020-02-11:
Dit is een situatie die ik nog niet eerder gezien heb. 

Ik denk dat je het beste terug kunt gaan naar v2.x en daarna rustig een nieuwe virtualenv ernaast aanmaakt om te kijken of je de pip issues voor v3 werkend kan krijgt. 

Vermoedelijk heb je hier voor nu genoeg aan:
```
sudo su - dsmr
git branch

# Als je daar v2 tussen ziet staan, doe dan:
git checkout v2

# Zo niet, doe:
git checkout -b v2 origin/v2
```

Als het goed is gegaan kun je dit doen: `cat dsmrreader/__init__.py`
En zie je dan zoiets:
```
...

VERSION = (2, 15, 1, 'final', 0)

...
```

Daarna de oude virtualenv van je vorige Python-versie weer terugzetten (er vanuitgaande dat je die of de Python-versie zelf niet hebt verwijderd):

```
# Zoals beschreven in: https://dsmr-reader.readthedocs.io/nl/v3/faq/v3_upgrade.html

deactivate
cd ~
mv .virtualenvs/v2-dsmrreader .virtualenvs/dsmrreader

cd dsmr-reader
./deploy.sh
```

Je zou dan weer op v2.15.1 moeten zitten.

dennissiemensma, 2020-02-11:
Nog een klein foutje gevonden, waarschijnlijk moet je in de laatste stappen hierboven nog `.virtualenvs/dsmrreader` doen om de nieuwe v3-virtualenv te verwijderen, voordat je die van v2 terugzet (je krijgt anders waarschijnlijk een foutmelding dat die map al bestaat).

dennissiemensma, 2020-02-11:
Als je trouwens die laatste stappen al eerder had gedaan, dan hoef je vermoedelijk alleen terug te wisselen naar de `v2` branch. 
Vermoedelijk gaat je installatie stuk omdat v3 code tegen een v2-virtualenv aan praat.

Als je alles gedaan hebt, kun je het alles geforceerd herstarten met root of pi gebruiker:
```
sudo supervisorctl restart all
```

Djjvb, 2020-02-11:
Dag Dennis,

Dank voor je snelle reactie. Ik kreeg geen foutmelding over de bestaande map.  Heb die extra regel dus niet toegevoegd.

Verder ging het goed t/m de version stap. Bij het terugzetten kreeg ik weer de 'oude melding' van Failed to call Django.

Omdat je schreef dat ik het misschien al eerder gedaan had en dus niet nodig zou zijn, heb ik de geforceerde start gebruikt met de volgende uitkomst:

pi@raspberrypi:~ $ sudo supervisorctl restart all
dsmr_mqtt: stopped
dsmr_backend: stopped
dsmr_datalogger: stopped
dsmr_webinterface: ERROR (spawn error)
dsmr_mqtt: ERROR (spawn error)
dsmr_datalogger: ERROR (spawn error)
dsmr_backend: ERROR (spawn error)
pi@raspberrypi:~ $ 

Kom er nog niet bij.

dennissiemensma, 2020-02-11:
Je kunt nu even wat dingen moeten checken, wat is de output hiervan:

```
sudo su - dsmr
git branch
python3 --version
pip3 freeze | grep django
cat dsmrreader/__init__.py
```

Djjvb, 2020-02-11:
Dit:

(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ git branch
  master
* v2
  v3
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ python3 --version
Python 3.6.9
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ pip3 freeze | grep django
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ cat dsmrreader/__init__.py
from django.utils.version import get_version

VERSION = (2, 15, 1, 'final', 0)

__version__ = get_version(VERSION)

dennissiemensma, 2020-02-11:
Had je hiervoor ook al `Python 3.6.9`? Of heb je die juist geinstalleerd om je eerst een oudere versie had?

Djjvb, 2020-02-11:
Had 3.5.3. Heb dus upgrade uitgevoerd conform handleiding.

dennissiemensma, 2020-02-11:
Dan lijkt het terugzetten van de virtualenv niet gelukt. Je hebt tussendoor wel een keer `logout` en `sudo  su - dsmr` gedaan neem ik aan?

Wat zie je met:
```
sudo su - dsmr
ls -l ~/.virtualenvs/
```

dennissiemensma, 2020-02-11:
Welke Python-versie zie je met:

```
sudo su - dsmr
deactivate

python3 --version
```

Djjvb, 2020-02-11:
Ik log  braaf steeds uit en in. Ben echt een noob (had je al door ;-)), maar dat had ik al geleerd na een paar keer de vraag om het dsmr wachtwoord. :-)

Op je virtualenvs vraag:
totaal 4
drwxr-xr-x 7 dsmr dsmr 4096 feb 11 21:54 dsmrreader

Op de python versie:
Python 3.6.9

dennissiemensma, 2020-02-11:
Je zult nu namelijk op zoek moeten naar je oude Python-versie, die zou nog gewoon beschikbaar moeten zijn.

Geeft dit output?

```
sudo su - dsmr
deactivate

python3.5 --version
```

Djjvb, 2020-02-11:
Yep: Python 3.5.3

dennissiemensma, 2020-02-11:
Want dan is het een kwestie van gewoon een nieuwe virtualenv maken met Python3.5. Probeer dit maar:

```
sudo su - dsmr
deactivate
cd ~
mv .virtualenvs/dsmrreader .virtualenvs/OLD-dsmrreader

cd dsmr-reader
virtualenv /home/dsmr/.virtualenvs/dsmrreader --no-site-packages --python python3.5
source ~/.virtualenvs/dsmrreader/bin/activate

pip3 install -r dsmrreader/provisioning/requirements/base.txt
pip3 install -r dsmrreader/provisioning/requirements/postgresql.txt
./deploy.sh
```

En als dat goedgaat kun je weer eens dit proberen:
```
sudo supervisorctl restart all
```

Djjvb, 2020-02-11:
HELD! Nu zit ik weer op een werkende v2.15.1 Wat is nu handig tav migratie naar v3?

Nogmaals/alvast dank voor alle hulp. Sorry dat ik niet meer zelf kan doen, maar dit gaat mijn pet echt te boven.

dennissiemensma, 2020-02-12:
Ik denk dat je eerst nogmaals moet proberen om een v3-virtualenv aan te maken met de nieuwere Python, maar dan als proef en dus nog DSMR-reader lekker op v2 laten draaien.

Je kunt dit proberen:
```
sudo su - dsmr
deactivate

virtualenv ~/.virtualenvs/test-v3-dsmrreader --no-site-packages --python python3.6
source ~/.virtualenvs/test-v3-dsmrreader/bin/activate

pip3 install -r dsmrreader/provisioning/requirements/base.txt
pip3 install -r dsmrreader/provisioning/requirements/postgresql.txt
```

Kijk eerst maar of dat lukt zonder fouten.

Djjvb, 2020-02-12:
Dat lukt dus helaas niet. Dit is de output:

`pi@raspberrypi:~ $ sudo su - dsmr
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ deactivate
dsmr@raspberrypi:~/dsmr-reader $ 
dsmr@raspberrypi:~/dsmr-reader $ virtualenv ~/.virtualenvs/test-v3-dsmrreader --no-site-packages --python python3.6
Running virtualenv with interpreter /usr/local/bin/python3.6
Using base prefix '/usr/local'
New python executable in /home/dsmr/.virtualenvs/test-v3-dsmrreader/bin/python3.6
Also creating executable in /home/dsmr/.virtualenvs/test-v3-dsmrreader/bin/python
Installing setuptools, pkg_resources, pip, wheel...done.
dsmr@raspberrypi:~/dsmr-reader $ source ~/.virtualenvs/test-v3-dsmrreader/bin/activate
(test-v3-dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ 
(test-v3-dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ pip3 install -r dsmrreader/provisioning/requirements/base.txt
pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Collecting https://qpypi3.qpython.org/repository/1745924/PyCRC-1.21.tar.gz (from -r dsmrreader/provisioning/requirements/base.txt (line 20))
Exception:
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 423, in send
    timeout=timeout
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 582, in urlopen
    conn = self._get_conn(timeout=pool_timeout)
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 255, in _get_conn
    return conn or self._new_conn()
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 814, in _new_conn
    raise SSLError("Can't connect to HTTPS URL because the SSL "
requests.packages.urllib3.exceptions.SSLError: Can't connect to HTTPS URL because the SSL module is not available.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/lib/python3.6/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/lib/python3.6/site-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/lib/python3.6/site-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/lib/python3.6/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/lib/python3.6/site-packages/pip/req/req_set.py", line 620, in _prepare_file
    session=self.session, hashes=hashes)
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/lib/python3.6/site-packages/pip/download.py", line 821, in unpack_url
    hashes=hashes
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/lib/python3.6/site-packages/pip/download.py", line 659, in unpack_http_url
    hashes)
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/lib/python3.6/site-packages/pip/download.py", line 853, in _download_http_url
    stream=True,
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 501, in get
    return self.request('GET', url, **kwargs)
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/lib/python3.6/site-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 488, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/home/dsmr/.virtualenvs/test-v3-dsmrreader/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 497, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: Can't connect to HTTPS URL because the SSL module is not available.
(test-v3-dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ pip3 install -r dsmrreader/provisioning/requirements/postgresql.txt
pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Collecting psycopg2-binary==2.8.4 (from -r dsmrreader/provisioning/requirements/postgresql.txt (line 1))
  Cache entry deserialization failed, entry ignored
  Could not find a version that satisfies the requirement psycopg2-binary==2.8.4 (from -r dsmrreader/provisioning/requirements/postgresql.txt (line 1)) (from versions: )
No matching distribution found for psycopg2-binary==2.8.4 (from -r dsmrreader/provisioning/requirements/postgresql.txt (line 1))`

dennissiemensma, 2020-02-12:
Wat voor OS gebruik je?

```
lsb_release -a
```

Djjvb, 2020-02-12:
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.6 (stretch)
Release:	9.6
Codename:	stretch

dennissiemensma, 2020-02-12:
Dan wordt het een kwestie van proberen. Als ik een random oplossing bekijk: https://stackoverflow.com/a/44758621

Dan kun je proberen om die modules te installeren en dan Python 3.6 nogmaals compilen (volgens de upgrade instructies):
```
sudo apt-get install libreadline-gplv2-dev libncursesw5-dev libssl-dev libsqlite3-dev tk-dev libgdbm-dev libc6-dev libbz2-dev -y

# Uit de guide
wget https://www.python.org/ftp/python/3.6.9/Python-3.6.9.tar.xz
tar xJf Python-3.6.9.tar.xz
cd Python-3.6.9
./configure
make
sudo make install
sudo pip3 install --upgrade pip
```

Het is dus geen garantie dat het het oplost, maar iemand daaronder meldt dat het werkt voor jouw OS, dus wie weet.

Als daarna `python3.6 --version` nog werkt kun je het nogmaals proberen:

```
sudo su - dsmr
deactivate

#### Extra stap om je eerdere poging weg te halen.
rm -rf ~/.virtualenvs/test-v3-dsmrreader

virtualenv ~/.virtualenvs/test-v3-dsmrreader --no-site-packages --python python3.6
source ~/.virtualenvs/test-v3-dsmrreader/bin/activate

pip3 install -r dsmrreader/provisioning/requirements/base.txt
pip3 install -r dsmrreader/provisioning/requirements/postgresql.txt
```

dennissiemensma, 2020-02-12:
En als dat geen oplossing is, dan kun je of verder zoeken/prutsen of overwegen om simpelweg je Raspberry een opnieuw te installeren met een nieuwere versie van het OS. 
Daar krijg je dan automatisch een (veel) nieuwere versie van Python3 bij die sowieso werkt (ik gebruik die zelf ook).

```
pi@rpi4:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

pi@rpi4:~ $ python3 --version
Python 3.7.3
```

https://packages.debian.org/buster/python3

Het nadeel is vanzelfsprekend dat het je wat meer tijd kost en dat je goed de database-backup voor DSMR-reader moet overzetten.

Djjvb, 2020-02-12:
Ik ben een beetje gaan snuffelen en kwam op de volgende oplossing:

https://gist.github.com/dschep/24aa61672a2092246eaca2824400d37f 

Heb daar de required build tools uit geïnstalleerd van stap 1 met bijbehorende code. En daarna python 3.6 opgepakt uit jouw instructies:

wget https://www.python.org/ftp/python/3.6.9/Python-3.6.9.tar.xz
tar xJf Python-3.6.9.tar.xz
cd Python-3.6.9
./configure
make
sudo make install
sudo pip3 install --upgrade pip

Daarna jouw verdere instructies gevolgd:

sudo su - dsmr
deactivate

#### Extra stap om je eerdere poging weg te halen.
rm -rf ~/.virtualenvs/test-v3-dsmrreader

virtualenv ~/.virtualenvs/test-v3-dsmrreader --no-site-packages --python python3.6
source ~/.virtualenvs/test-v3-dsmrreader/bin/activate

pip3 install -r dsmrreader/provisioning/requirements/base.txt
pip3 install -r dsmrreader/provisioning/requirements/postgresql.txt

En nu alles zonder fouten! Wat is nu de volgende stap? Durf niet zomaar code uit de handleiding te kopiëren.

dennissiemensma, 2020-02-12:
In principe zouden die instructies gewoon moeten werken nu. En je kunt telkens terug naar v2.

```
sudo su - dsmr
deactivate
rm -rf ~/.virtualenvs/test-v3-dsmrreader
logout
```

Daarna:
```
sudo su - dsmr
deactivate
cd ~
mv .virtualenvs/dsmrreader .virtualenvs/v2-dsmrreader

virtualenv /home/dsmr/.virtualenvs/dsmrreader --no-site-packages --python python3.6
logout
```

En dan:

```
sudo su - dsmr
git fetch
git checkout v3

# Make sure you're at v3 now:
git branch

git pull
pip3 install -r dsmrreader/provisioning/requirements/base.txt
pip3 install -r dsmrreader/provisioning/requirements/postgresql.txt

# Now redeploy
./deploy.sh

# (Re)start all processes
logout
sudo supervisorctl restart all
```

dennissiemensma, 2020-02-12:
Vergeet trouwens niet om DSMR-reader uit te zetten:
```
sudo supervisorctl stop all
```

Djjvb, 2020-02-12:
En ik ben op versie 3.3!! Yes... ! Dank, dank, dank Dennis! Echt niet normaal wat je voor ons doet echt super. Voel me echt een beetje schuldig dat je mij als gebruiker met slechts zeer beperkte Linux vaardigheden en geen pyhton kennis en ervaring zo op weg moet helpen.

Zou het dan in de required build tools voor python kunnen liggen uiteindelijk?

dennissiemensma, 2020-02-12:
Mooi! De exacte reden van de fouten weet ik niet. Maar erg bedankt voor het melden in ieder geval, de upgrade-guide is nog wat bijgewerkt.

closed 2020-02-12 by dennissiemensma