dsmrreader/dsmr-reader #874: Fout melding bij 'Notifications' na upgrade naar V3

D3rax, 2020-02-11:
Nadat ik een upgrade naar V3 gedaan heb krijg ik een foutmelding als ik 'notifications' open:

```
Server Error

Sorry, something unexpected happened.
Exception:

NoReverseMatch: Reverse for 'v3-upgrade-redirect' not found. 'v3-upgrade-redirect' is not a valid view function or pattern name.

Traceback:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/handlers/base.py", line 143, in _get_response
    response = response.render()

[.....]
```

dennissiemensma, 2020-02-11:
Wil je kijken of dit het voor je oplost:

```
sudo su - dsmr
./manage.py migrate
```

Zo nee, wat krjg je hier te zien:
```
sudo su - postgresql
psql dsmrreader

select * from dsmr_frontend_notification where redirect_to like '%v3-upgrade-redirect%';
```

D3rax, 2020-02-12:
Met behulp van ./manage.py migrate is het opgelost, bedankt!

dennissiemensma, 2020-02-12:
Top, bedankt voor de terugkoppeling

closed 2020-02-12 by dennissiemensma