dsmrreader/dsmr-reader #875: Actuele Amperes weergeven

dgeest66, 2020-02-12:
Ik zou graag ook het actueel gebruik in Ampere per fase zien in dsmr-reader. En deze ook door kunnen zetten via mqtt. 

De applicatie die mijn laadpaal aanstuurt heeft   [https://github.com/snaptec/openWB](url) deze gegevens nodig heeft. 
Het gaat om de volgende OBIS-codes:
1-0:31.7.0(001*A)
1-0:51.7.0(000*A)
1-0:71.7.0(000*A)

Kan ik deze zelf toevoegen, of kan de code aangepast worden?

dennissiemensma, 2020-02-12:
Bedankt voor je verzoek. Ik zal er op termijn inhoudelijk naar kijken.

Je kunt ook nog overwegen om het zelf uit te lezen met [dsmr_parser](https://github.com/ndokter/dsmr_parser) en daarna direct door te geven aan die andere applicatie. Daar ben je wellicht wat sneller mee geholpen.

DarkShadowNight, 2020-02-16:
Dit zou een zeer zinnige functionaliteit kunnen zijn, daar je aansluiting beperkt is op stroom en niet op vermogen.
Een kleine grafiek net zoals de spanning zou al een leuke toevoeging zijn.

dennissiemensma, 2020-02-20:
@dgeest66 @DarkShadowNight ik heb een opzet klaar. Alleen ben ik er nog niet helemaal uit welke grafiekstijl het beste hiervoor is. Hebben jullie input? 

- Bij stacked staan de fasen altijd boven elkaar en kunnen ze elkaar nooit kruizen.
- Bij unstacks gaan ze door elkaar heen en kunnen ze elkaar kruizen wanneer ze wisselend hoger of lager dan elkaar zijn (zie voltages).

Zie de screenshots:

## Stacked bars
[image]

## Stacked lines
[image]

## Unstacked lines
[image]

dennissiemensma, 2020-02-20:
Voor nu in branch klaargezet met stacked bars als voorstel. Wacht op input.

dgeest66, 2020-02-20:
@dennissiemensma dank, ik zal zsm terugkoppelen

Ferryb, 2020-02-23:
Ik zou de Ampere's ook fijn vinden naar MQTT. Hopelijk snel een update. :)

dgeest66, 2020-02-23:
Misschien domme vraag, maar welke stappen moet ik nemen om van de huidige versie naar deze branch te updaten?

dennissiemensma, 2020-02-23:
Je kunt wisselen met:
```
sudo su - dsmr
git fetch
git checkout -b feature/875-power-current-graph wip/feature/875-power-current-graph
./deploy.sh
```

Zorg er wel voor dat je voor de zekerheid een recente database-backup achter de hand hebt.

dennissiemensma, 2020-02-24:
Ik heb de branch alvast gemerged naar de release branch. Mocht het nog een ander type moeten worden, dan pas ik het tussendoor nog wel aan.

closed 2020-02-24 by dennissiemensma

dennissiemensma, 2020-02-29:
Uitgebracht in [v3.5](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v3.5.0)

Ferryb, 2020-03-03:
Dennis, zijn de stoomwaardes nu ook via MQTT te versturen? via gesplitst topic? Ik zie ze er namelijk niet bij staan.

dennissiemensma, 2020-03-03:
Ik heb daar niet bij stil gestaan. Echter gaat alles dynamisch, dus je kunt kijken of deze mapping voor je werkt:

## JSON
```
phase_power_current_l1 = phase_power_current_l1
phase_power_current_l2 = phase_power_current_l2
phase_power_current_l3 = phase_power_current_l3
```

## Split topic
```
phase_power_current_l1 = dsmr/reading/phase_power_current_l1
phase_power_current_l2 = dsmr/reading/phase_power_current_l2
phase_power_current_l3 = dsmr/reading/phase_power_current_l3
```

Ik zal de beschrijvingen met bovenstaande waardes bijwerken.

Ferryb, 2020-03-03:
Ook deze werkt inderdaad. Alle waardes worden retained verstuurd.
Dank weer.

dgeest66, 2020-03-04:
@dennissiemensma , dankjewel voor de toevoeging!

dennissiemensma, 2020-03-06:
Bijgewerkt in [v3.6](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v3.6.0)

RubenDijk, 2020-05-15:
Volgens mij zijn de waardes die hij over mqtt weg stuurt geen ampère maar wattage. (versie 3.9.1)

RubenDijk, 2020-05-15:
[image]
[image]

dennissiemensma, 2020-05-15:
@RubenDijk bedankt voor je melding. 

Kan het zijn dat je in de verkeerde grafiek kijkt? Er staat expliciet Ampere in als ik het hier test. 
Mocht je geen DSMR v5 meter hebben, dan zijn deze gegevens overigens niet beschikbaar.

[image]

RubenDijk, 2020-05-15:
Hallo Dennis,

Ik zie wel Amperes in DSMR-Reader webpage. 
De waarde word niet door gestuurd naar ha (mqtt)

dennissiemensma, 2020-05-15:
Als je DSMR-reader al gebruikte voordat v3.5 er was, zul je ze met de hand moeten toevoegen aan de MQTT config. Dat gaat niet automatisch: https://github.com/dennissiemensma/dsmr-reader/issues/875#issuecomment-594137011

RubenDijk, 2020-05-15:
zo als hier boven zijn de regels toegevoegd.
Heb de supervisor reload en opnieuw opgestart.
De rpi gereboot, DSRM versie nr:5

RubenDijk, 2020-05-15:
Dennis,

Zou het kunnen dat HA het niet ondersteund?
Mqtt explorer ziet hem wel
[image]

dennissiemensma, 2020-05-16:
Dat zou kunnen. Ik gebruik zelf geen HA dus ik kan je op dat gebied niet helpen. Je kunt eventueel op het Tweakers forum om hulp vragen.