dsmrreader/dsmr-reader #876: Fails to complete "dsmr_backup_create --full"

PeterEIER, 2020-02-12:
Making a backup, using the mentioned command, to a 128GB external SSD of all my DSMRreader data which is stored on a 32GB SDcard fails with the message after hours and hours with the message "disk full" which is not the case... 

Struggling for days now of creating a backup need help!

dennissiemensma, 2020-02-12:
Bedankt voor je vraag. Kan het zijn dat je database heel groot is? Je kunt dit zien met:
```
sudo du -msc /var/lib/postgresql/
```

Standaard wordt de backup in dezelfde locatie gezet als je andere backups en waar de applicatie draait. Deze locatie is niet aan te passen.

Het commando is echter gewoon een alias voor het maken van een database backup. Dezek un je dus ook handmatig uitvoeren en direct wegschrijven op je SSD.

```
sudo -u postgres pg_dump dsmrreader | gzip --fast > /pad/naar/je/ssd/dsmrreader-backup.sql.gz
```

PeterEIER, 2020-02-12:
Bedankt voor de snelle reactie!

Response op "sudo du -msc /var/lib/postgresql/"
15650	/var/lib/postgresql/
15650	total

Ik ben nu aan het stoeien met "sudo -u postgres" hopelijk lukt dat wel... ben de wanhoop nabij ;-)

dennissiemensma, 2020-02-12:
Wellicht werkt het beter in twee stappen: dump maken en daarna moven naar SSD

```
sudo su - postgres
pg_dump dsmrreader | gzip --fast > dsmrreader-backup.sql.gz

logout
sudo mv /var/lib/postgresql/dsmrreader-backup.sql.gz /pad/naar/je/ssd/
```

dennissiemensma, 2020-02-12:
Maar ik denk dat het issue vooral is dat je DB te groot is om te dumpen naar je SD-kaart (gezien je OS ook nog wat GB in beslag neemt vermoedelijk).

Overigens zou je hier ook tegenaan lopen bij de automatische backups van DSMR-reader. Wellicht nemen je huidige backups ook nog aardig wat ruimte in beslag:

```
 sudo du -msc /home/dsmr/dsmr-reader/backups/
```

PeterEIER, 2020-02-12:
maar ik heb helemaal geen backups op mijn SDcard staat... ooit maakte ik backups naar Dropbox... dropbox ingeruild voor OneDrive dat "gratis" is bij Office365.

Ofwel ik heb geen enkele backup... nu wil ik ongeveer 2 jaar data toch maar eens gaan backup-en naar een externe 128GB SSD en dat lukt vooralsnog niet.

dennissiemensma, 2020-02-12:
Ik denk dat je dan van geluk mag spreken dat je SD-kaartje inmiddels nog niet gecrashed is. Maak dan nu vooral even direct een backup van alleen je dagtotalen. Dat is een paar honderd KB.

```
sudo su - postgres
pg_dump dsmrreader --data-only --table=dsmr_stats_daystatistics --table=dsmr_stats_hourstatistics > dsmrreader-partial.sql
```

dennissiemensma, 2020-02-12:
Een ander alternatief is om even een CSV-export te maken van al je dagtotalen en uurtotalen. Dat kan via de webinterface onder export.

dennissiemensma, 2020-02-12:
En als mogelijke oplossing voor je huidige probleem. In DSMR-reader is het mogelijk om oude metingen (**alleen de metingen!**) automatisch op te schonen.  Heb je daar al eens naar gekeken?

https://dsmr-reader.readthedocs.io/en/latest/admin/datalogger.html#data-cleanup-retention

Voor de vergelijking, ik heb na ruimte 4 jaar meten slechts 200 MB aan data. Alle metingen van vroeger maken mij zelf niet zoveel uit, zolang ik de dag- en uurtotalen maar nog heb (die blijven altijd bewaard).

PeterEIER, 2020-02-12:
ja eens, mijn "SDcard" is ook eigenlijk geen "gewone" SDcard maar zo'n MO-300 SSD SATA ding met Pi-hat etc... echter die is nu dus ook gecrashed... na repair werkt alles weer dus nu zaak om snel alle (2jaar) data veilig stellen... helaas wil dat nog niet echt lukken en time running out ;-) Ja, wacht met smart op een OneDrive oplossing...

PeterEIER, 2020-02-13:
Dennis ik lees  "_The application will slowly apply retention, cleaning up a maximum of 24 hours worth of data on each backend run executed._"

Hoevaak draait die "backend run"?

Ik heb de pi losgekoppeld van mijn slimmemeter inmiddels, die backend run blijft dan nog wel werken? 

Ik moet zet 2 jaar DSMR5 data schonen... hoe lang zou dat ongeveer inbeslagnemen?
en hoe weet is dat het klaar is?

Nogmaals graag advies, bedankt alvast!

dennissiemensma, 2020-02-13:
De backend draait constant met een kort sleep van een paar seconden en die draait inderdaad los van de slimme meter. 

Hoelang elke run duurt hangt vanzelfsprekend af van hoelang de taak duurt. Je kunt dit het beste in de log zelf zien, wanneer je debug logging aanzet: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging
Vervolgens kun je met `tail -f /var/log/supervisor/dsmr_backend.log` zien wat die doet. Als retentie ingeschakeld staat, zou je iets moeten zien als `Retention: Cleaning up: ...`. 

Aan de hand daarvan kun je zelf het beste een inschatting maken hoelang het duurt.

PeterEIER, 2020-02-15:
Dennis het zal vast ergens in de uitgebreide documentatie staan... kan het alleen niet vinden... ik raak steed verder de weg kwijt... backup/restore van 2.5jaar data heb ik al opgegeven... jammer.

Ik wil een backup schrijven naar \media\mijnssd   De SSD wordt automatisch gemount bij booten, ik kan em ook vinden zowel als pi user als dmsr user... echter wat ik ook probeer DSMR-READER blijft schrijven in zijn eigen media folder...

Welk config file moet ik aanpassen?

dennissiemensma, 2020-02-16:
DSMR-reader is niet in te stellen qua pad. Je zult handmatig een backup moeten maken.

Ik zou verwachten dat dit werkt:
```
sudo su - postgres
pg_dump dsmrreader --data-only --table=dsmr_stats_daystatistics --table=dsmr_stats_hourstatistics > \media\mijnssd\dsmrreader-partial.sql
```

Als dat goed werkt, kun je een volledige backup maken:
```
sudo su - postgres
pg_dump dsmrreader | gzip --fast > \media\mijnssd\dsmrreader-full.sql.gz
```

PeterEIER, 2020-02-18:
> Reply to: dennissiemensma/dsmr-reader <reply@reply.github.com>

Dennis,

Ik moet vast niet bij jou zijn…. Echter  ik probeer nu UALEX73 zijn docker versie te draaien, hoe ik UALEX73 moet bereiken geen flauw idee… docker geeft de volgende error boodschap standard_init_linux.go:211: exec user process caused "exec format error"
Maak jij toevallig gebruik van Docker?

Mvg,
Peter

dennissiemensma, 2020-02-18:
Lijkt op: https://github.com/xirixiz/dsmr-reader-docker/issues/46

Waarschijnlijk gebruik je de verkeerde docker architectuur voor de hardware die je gebruikt. Je kunt ook eventueel de versie van xirixiz proberen. En als dat niet werkt bij hem een issue aanmaken. 

Ualex heeft in zijn project het aanmaken van namelijk issues uit staan.

PeterEIER, 2020-02-18:
Hoi Dennis,

Klopt…. Mijn verse raspbian image was het probleem, onderstaande installatie vergeten uit te voeren dom van mij, opgelost. Dank, en sorry dat ik je de laatste dagen zo lastig val…

apt update
🐳 apt install -y qemu qemu-user-static qemu-user binfmt-support

apt update

🐳 apt install -y qemu qemu-user-static qemu-user binfmt-support

Date: Tuesday, 18 February 2020 at 22:37

Lijkt op: xirixiz/dsmr-reader-docker#46<https://github.com/xirixiz/dsmr-reader-docker/issues/46>

dennissiemensma, 2020-02-19:
Is het je inmiddels al wel gelukt om die database backup te maken? En/of de database te verkleinen via retentie?

PeterEIER, 2020-02-19:
Nee, heb het opgegeven Dennis. Heb dan ook “format C:\” gedaan ik vond 20-30 uur stoeien met herstel/restore van een DSMR database wel genoeg... Clean install gedaan alles werkte weer maar ik dacht hmmm ja nu heb ik over 2 jaar weer ellende... nee ik wil programma en data op aparte SSD schijven hebben staan... dat wil niet zoals ik wil... dus weer ‘uninstall’ gedaan, ofwel opgegeven. Nu aan het stoeien met Docker, dan hoef ik in elk geval niet de hele DSMR setup te doorlopen is het idee daarachter... maar helaas alles behalve DSMR krijg ik aan de praat grrrr De dsmr en postgres etc containers draaien als een zonnetje maar de webinterface krijg ik niet te zien. Nog even en dan gebruik ik weer pen en papier werkt altijd... ;-) Tja 2.5jaar genoten van je software nu zit het ff tegen, leg het maar een maandje naast mij neer niet leuk meer.

Bedankt voor je geweldige support, jij kunt er niets aan doen dat mijn kennis te kort schiet.

dennissiemensma, 2020-02-20:
Spijtig om te horen! Heb je dan tenminste een CSV-export gemaakt van alle dagtotalen via de webinterface? Daarvoor hoef je geen moeilijke dingen te doen en dan heb je tenminste nog de historie.

closed 2020-02-20 by dennissiemensma

PeterEIER, 2020-02-20:
Ja die csv-export heb ik gemaakt van alle dagtotalen.

dennissiemensma, 2020-02-20:
Dat is fijn. Mocht je ooit weer opnieuw bezig gaan met DSMR-reader, laat het dan even weten. Ik kan die export tzt wel eenmalig voor je omzetten naar dagtotalen in het Archief. 
Je krijgt dan een SQL-bestand van me waar queries in staan om de data in de applicatie te importeren.

PeterEIER, 2020-02-20:
DSMR 3.4 !! in een docker container aan de praat gekregen, en het ziet er zelfs ook naar uit dat de database en de backups op extern Samsung 128GB SSD staan... finger crossed.

Zodra het stabiel draait allemaal maak ik graag gebruik van je aanbod.

PeterEIER, 2020-03-21:
Hallo Dennis,
  

  
Inmiddels heb ik DSMR-reader tot volle tevredenheid in een docker container draaien. Wanneer het jou gelegen is wil ik graag gebruik maken van onderstaand aanbod, ik zal je voor werk dan weer belonen met een donatie. Nb. Er is totaal geen haast bij, laat maar weten wat je van mij nodig hebt om dat sql bestand te maken.
  
  
  
  
  
Mvg,
  
Peter
 Sent from my iPad Pro

dennissiemensma, 2020-03-22:
Ik heb je de queries gemaild. Even hier voor mezelf een note hoe het omgezet is:

```
1) Header verwijderen (check volgorde velden)

2) Datum escapen:
FIND:      ^([^,]+),
REPLACE:   '$1',

3) SQL maken:
FIND:      ^(.+)$
REPLACE:   INSERT INTO dsmr_stats_daystatistics(day, electricity1, electricity2, electricity1_returned, electricity2_returned, gas, electricity1_cost, electricity2_cost, gas_cost, total_cost) VALUES($1);
```

dennissiemensma, 2020-03-22:
@PeterEIER afhankelijk van welke Docker-container gebruikt, kun je het beste even bij de degene die het onderhoud vragen hoe je SQL kunt uitvoeren.

Als dat `xirixiz` is, kun je in die Github-repo gewoon een issue maken (of even zoeken in de gesloten issues daar wellicht). Als het iemand anders is, laat maar even weten.