dsmrreader/dsmr-reader #877: ImportError: cannot import name 'read_serial_port'

Calimerorulez, 2020-02-14:
DSMR-Reader v3.3.0
DSMR v4 op een Intel NUC in een docker container

Sinds de upgrade naar versie 3.3.0 krijg ik onderstaande error:

`ImportError: cannot import name 'read_serial_port' from 'dsmr_datalogger.scripts.dsmr_datalogger_api_client' (/dsmr/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py)`

dennissiemensma, 2020-02-14:
Ik denk dat je dan het beste even voor nu terug kan gaan naar v3.2

Calimerorulez, 2020-02-14:
Doe ik dat, dank :)

dennissiemensma, 2020-02-15:
@Calimerorulez maar het moet dan nog wel opgelost worden. Welke docker image gebruik je?

Phontana, 2020-02-18:
@dennissiemensma Ik heb hetzelfde issue. Ik gebruik de docker image `ualex73/dsmr-reader-docker:arm32v6`.

dennissiemensma, 2020-02-18:
Bedankt voor je aanvulling. Ik denk dat ik het dan al weet. Ik heb een script aangepast, wat ualex los in zijn repo heeft. Opzich logisch, want dat script werkt ik bijna nooit bij.

Ik zal bij hem een issue maken zodat die zijn eigen repo kan bijwerken met de nieuwere versie van het script.

dennissiemensma, 2020-02-18:
Ik kan geen issue maken, dus heb het indirect via een PR gedaan. https://github.com/ualex73/dsmr-reader-docker/pull/2

dennissiemensma, 2020-02-19:
Ik sluit deze voor nu. Mocht Ualex er zelf niet uitkomen, dan kan ik altijd nog assisteren waar mogelijk.

closed 2020-02-19 by dennissiemensma

ualex73, 2020-02-20:
Het is nu gefixt, een rebuild van 3.3.0 loopt nu ;-) @dennissiemensma, je PR was 99% goed. Ik moest alleen "SLEEP=..." en "import os" toevoegen en toen was Python ook weer blij. Alleen ik kan het zelf niet testen, want ik gebruik de client zelf ook niet.

ualex73, 2020-02-20:
BTW - Ik heb een aangepast script gemaakt voor de client, anders moet zelf direct in de python scripts gaan editten om het werkend te krijgen (volgens mij werkt de xirixiz image zo?)

dennissiemensma, 2020-02-20:
Bedankt voor de update! Wellicht dat ik het standaardscript dan een keer uitbreid met de envvars. Dan hoef je ook niet moeilijk te doen.